OpenCBM/OpenCBM #3: Unable to build on OS X

bodgit, 2018-03-08:
Using the bundled Portfile (and changing the git URL from git://git.code.sf.net/p/opencbm/code to https://github.com/OpenCBM/OpenCBM.git) I'm unable to get things to build. It seems to be failing on xum1541cfg:
```
...
:info:build /Applications/Xcode.app/Contents/Developer/usr/bin/make -C xum1541cfg -f LINUX/Makefile all
:info:build make[1]: Entering directory `/opt/local/var/macports/build/_Users_matt_Documents_ports_emulators_opencbm/opencbm/work/code/xum1541cfg'
:info:build gcc -g -O2 -Wall -I../include -I../include/LINUX -DPREFIX=\"/opt/local\" -DOPENCBM_CONFIG_FILE=\"/opt/local/etc/opencbm.conf\"  -g -I /opt/local/include -I dfu-programmer-0.5.4/src -Wall -o main.o -c main.c
:info:build main.c:33:10: fatal error: 'usb.h' file not found
:info:build #include "usb.h"
:info:build          ^
:info:build 1 error generated.
:info:build make[1]: *** [main.o] Error 1
:info:build make[1]: Leaving directory `/opt/local/var/macports/build/_Users_matt_Documents_ports_emulators_opencbm/opencbm/work/code/xum1541cfg'
:info:build make: *** [BUILDSYSTEM.xum1541cfg.all] Error 2
```

spiro-trikaliotis, 2018-03-09:
I know this problem, I was already informed about it via mail. Unfortunately, I needed some more info, but I did not get it yet, so I could not fix it.

If you want to do me a favor, can you please check if libusb-0.1 is installed (for development)? And if so, where is the file usb.h located on your machine?

If you just need a quick fix, then you can also fix it by removing the xum1541cfg generation. For this, remove the part "xum1541cfg" from LINUX/Makefile, line 22:

That is, change                                                                                                                                                                                              
SUBDIRS_PLUGIN_XUM1541 = opencbm/lib/plugin/xum1541 xum1541cfg
to
SUBDIRS_PLUGIN_XUM1541 = opencbm/lib/plugin/xum1541

bodgit, 2018-03-12:
I've got a libusb-legacy port installed which has /opt/local/include/libusb-legacy/usb.h

ttkoistinen, 2019-02-01:
Hi, I'm also trying to build OpenCBM for macOS Mojave 10.14.3 to get ZoomFloppy connected. When I tried the given Portfile the build stopped with the following error:
`/opt/local/var/macports/build/_Users_<name_removed>_ports_emulators_opencbm/opencbm/work/code/xu1541/lib
:info:build cc -I/opt/local/include/libusb-legacy -I/opt/local/include/ -I/opt/local/include -I../include/ -Wall -L/opt/local/lib/libusb-legacy -lusb-legacy -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-prebind -s -L/opt/local/lib -o usb_echo_test usb_echo_test.c -L../lib -lxu1541 -lusb
:info:build ld: warning: option -s is obsolete and being ignored
:info:build ld: warning: option -prebind is obsolete and being ignored
:info:build ld: library not found for -lusb
:info:build clang: error: linker command failed with exit code 1 (use -v to see invocation)
:info:build make[1]: *** [usb_echo_test] Error 1
:info:build make[1]: Leaving directory /opt/local/var/macports/build/_Users_tommikoistinen_ports_emulators_opencbm/opencbm/work/code/xu1541/misc
:info:build make: *** [plugin-xu1541] Error 2
:info:build make: Leaving directory `

I did also try to compile a clean clone directly from GitHub by removing the xu1541 plug-in installation. However, then for example the cbmctrl detect command just outputs "An error occurred opening OpenCBM, aborting..."

In both cases I had the following ports active:
cc65 @2.13.3_0 (active)
libusb-legacy @0.1.12_4 (active)

Any ideas what to try next? Or could someone confirm a successful installation with macOS Mojave? Thanks!

spiro-trikaliotis, 2019-02-01:
I do not have MacOS, so it is not very easy for me to debug.
It seems that libusb-legacy is not found.
Does MacOS have separate "development" packages for libraries? For example, on Debian, I have to install libusb-0.1-4 to actually USE libusb, but also libusb-dev for COMPILING programs that use it.

Is it the same on MacOS?

If you think it is installed, can you search for libusb.h on your HDD? Where is it located?

Note that the warnings about "-s" and "-prebind" being ignored might be the reason. I do not know what they are doing, but -s gives the path to /opt/local/lib where the libusb linker file is most probably located. Do you have a man page of the build tool that is used?

Perhaps someone with MacOS experience can help here?

ttkoistinen, 2019-02-02:
Here's an update on where I have ended so far.
When installing the port libusb-legacy, the following files are created in:

/opt/local/lib/libusb-legacy
	Libusb-legacy.a
	Libusb-legacy.dylib
/opt/local/include/libusb-legacy/usb.h
/opt/local/bin/libusb-legacy-config

The last one is used in OpenCBM/xu1541/include/common.mk makefile to get the library and include paths. The following is in common.mk:

CFLAGS_EXTRA =
 #determine libusb location
 OS=$(shell uname -s)
 ifeq "$(OS)" "Darwin"
   #MacOS compilation:
   LIBUSB_DIR = $(shell libusb-legacy-config --prefix)
   LDFLAGS_EXTRA = $(shell libusb-legacy-config --libs)
   CFLAGS_EXTRA = $(shell libusb-legacy-config --cflags)
…
CFLAGS_EXTRA+=-I$(LIBUSB_DIR)/include/

The libusb-legacy-config helper tool gives the following outputs:
LIBUSB_DIR = /opt/local
LDFLAGS_EXTRA = -L/opt/local/lib/libusb-legacy -lusb-legacy -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-prebind
CFLAGS_EXTRA = -I/opt/local/include/libusb-legacy

Now these values are used in OpenCBM/xu1541/misc/Makefile:

CFLAGS  = $(CFLAGS_EXTRA) -I$(LIBUSB_DIR)/include -I../include/ -Wall
LDFLAGS = $(LDFLAGS_EXTRA) -L$(LIBUSB_DIR)/lib
LIBS    = -L../lib -lxu1541$(LIB_WIN) -lusb-legacy

The Makefile generates: $(CC) $(CFLAGS) $(LDFLAGS) -o $@ $< $(LIBS)
Which is the same as the lines causing the build to fail:

'build cc -I/opt/local/include/libusb-legacy -I/opt/local/include/ -I/opt/local/include -I../include/ -Wall -L/opt/local/lib/libusb-legacy -lusb-legacy -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-prebind -s -L/opt/local/lib -o usb_echo_test usb_echo_test.c -L../lib -lxu1541 -lusb
:info:build ld: warning: option -s is obsolete and being ignored
:info:build ld: warning: option -prebind is obsolete and being ignored
:info:build ld: library not found for -lusb'

The -s option is to not include debugging info in output.

The problem seems to be the last library include option (-lusb) as there is no library called “libusb” in macOS port, but only “libusb-legacy”. The port file produces an error at this point. 

Stopped debugging the Portfiles installation here.

Then, attempting to edit the source files and compile directly from GitHub.

The OpenCBM part seemed to compile but then I realized the config.make file does a test whether libusb0 or libusb1 is in use. This test doesn’t work in Mac as the libusb-legacy cannot be found.

Test in config.make file:
HAVE_LIBUSB0_USB_H = ${shell test -e /usr/include/usb.h && echo -DHAVE_LIBUSB0_USB_H=1}
HAVE_LIBUSB1_LIBUSB_H = ${shell test -e /usr/include/libusb-1.0/libusb.h && echo -DHAVE_LIBUSB1_LIBUSB_H=1}

ifneq ($(strip $(HAVE_LIBUSB0_USB_H)),)
  HAVE_LIBUSB=-DHAVE_LIBUSB=1
endif

ifneq ($(strip $(HAVE_LIBUSB1_LIBUSB_H)),)
  HAVE_LIBUSB=-DHAVE_LIBUSB=1
endif

In top level makefile the usb0 option related to above test was empty, so copied the same plugins in both:

ifeq ($(strip $(HAVE_LIBUSB)),)
SUBDIRS_PLUGIN_XU1541 = opencbm/lib/plugin/xu1541
SUBDIRS_PLUGIN_XUM1541 = opencbm/lib/plugin/xum1541 xum1541cfg
else
SUBDIRS_PLUGIN_XU1541 = opencbm/lib/plugin/xu1541
SUBDIRS_PLUGIN_XUM1541 = opencbm/lib/plugin/xum1541 xum1541cfg
endif

Now the make starts to compile the xum1541 part but stops right away with the following:

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C opencbm/lib/plugin/xum1541 -f LINUX/Makefile install
install -d /opt/local/lib/opencbm/plugin/
install -m 755 libopencbm-xum1541.0.4.99.99.dylib /opt/local/lib/opencbm/plugin/
install: libopencbm-xum1541.0.4.99.99.dylib: No such file or directory
make[1]: *** [install-plugin] Error 71
make: *** [BUILDSYSTEM.opencbm/lib/plugin/xum1541.install] Error 2

Will try to continue from here...any hints would be great! Of course I could try the same in a Windows machine but would prefer getting this running in Mac.

silverdr, 2019-02-02:
FWIW and FYI - I got it running on Mac some time ago so it certainly was possible (and AFAIR not /that/ hard) but it was surely not on "Mojave" and things might have changed since then. If you don't get it done, please drop a line and I'll see if I can find time to check it on the current OS.

ttkoistinen, 2019-02-03:
silvered, thanks for the offer! I will one more time start over and try also some older versions from the source forge file directory and will then update if I got it running.

greg-king5, 2019-02-03:
> **...**
> The last one is used in the "OpenCBM/xu1541/include/common.mk" makefile to get the library and include paths. The following is in "common.mk":
> 
> CFLAGS_EXTRA =
> #determine libusb location
> OS=$(shell uname -s)
> ifeq "$(OS)" "Darwin"
> #MacOS compilation:
> LIBUSB_DIR = $(shell libusb-legacy-config --prefix)
> LDFLAGS_EXTRA = $(shell libusb-legacy-config --libs)
> CFLAGS_EXTRA = $(shell libusb-legacy-config --cflags)
> …
> CFLAGS_EXTRA+=-I$(LIBUSB_DIR)/include/
> 
> The "libusb-legacy-config" helper tool gives the following outputs:
> LIBUSB_DIR = /opt/local
> LDFLAGS_EXTRA = -L/opt/local/lib/libusb-legacy -lusb-legacy -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-prebind
> CFLAGS_EXTRA = -I/opt/local/include/libusb-legacy
> 
> Now, these values are used in "OpenCBM/xu1541/misc/Makefile":
> 
> CFLAGS = $(CFLAGS_EXTRA) -I$(LIBUSB_DIR)/include -I../include/ -Wall
> LDFLAGS = $(LDFLAGS_EXTRA) -L$(LIBUSB_DIR)/lib
> LIBS = -L../lib -lxu1541$(LIB_WIN) -lusb-legacy
> 
> The Makefile generates: $(CC) $(CFLAGS) $(LDFLAGS) -o $@ $< $(LIBS)
> Which is the same as the lines causing the build to fail:
> 
> 'build cc -I/opt/local/include/libusb-legacy -I/opt/local/include/ -I/opt/local/include -I../include/ -Wall -L/opt/local/lib/libusb-legacy -lusb-legacy -Wl,-framework -Wl,IOKit -Wl,-framework -Wl,CoreFoundation -Wl,-prebind -s -L/opt/local/lib -o usb_echo_test usb_echo_test.c -L../lib -lxu1541 -lusb
> 
> :info:build ld: library not found for -lusb'
> 
> The problem seems to be the last library include option (-lusb), as there is no library called “libusb” in macOS port, but only “libusb-legacy”. The port file produces an error at this point.

The problem is that the line is too long!  Therefore, its end was chopped off [compare the value of $(LIBS) with the end of that command].  The line is too long because "Makefile" put things in the line that don't belong there!

Change these lines in "OpenCBM/xu1541/misc/Makefile" to:
```make
CFLAGS = $(CFLAGS_EXTRA) -I../include -Wall
LDFLAGS = $(LDFLAGS_EXTRA)
LIBS = -L../lib -lxu1541$(LIB_WIN)
```

ttkoistinen, 2019-02-03:
Yes, I noticed a typo in my previous message, the line in OpenCBM/xu1541/misc/Makefile which is currently in github and gets downloaded as part of Mac port is:
LIBS = -L../lib -lxu1541$(LIB_WIN) -lusb
I changed it to "-lusb-legacy" in my later attempts but still run into some other issues as in my previous email. Thanks for commenting, I will try that also.

ttkoistinen, 2019-02-03:
Got it working after hard work! ;-) A couple of things I had to do:
- For some reason, the top level make did not process the subdirectories for plugin libraries, so I had to go to each plugin directory and run the sub-makefile with "build-lib" for each, for example for OpenCBM/lib/plugin/xum1541: 
'sudo make -f LINUX/Makefile PREFIX=/opt/local MANDIR=/opt/local/share/man/man1 build-lib -Wall'
- Then I encountered an error with usb.h in /xum1541cfg/Linux/Makefile: "fatal error: 'usb.h' file not found" (probably the same thing as reported by the original creator of this thread). There is the following line: LINK_FLAGS= -L/opt/local/lib -lusb in the xum1541cfg makefile where the lib is not found due the name, anyway removed the whole xum1541cfg.
The build was then completing.

The ZoomFloppy connects now and I'm able to see my files in the real 1541 drive!

cbmctrl detect
8: 1540 or 1541

cbmctrl status 8
00, ok,00,00

Thanks for the help! At least for me, the automatic Portfile installation did not work, but fetching the package from GitHub and performing those above mentioned changes helped.

bodgit, 2019-02-03:
I've literally just gotten a new Mac running Mojave so I'm in the process of trying to (re)install all my favourite tools and remembered I didn't get this working on my old Mac.

The MacPorts `libusb-legacy` port, which I think is the one we need to be using, ships a `.pc` file so instead of hardcoding the directories and/or library names you can instead do:
```
$ pkg-config --cflags libusb-legacy
-I/opt/local/include/libusb-legacy
$ pkg-config --libs libusb-legacy
-L/opt/local/lib/libusb-legacy -lusb-legacy
```
Using `pkg-config` in the Makefile might make things a bit more portable between systems. I will see if I can get it building reliably and submit a PR if I'm successful.

bodgit, 2019-02-04:
I've created PR #12 which I think fixes the build on my Mojave installation. It now builds without error, (there are lots of warnings but these don't stop the build itself) and the port contains the following files:
```
$ port contents opencbm
Port opencbm contains:
  /opt/local/bin/cbmcopy
  /opt/local/bin/cbmctrl
  /opt/local/bin/cbmformat
  /opt/local/bin/cbmforng
  /opt/local/bin/cbmlinetester
  /opt/local/bin/cbmread
  /opt/local/bin/cbmrpm41
  /opt/local/bin/cbmwrite
  /opt/local/bin/d64copy
  /opt/local/bin/d82copy
  /opt/local/bin/frm_analyzer
  /opt/local/bin/imgcopy
  /opt/local/bin/opencbm_plugin_helper_tools
  /opt/local/bin/samplelibtransf
  /opt/local/bin/testlines
  /opt/local/bin/xum1541cfg
  /opt/local/etc/opencbm.conf
  /opt/local/etc/opencbm.conf.d/00opencbm.conf
  /opt/local/etc/opencbm.conf.d/10xu1541.conf
  /opt/local/etc/opencbm.conf.d/10xum1541.conf
  /opt/local/include/cbmcopy.h
  /opt/local/include/d64copy.h
  /opt/local/include/opencbm.h
  /opt/local/lib/libopencbm.0.4.99.99.dylib
  /opt/local/lib/libopencbm.0.dylib
  /opt/local/lib/libopencbm.a
  /opt/local/lib/libopencbm.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xu1541.0.4.99.99.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xu1541.0.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xu1541.a
  /opt/local/lib/opencbm/plugin/libopencbm-xu1541.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xum1541.0.4.99.99.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xum1541.0.dylib
  /opt/local/lib/opencbm/plugin/libopencbm-xum1541.a
  /opt/local/lib/opencbm/plugin/libopencbm-xum1541.dylib
  /opt/local/lib/opencbm/plugin/xu1541/read_event_log
  /opt/local/lib/opencbm/plugin/xu1541/usb_echo_test
  /opt/local/share/man/man1/cbmcopy.1.gz
  /opt/local/share/man/man1/cbmctrl.1.gz
  /opt/local/share/man/man1/cbmformat.1.gz
  /opt/local/share/man/man1/cbmforng.1.gz
  /opt/local/share/man/man1/cbmlinetester.1.gz
  /opt/local/share/man/man1/cbmread.1.gz
  /opt/local/share/man/man1/cbmrpm41.1.gz
  /opt/local/share/man/man1/cbmwrite.1.gz
  /opt/local/share/man/man1/d64copy.1.gz
  /opt/local/share/man/man1/d82copy.1.gz
  /opt/local/share/man/man1/frm_analyzer.1.gz
  /opt/local/share/man/man1/imgcopy.1.gz
  /opt/local/share/man/man1/samplelibtransf.1.gz
  /opt/local/share/man/man1/testlines.1.gz
  /opt/local/share/man/man1/xum1541cfg.1.gz
```
Note `xum1541cfg` and the `libopencbm-xum1541*.dylib` files are present which previously either weren't being built or would stop the build from completing.

I haven't tested it with my ZoomFloppy yet as I need to dig it out from storage along with a 1541 or 1571 to test it with so if someone else wants to try it out, you will need to apply this small diff to the `Portfile` to point to my branch:
```diff
--- opencbm/ports/MacPorts/Portfile     2019-02-03 15:55:36.000000000 +0000
+++ /Users/matt/ports/emulators/opencbm/Portfile        2019-02-03 21:57:09.000000000 +0000
@@ -23,8 +23,8 @@
 depends_lib         port:libusb-legacy
 
 fetch.type          git
-git.url             https://github.com/OpenCBM/OpenCBM.git
-git.branch          master
+git.url             https://github.com/bodgit/OpenCBM.git
+git.branch          darwin
 
 checksum {}
 
```

closed 2019-02-04 by spiro-trikaliotis

spiro-trikaliotis, 2019-02-04:
I removed the .gitignore, because its against the way I work. Furthermore, I had to a some small fix in the Makefile of the xu1541/misc/ directory.
Other than that, the patch looks good. 

Thank you for your work.

PR #12 (2019-02-04): Fix building on macOS/OS X/Darwin
* Unconditionally define `HAVE_LIBUSB` rather than test for it
* Fix building of xu1541 and xum1541cfg
* Correct git URL to point to github instead of sourceforge
* Add basic .gitignore to ignore compiler output

Fixes #3