OpenCBM/OpenCBM #10: ioctl() usage in xa1541 plugin

Zirias, 2018-11-05:
This weekend, I ported the kernel module to FreeBSD and was hoping not to touch any other code, which unfortunately didn't work because of the way ioctls are used:

1. All ioctls are defined as operations without data (only `_IO()` macros are used, no `_IOW()`, `_IOR()`, `_IOWR()`) and calls that need to pass data to the driver just do it without declaring it. While this seems to work on Linux, in FreeBSD, the data isn't received at the driver side.
2. For reading data from the driver, the `ioctl()` return value is used. This doesn't work on FreeBSD either, `ioctl()` will only return `0` or `-1` with `errno` set.

So, In my ported driver, I defined the ioctls correctly and adapted the code in the plugin. It works, but it would be much nicer to have them defined correctly upstream. Of course, the Linux module would need some changes for this as well. I can't do them myself because I have no Linux machine on metal for testing right now. But I could prepare a patch for the plugin code which someone with a Linux machine could apply and adapt the driver there?

Finally, the parburst calls read and write directly to/from userspace memory. AFAIK, this won't work on all architectures, so I'd like to change this as well by having an in-kernel buffer that's large enough -- what do you think about that?

Zirias, 2018-11-05:
The patches needed for this to work with FreeBSD are now in my port in the files directory:
https://github.com/Zirias/zfbsd-ports/tree/master/new/archivers/opencbm

spiro-trikaliotis, 2018-11-17:
Hello Felix,
this is just a short notice that I have seen your patches and they look reasonable to me.
Unfortunately, at the moment, I am lacking the time to test them.

I will do as soon as possible.

Zirias, 2018-11-24:
Hi Spiro,

thanks for feedback, I'm glad parallel-port isn't considered legacy and "wontfix" right away ;) The patches aren't exactly clean right now, an unrelated one e.g. just changes a hardcoded `make`  to `gmake` so it builds on FreeBSD. If you are interested to incorporate the changes, I can offer to prepare a clean pull-request. Of course, as stated above, the necessary changes in the Linux kernel module need an installation of Linux *on metal* to test (AFAIK, virtualizing the parallel port won't do because timing won't be exact). I can *try* to do these changes as well, but someone else would have to test.

As for always copying the data for the parburst calls, I did that in my FreeBSD driver and it works quite well. Given the recent use of SMAP, I'd suggest to do this in the Linux driver as well -- it's the easiest "way out" :) I'd also suggest to drop "direct port access" from the Linux driver, to simplify the code a bit.

Would you be interested to adopt the FreeBSD driver in your code base or do you prefer to keep it external while giving an option to build the xa1541 plugin on FreeBSD without a driver?

BR, Felix

spiro-trikaliotis, 2018-11-24:
Of course, a clean pull request would be easiest for me. Of course, I would have to test it before merging it.

If it is possible to include the FreeBSD driver without too much special casing, I would prefer having it in the repository. I think it is not optimal if we split the sources into different places.

Regarding the copying of the data in the parbust calls, I am not sure if I understand you correctly. Do you suggest to copy the data around, as it does not have any negative effect? Or what is your proposal?

I must admit that I did not follow Kernel activity for some years now, so I am not up-to-date in this respect.

Zirias, 2018-11-25:
> Regarding the copying of the data in the parbust calls, I am not sure if I understand you correctly. Do you suggest to copy the data around, as it does not have any negative effect?

Exactly. For a bit of background, I stumbled upon this problem while porting the driver, looking for a FreeBSD equivalent to `access_ok()` -- there doesn't seem to be any. I tested a ported version that just omits this test and it worked for me, but then changed the implementation to use `copyin()`/`copyout()` instead (I think the Linux equivalents are `copy_from_user()` etc ...) and this worked flawlessly as well.

The version with direct memory access stops working on the current FreeBSD 11.2, because [SMAP](https://en.wikipedia.org/wiki/Supervisor_Mode_Access_Prevention) is active there. As I read `access_ok()` on Linux really only tests that the memory is indeed in userspace and temporarily deactivating SMAP takes a bit more code, I *assume* the parburst calls might not work any more on recent Linux versions -- the simplest fix is to use the copy in/out functions. On a side note, I think it's a bit strange how the current implementation carefully uses these copy functions on a `struct` with just a pointer and the length, but then goes on dereferencing this pointer to userspace :)

Zirias, 2019-01-02:
Any chance to get my pull req tested on Linux (successfully tested by myself on FreeBSD of course) and integrated? If there's anything I should change or could help with, please let me know :) Btw, happy new year!

spiro-trikaliotis, 2019-01-06:
I am sorry, I do not have the time at the moment to do the tests. At the moment, that's the only thing that prevents your changes from being merged.

spiro-trikaliotis, 2019-02-06:
I just merged your changes. Thank you very much.
I had to do some fixes, though, to make it compile.

closed 2019-02-06 by spiro-trikaliotis