OpenCBM/OpenCBM #13: Can't see ZoomFloppy on OS X

bodgit, 2019-02-09:
Hello,

After getting OpenCBM to build on OS X, I managed to find my ZoomFloppy board and plugged it in. The LED is lit however it doesn't seem to be detected by any of the various utilities. From a bit of googling, I found I can set `XUM1541_DEBUG=3` to increase the debugging. Here's what I get:
```
$ cbmctrl detect
[XUM1541] scanning usb ...
[XUM1541] scanning bus 001
[XUM1541] device 05ac:8007 at 000-05ac-8007-09-ff
[XUM1541] scanning bus 002
[XUM1541] device 05ac:8007 at 000-05ac-8007-09-ff
error: no xum1541 device found
An error occurred opening OpenCBM, aborting...
```
The weird thing is my Mac _can_ see the ZoomFloppy USB device:

[image]

I have the ZoomFloppy directly attached, albeit with a USB C to A adapter as I have only C ports. Any ideas what else I can try?

bodgit, 2019-02-10:
While I still have my old Mac, I've built OpenCBM on that machine using the same Portfile and on that machine I can see the ZoomFloppy:
```
$ cbmctrl detect
[XUM1541] scanning usb ...
[XUM1541] scanning bus 001
[XUM1541] device 05ac:8006 at 000-05ac-8006-09-ff
[XUM1541] device 0424:2513 at 001-0424-2513-09-00
[XUM1541] device 05ac:8404 at 002-05ac-8404-00-00
[XUM1541] device 16d0:0504 at 003-16d0-0504-ff-00
[XUM1541] found xu/xum1541 version 0207 on bus 001, device 003-16d0-0504-ff-00
[XUM1541] xum1541 name: xum1541 floppy adapter (ZOOMFLOPPY)
[XUM1541] xum1541 serial number:   0
[XUM1541] firmware version 7, library version 8
xum1541 firmware version too low (7 < 8)
please update your xum1541 firmware
[XUM1541] Closing USB link
An error occurred opening OpenCBM, aborting...
```
It's plugged directly into one of the USB A ports:

[image]

So it works on a 2011 MacBook Air running 10.13.x High Sierra but doesn't work on a 2018 MacBook Pro running 10.14.x Mojave, however I think the OS version is probably irrelevant. Annoyingly I can't upgrade the older machine to Mojave as it's not supported. My hunch is this is something to do with enumerating the USB bus.

spiro-trikaliotis, 2019-02-11:
As I don't know Mac OS, I cannot help you here.
Perhaps, someone else can jump into the discussion?

spiro-trikaliotis, 2019-02-11:
Oh, and I forgot: you should update your firmware.
This should not affect your current problem, though.

bodgit, 2019-02-11:
> Oh, and I forgot: you should update your firmware.
This should not affect your current problem, though.

Yes, I'm amazed my ZoomFloppy was even on version 7 given how long I've had it 😄 

I've been thinking that using the legacy version of libusb might have something to do with this problem given it was likely written before USB 3.x was a thing and from a quick look the code only seems to be doing something as simple as looping through every device on every bus looking for a specific product and vendor ID so I'm going to write some separate test code that uses both the legacy libusb and the current version of the library and see if I can just find the device by its ID's. If I can't find it using the legacy version then that's the likely cause.

silverdr, 2019-02-11:
> So it works on a 2011 MacBook Air running 10.13.x High Sierra but doesn't work on a 2018 MacBook Pro running 10.14.x Mojave, however I think the OS version is probably irrelevant

Actually that'd be the first thing I'd look into (whether the libusb variant in use interfaces properly with the OS inner structures, which might have changed between the OS versions in question).

bodgit, 2019-02-11:
You can see from my first debug snippet that the code does seem to be enumerating at least some, possibly all, of the USB buses, so it's not like it can't see them. However, I think I'm correct in saying this legacy version of libusb is now unmaintained so if it turns out it doesn't work with USB 3 (or whatever is special about this machine), it's likely not going to be updated and the fix is probably to migrate to the newer version of the library given it's only going to become the norm for hardware going forwards.

Anyway, I'll try and whip up some test code first and see what happens.

bodgit, 2019-02-11:
Okay, I've done some testing with both libusb-0.1.12 (the legacy version) and libusb-1.0.22 (the current version). Both source repositories contain an example/test `testlibusb.c` program to list every available USB device so I didn't have to write anything in the end.

With libusb-0.1.12:
```
$ gcc -o testlibusb testlibusb.c $(pkg-config --cflags --libs libusb-legacy)
$ otool -L ./testlibusb
./testlibusb:
        /opt/local/lib/libusb-legacy/libusb-legacy-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.200.5)
$ ./testlibusb
Dev #0: 05AC - 8007
Dev #0: 05AC - 8007
```
These two devices are apparently "XHCI Root Hub USB 2.0 Simulation" if I google the vendor/product ID.

With libusb-1.0.22:
```
$ gcc -o testlibusb testlibusb.c $(pkg-config --cflags --libs libusb-1.0)
$ otool -L ./testlibusb
./testlibusb:
        /opt/local/lib/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.200.5)
$ ./testlibusb
Dev (bus 20, device 1): Nate Lawson and OpenCBM team - xum1541 floppy adapter (ZOOMFLOPPY)
Dev (bus 128, device 2): Apple Inc. - Apple Internal Keyboard / Trackpad
Dev (bus 128, device 6): Apple - Headset
```
There is the ZoomFloppy, interestingly the two devices listed by the legacy version aren't there and it's also found the keyboard/trackpad and headset devices which also weren't listed previously.

I've also found there is a libusb-compat library which shims the newer libusb library with the older  legacy API so if I recompile the legacy version of testlibusb.c using the compat library:
```
$ gcc -o testlibusb testlibusb.c $(pkg-config --cflags --libs libusb)
$ otool -L ./testlibusb
./testlibusb:
        /opt/local/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.200.5)
$ ./testlibusb 
Dev #1: Nate Lawson and OpenCBM team - xum1541 floppy adapter (ZOOMFLOPPY)
Dev #2: Apple Inc. - Apple Internal Keyboard / Trackpad
Dev #6: Apple - Headset
```
This now also finds the ZoomFloppy, so it looks like using a newer libusb is the way to go.

Rather than try and port the whole codebase to use the newer libusb API (unless you've already started to do this?), it might be worth using this compat library instead. I will try and create a PR with the necessary changes.

spiro-trikaliotis, 2019-02-12:
Thank you for your analysis.
The patch would only affect MacOS, wouldn't it?

I am already working on libusb1, I even have a patch. Unfortunately, it does not apply on the current repository anymore, so it is not as easy as it should be.

bodgit, 2019-02-12:
> The patch would only affect MacOS, wouldn't it?

Yes, it's actually quite a small set of changes, I have a branch with them here:

https://github.com/bodgit/OpenCBM/tree/libusb-compat

It compiles and `cbmctrl detect` now finds my ZoomFloppy (and complains about its firmware version), however I should probably test it with an actual drive to make sure it works fully before you merge it.

It would be also helpful if anyone has a Mac that can run both High Sierra (10.13.x) and Mojave (10.14.x) to see if it's the OS version or the hardware change that broke things.

> I am already working on libusb1, I even have a patch. Unfortunately, it does not apply on the current repository anymore, so it is not as easy as it should be.

If you need any help testing that, I'm happy to help. It definitely seems the way forward given that legacy version of the library hasn't been touched since 2006. However it looks like the API has changed quite a bit.

closed 2019-02-13 by spiro-trikaliotis

spiro-trikaliotis, 2019-03-02:
Hello @bodgit,
I pushed the libusb1 support on the branch libusb0_and_libusb1. If it finds a libusb1 development environment, it will use that instead of libusb0.

Do you want to give it a try?

PR #14 (2019-02-13): Switch to using libusb-compat port on OS X
Using this build I was able to see my attached ZoomFloppy under OS X 10.14.x Mojave, update it to the version 8 firmware and also read a disk with a connected 1571 drive.

Fixes #13