OpenCBM/OpenCBM #16: Cannot compile on OS X 10.14.2

alien2108, 2019-09-27:
Compilation dies here:

_make[2]: Entering directory `/opt/local/var/macports/build/_Users_alien_ports_emulators_opencbm/opencbm/work/code/xu1541/lib'
cc -I/opt/local/include -I/opt/local/include/ -g -Wall -pedantic -I../include/   -c -o close.o close.c
In file included from close.c:1:
../include/xu1541lib.h:57:8: error: unknown type name 'libusb_device_handle'
extern libusb_device_handle * xu1541lib_find(void);
       ^
../include/xu1541lib.h:58:8: error: unknown type name 'libusb_device_handle'
extern libusb_device_handle * xu1541lib_find_in_bootmode(unsigned int *p_soft_bootloader_mode);
       ^
../include/xu1541lib.h:59:47: error: unknown type name 'libusb_device_handle'
extern void                   xu1541lib_close(libusb_device_handle *handle);
                                              ^
../include/xu1541lib.h:60:57: error: unknown type name 'libusb_device_handle'
extern int                    xu1541lib_get_device_info(libusb_device_handle *handle, xu1541_device_info_t *device_info, unsigned int device_info_length);
                                                        ^
../include/xu1541lib.h:61:61: error: unknown type name 'libusb_device_handle'
extern void                   xu1541lib_display_device_info(libusb_device_handle *handle);
                                                            ^
../include/xu1541lib.h:62:63: error: unknown type name 'libusb_device_handle'
extern int                    xu1541lib_is_in_bootloader_mode(libusb_device_handle *handle);
                                                              ^
../include/xu1541lib.h:63:46: error: unknown type name 'libusb_device_handle'
extern void                   xu1541lib_wait(libusb_device_handle *handle);
                                             ^
../include/xu1541lib.h:64:58: error: unknown type name 'libusb_device_handle'
extern int                    xu1541lib_set_to_boot_mode(libusb_device_handle *handle);
                                                         ^
../include/xu1541lib.h:65:54: error: unknown type name 'libusb_device_handle'
extern int                    xu1541lib_get_pagesize(libusb_device_handle *handle);
                                                     ^
close.c:9:22: error: unknown type name 'libusb_device_handle'
void xu1541lib_close(libusb_device_handle *handle) {
                     ^
close.c:11:13: warning: implicit declaration of function 'libusb_release_interface' is invalid in C99 [-Wimplicit-function-declaration]
  int ret = libusb_release_interface(handle, 0);
            ^
close.c:12:14: error: use of undeclared identifier 'LIBUSB_SUCCESS'
  if (ret != LIBUSB_SUCCESS) {
             ^
close.c:13:40: warning: implicit declaration of function 'libusb_error_name' is invalid in C99 [-Wimplicit-function-declaration]
    fprintf(stderr, "USB error: %s\n", libusb_error_name(ret));
                                       ^
close.c:13:40: warning: format specifies type 'char *' but the argument has type 'int' [-Wformat]
    fprintf(stderr, "USB error: %s\n", libusb_error_name(ret));
                                ~~     ^~~~~~~~~~~~~~~~~~~~~~
                                %d
close.c:17:3: warning: implicit declaration of function 'libusb_close' is invalid in C99 [-Wimplicit-function-declaration]
  libusb_close(handle);
  ^
4 warnings and 11 errors generated._

The issue has to do something with the fact that wrong/no headers are included.

"libusb_device_handle" function is found in "/opt/local/include/libusb-1.0/libusb.h" but NOT in "/opt/local/include/usb.h"

As you can see "cc..." is totally missing the -DHAVE_LIBUSB=1 -DHAVE_LIBUSB1=1 -DHAVE_LIBUSB_1_0=1 stuff....thus "../include/xu1541lib.h" does not include anything:

#if HAVE_LIBUSB0
#include <usb.h>
#elif HAVE_LIBUSB1
#include <libusb.h>
#endif

lafiosca, 2019-10-05:
I'm having a similar problem, although my output is not the same as yours. I am on a fresh-ish OS X 10.14.6 system. I installed MacPorts specifically for this, following the directions at https://github.com/OpenCBM/OpenCBM/blob/master/README-OSX.txt

Here's the failure output, although it's not from the first time I ran it. I tried more than once: https://gist.github.com/lafiosca/385c887e898c0e8da680ed0218a30df0

I found it suspicious that there were a lot of lines with `:info:build /bin/sh: pkg-config: command not found`, like: https://gist.github.com/lafiosca/385c887e898c0e8da680ed0218a30df0#file-main-log-L367

`pkg-config` was not listed as a prerequisite in the OSX README, but it seems that it is what's used to recognize which version of the USB library is installed: https://github.com/OpenCBM/OpenCBM/blob/447dbcb0e99ebf57b313b53783abd2e3123d294f/opencbm/LINUX/config.make#L113-L114

I installed `pkg-config` via homebrew and made sure it was in my path, but it didn't seem to help. Even after cleaning all ports and running the install command again, I got the failure output from my gist above. The key error was on this line: https://gist.github.com/lafiosca/385c887e898c0e8da680ed0218a30df0#file-main-log-L404

Which came from this section: https://github.com/OpenCBM/OpenCBM/blob/447dbcb0e99ebf57b313b53783abd2e3123d294f/opencbm/libmisc/usbcommon.h#L30-L37

This seems to indicate that the failure to define either of those values on those `pkg-config` steps could be the root cause.

I am unclear on why the Makefile is unable to find/run pkg-config even after it's been installed to my standard path:
```
$ which pkg-config
/usr/local/bin/pkg-config
```

(I have not worked with Makefiles in a long time, and never before in an OSX context.)

lafiosca, 2019-10-05:
I see that @bodgit suggested the switch to `pkg-config` in a previously resolved issue: https://github.com/OpenCBM/OpenCBM/issues/3#issuecomment-460053897

Any ideas that could help us?

I manually cleaned the work directory and ran `sudo port -v install opencbm` to get verbose debug output, available here: https://gist.github.com/lafiosca/79f15d102ee169e248b78d0ca68a4ede

I'm not familiar with MacPorts, but I find it odd that when I run the [exact command listed in that log](https://gist.github.com/lafiosca/79f15d102ee169e248b78d0ca68a4ede#file-gistfile1-txt-L4), I don't get the exact same output: https://gist.github.com/lafiosca/2a40834d88acde2338458e1d8f0d9054

It's similar but doesn't include any of the verbose errors about missing `pkg-config`. The ultimate result appears to be the same unfortunately. I'm wondering if port is running a different version of make or calling it differently behind the scenes or something, but I also think this might be a red herring.

lafiosca, 2019-10-05:
Uninstalling the `brew` version of `pkg-config` and manually installing the `port` version (`sudo port install pkgconfig`) got me a bit further. Now my errors are looking more similar to the original post here: https://gist.github.com/lafiosca/8bcb22d18ce09193cb845c13028b9f7e

lafiosca, 2019-10-05:
I create PR #18 to add `port:pkgconfig` to the build dependencies. Note that this does NOT resolve the original issue described above, but it will help people avoid the problem I ran into along the way.

bodgit, 2019-10-06:
`port:pkgconfig` is a dependency of libusb-compat so it should be pulled in, although I agree it should be also added as a first class dependency for OpenCBM if it calls it directly.

The reason the Homebrew version didn't likely work is that it is probably looking for the `.pc` files under `/usr/local` somewhere whereas the MacPort version will be looking under `/opt/local`. It will be set at compile-time although you could coax either version to look in both places.

I will try with the latest version of the source and see if it breaks for me too, I might be able to offer a fix.

lafiosca, 2019-10-06:
Strange that pkgconfig was not even installed indirectly then! Thanks for taking a look, @bodgit !

spiro-trikaliotis, 2019-10-09:
As I wrote in #18, I do not know anything about MacOS build environment, so it is hard for me to debug such an issue.

One thing that comes to mind:
1. Install libusb0 (or libusb1)
2. Then, "hard-code" HAVE_LIBUSB0 (or HAVE_LIBUSB1) to "1" in OpenCBM/opencbm/LINUX/config.make.

Does it solve the build problem?

Can you re-do it for the "other" libusb0 or libusb1?

If it helps, we have a problem in the detection only. If not, please give me a newer output, so I can have a look again.

bodgit, 2019-10-10:
I've raised a PR #19 which I think fixes the issue, everything now compiles at least. I'll have to dig out a 1541/1571 to test it properly.

lafiosca, 2019-10-10:
I believe #19 has fixed this. I was able to install opencbm and detect the drive. Then I installed nibtools and was able to read one disk (semi-?)successfully.

(Unfortunately all subsequent nibreads have been failing for me with `NOSYNC!` errors, but I am assuming that is not the fault of the opencbm library. I imagine I'll need to start by physically opening up the drive and doing some maintenance. Why is this stuff so frustrating?)

bodgit, 2019-10-10:
I presume you have a parallel interface for your 1541? I thought that was a requirement of nibtools, so I haven't tried them yet with any of my drives. I do have a parallel interface kit but because I have a 1541-II the VIA chip isn't socketed so I need to dismantle the drive and desolder the chip from the controller board, it's on the todo pile.

closed 2019-10-10 by spiro-trikaliotis

spiro-trikaliotis, 2019-10-10:
@lafiosca, I would assume the R/W head needs some cleaning. You can read how to do it on the net.
It can be that your disk looses its surface, and the surface gets on the R/W head. If this is the case, and you try to read this or other disk, chances are very high that you will ruin all the other disks that you put into the drive! So, first of all, clean it!

@bodgit, for the 1541, one needs a parallel interface. For the 1570 and 1571, it also works without parallel interface.

lafiosca, 2019-10-10:
Yes @bodgit, I modified my 1541 a couple years ago to add the parallel port. I also built the XAP1541 cable, and I was able to nibble some disks on an ancient Windows laptop, but that stopped working unexpectedly. So I purchased a ZoomFloppy to try to use my more modern computers with it instead of scrounging up parallel ports :)

@spiro-trikaliotis This could explain a lot about how I also recently had TWO other drives go bad back-to-back. I will try cleaning all of them and checking alignment when I have some more time to work on this.

Big thanks to both of you for your help.

alien2108, 2019-10-20:
Confirming successful compile for me to now!

PR #19 (2019-10-10): Fix compilation on OS X
There was still a condition for Darwin to use the old `libusb-config`, instead let it fall through to the catch-all condition which relies on `pkg-config` instead.

Also update the `Portfile` with the correct dependencies.

Fixes #16 

(this also probably supersedes #18 too)