OpenCBM/OpenCBM #17: Building with `-j` causes linking errors

PotcFdk, 2019-10-05:
This works: 
`make -j1 -f LINUX/Makefile opencbm plugin-xum1541 plugin-xu1541`
This doesn't: 
`make -j8 -f LINUX/Makefile opencbm plugin-xum1541 plugin-xu1541`

It fails with:
```
(...)
make[1]: Leaving directory '/home/user/test/opencbm/OpenCBM/opencbm/libtrans'
gcc -g -O2 -Wall -I../include -I../include/LINUX -DPREFIX=\"/usr/local\" -DOPENCBM_CONFIG_FILE=\"/etc/opencbm.conf\"  -o ../libd64copy/s2.o -c ../libd64copy/s2.c
gcc cbmctrl.o pport.o -o cbmctrl -L../lib -L../arch/linux -L../libmisc -lopencbm -larch -lmisc
/usr/bin/ld: ../libmisc/libmisc.a(getpluginaddress.lo): in function `plugin_load':
getpluginaddress.c:(.text+0x10): undefined reference to `dlopen'
/usr/bin/ld: getpluginaddress.c:(.text+0x18): undefined reference to `dlerror'
/usr/bin/ld: ../libmisc/libmisc.a(getpluginaddress.lo): in function `plugin_get_address':
getpluginaddress.c:(.text+0x51): undefined reference to `dlsym'
/usr/bin/ld: ../libmisc/libmisc.a(getpluginaddress.lo): in function `plugin_unload':
getpluginaddress.c:(.text+0x61): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
make[1]: *** [../LINUX/prgrules.make:61: cbmctrl] Error 1
```

Running the same command again seems to finish the compilation.

spiro-trikaliotis, 2019-10-09:
You are right.
From a first look, it seems I would need to generally rework the build system in order to fix this issue.
Thus, I will leave it open, but give the suggestion to always use -j1.

I am sorry, but I do not see an easy solution at the moment, and I am lacking the time to do a proper one.

hpingel, 2019-10-09:
@PotcFdk: Could you have a look if this solves the issue:
https://github.com/OpenCBM/OpenCBM/pull/15/commits/cf90f8b753b61fcd4780472f8697fc3db1dd0185

PotcFdk, 2019-10-09:
@hpingel It does not.

PotcFdk, 2019-10-09:
Interestingly enough, this makes the error disappear. It's probably ugly, though.
```patch
diff --git a/opencbm/cbmctrl/LINUX/Makefile b/opencbm/cbmctrl/LINUX/Makefile
index 7dc77b23..5f50c7e9 100644
--- a/opencbm/cbmctrl/LINUX/Makefile
+++ b/opencbm/cbmctrl/LINUX/Makefile
@@ -4,6 +4,7 @@ include ${RELATIVEPATH}LINUX/config.make
 PROG    = cbmctrl
 INC     = tdchange.inc
 
+LINK_FLAGS += -ldl
 OBJS = cbmctrl.o pport.o
 
 include ${RELATIVEPATH}LINUX/prgrules.make
```

spiro-trikaliotis, 2019-10-10:
Now I am a little bit surprised. I would assume that the problem is that the Makefile do not clearly specify an order in which to compile.
I might be wrong here, though.

I will have a look again.

spiro-trikaliotis, 2020-07-10:
The build dependencies have been fixed with commit b8af366e.

Does this solve your problem, or does it not solve it?

PotcFdk, 2020-08-18:
Looks good I think.

closed 2020-08-18 by PotcFdk