OpenCBM/OpenCBM #20: Segfault with no device connected

bodgit, 2019-10-10:
I've just discovered that if I run `cbmctrl detect` it will segfault if I don't have a ZoomFloppy connected:
```
$ cbmctrl detect
error: no xum1541 device found
Segmentation fault: 11
```
I've tried to compile it with debugging symbols enabled but I'm having a hard time getting GDB on my Mac to find any debugging symbols. It might be easier to try and reproduce this on a Linux machine.

spiro-trikaliotis, 2019-10-11:
I can reproduce the problem, so, there is no need for debugging symbols from your side.
I will have a look.

spiro-trikaliotis, 2019-10-11:
Fixed with commit 52a0a364c19e4.
Note that there was a similar (but technically different) problem in xum1541cfg, which I fixed, too.

closed 2019-10-11 by spiro-trikaliotis