OpenCBM/OpenCBM #21: Divergent xum1541cfg?

lafiosca, 2019-10-10:
Question: is the xum1541cfg in this OpenCBM repo a distinct fork from https://github.com/silverdr/xum1541cfg ? It seems that other version includes a `devinfo` command that this one does not. If it is distinct, can/should improvements from that other repo be merged here? Thank you.

spiro-trikaliotis, 2019-10-10:
Good question. I did not know about Patryk's version, or at least, I do not remember.
Unfortunately, the formatting is quite dissimilar, so it is hard to tell what was changed on his version.
I will contact him and ask him if he can tell anything about the history.

spiro-trikaliotis, 2019-10-10:
Ok, the only relevant patch is 458f580da2103e. Everything else is just formatting and infrastructure. It should be rather easy to integrate that.

silverdr, 2019-10-10:
Yes - I basically did it before we moved to github and there should be only one part that is relevant for the new functionality. The rest is formatting change, which I did in order to be able to analyse the code faster. Unless you can easily pick it yourself I may try to find some time to build a PR without the formatting changes. Otherwise feel free to use it yourself. If this makes it to the main repository, I'll remove mine.

spiro-trikaliotis, 2019-10-10:
The patch is straightforward and can be integrated rather easily. Give me half an hour (not today!), and I will have it done.

silverdr, 2019-10-10:
FWIW this one: https://github.com/silverdr/xum1541cfg/commit/5556a2c79facc2cbfa4a35a732f102bf3b56f572
might hold some (minor) value other than formatting stuff.

spiro-trikaliotis, 2019-10-11:
Integrated 'devinfo' in commit 544ba45e5dc8a47.

spiro-trikaliotis, 2019-10-11:
> FWIW this one: [silverdr/xum1541cfg@5556a2c](https://github.com/silverdr/xum1541cfg/commit/5556a2c79facc2cbfa4a35a732f102bf3b56f572)
> might hold some (minor) value other than formatting stuff.

What is this other value? I only see an 'if' that has its logic reverses, and 'then' and 'else' changed.
Do I miss something else?

silverdr, 2019-10-11:
It's been some time but checking it out yesterday I noticed that there was something else than formatting. Looking at it now it seems that the thing was about passing through the incoming value of ret rather than hardcoded numbers. I think I changed it because I try not to clobber what I basically only pass through.

closed 2019-10-11 by spiro-trikaliotis

lafiosca, 2019-10-11:
Thank you both!