OpenCBM/OpenCBM #22: XA1541 kernel module does not compile with current kernels

spiro-trikaliotis, 2019-10-11:
Compiling on Debian Buster, kernel 4.19.0-6-amd64 (Linux XXX 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux), the compilation fails:
`make[2]: Entering directory '/usr/src/linux-headers-4.19.0-6-amd64'
  CC [M]  /home/spiro/Work/cbm/opencbm.git/opencbm/sys/linux/cbm_module.o
/home/spiro/Work/cbm/opencbm.git/opencbm/sys/linux/cbm_module.c: In function ‘wait_for_listener’:
/home/spiro/Work/cbm/opencbm.git/opencbm/sys/linux/cbm_module.c:364:30: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
     while (cbm_irq_count && !signal_pending(current))
                              ^~~~~~~~~~~~~~
                              timer_pending
cc1: some warnings being treated as errors`

penfold42, 2019-10-11:
This might be the same

https://forum.manjaro.org/t/solved-error-with-rtl8812au/24066

spiro-trikaliotis, 2019-10-11:
Indeed, this might be it.
I will test it tomorrow.

commit 32a19cb1 (2019-10-12): Fix compiler error on kernel module compilation

Before, we tried to determine with HAVE_LINUX_SCHED_SIGNAL_H if we need
to include <linux/sched/signal.h>. Unfortunately, the mechanism was not
robust.

So, reworked it and hardcoded it depending on the kernel version (>=
4.11.0). This fixes #22.
closed 2019-10-12 by spiro-trikaliotis