OpenCBM/OpenCBM #24: Installing xa1541 or xm1541 with install.cmd fails

spiro-trikaliotis, 2020-03-13:
Trying to install the xa1541 plugin (which includes the non-USB xa1541 and xm1541 cables), the installation fails. In the output, you can find the following file:

Copying '.\cbm4wdm.sys' to 'C:\WINDOWS\system32\DRIVERS\cbm4wdm.sys' FAILED!

(Note that the paths can differ!)

This is an issue with the install.cmd script, as it forgets to copy the cbm4wdm.sys before running instcbm.exe

This issue has been fixed with commit f2bab60ec.

To fix this issue with the 0.4.99.99 binary package, replaced the install.cmd file with the version you get from here:
https://github.com/OpenCBM/OpenCBM/blob/master/opencbm/WINDOWS/install.cmd

The problem has been independantly reported by Charles Campbell and Ashley Powell.

khaverblad, 2020-04-04:
Hi Spiro, while this fixes the issue with cbm4wdm.sys when trying to install xa1541 it fails next on cbm4nt.sys which is missing as well. So the install.cmd needs to be updated with this as well. Copied the mentioned files into the OpenCBM folder which makes the installation of the xa1541 to work.

spiro-trikaliotis, 2020-04-05:
@khaverblad, I am surprised. The patch copies all *.sys files into the destination. How can it miss the cbm4nt.sys file?

khaverblad, 2020-04-05:
While I'm executing everything as local admin and my best guess is that it's Win 7 is the issue here or another or AV (which I checked and nothing been blocked). But, let me check on another system and see how it works out. So I'll keep you posted.

spiro-trikaliotis, 2020-04-05:
You are using the latest install.cmd? I did not update the ZIP yet, so you HAVE to replace the install.cmd with the latest from git.
I just tested it here on Windows 10, it worked as it should. That's why I am surprised.

(But: Yes, I should remove the dependancy on cbm4nt.sys on a Post-NT4 system, because that file is only needed for NT4. For never systems like 2000, XP and so on, it is cbm4wdm.sys)

khaverblad, 2020-04-05:
Yes, pulled the latest install.cmd from git. But, I'll give a spin again. 

Sounds like that there is no need to make too much fuzz about but I'll test. If the mentioned files can be removed that would be for the best since I can't see that there that many if any who runs opencbm on NT4. I'm running it on an old Thinkpad from 2004 which I replaced the mechanical disk with a PATA interface + SSD drive and Win 7 had proper driver support for more or less all hardware except BT and Intel 855GME.

Curious what dongle/cable are you using to get it to work on Win 10 (x64?)?

spiro-trikaliotis, 2020-06-14:
Please check the test version at https://spiro.trikaliotis.net/download/opencbm-0.4.99.101/opencbm-0.4.99.101.zip
It should fix your problem.

spiro-trikaliotis, 2020-06-24:
I am closing this ticket, as I believe it has been fixed.
If you find another problem, feel free to reopen, or file another ticket.

closed 2020-06-24 by spiro-trikaliotis