OpenCBM/OpenCBM #26: Question (or brainstorming)

Zibri, 2020-05-04:
Is there something like this:

1 iec to connect to a real c64
1 usb to connect to a pc.

purpose:
let the pc to emulate a drive or printer so the C64 things there's is a 1541 connectd.. in this way saving from the C64 will just create G64 or similar disks and it would even be possible to share directories from a pc.

I was thinking that *maybe* an ftdi232r is all that is needed...

any thoughts?

spiro-trikaliotis, 2020-05-04:
This has been thought out more than once.
The current HW does not allow for it for two reasons:
1. The ATN line is not an input line currently, but this is needed.

2. For being a device (not a controller) on the IEC bus, one has to react in a timely manner to an assertion of the line DATA by asserting CLOCK.  The timeout is no longer than 200us. While this could be accomplished somehow, it would be much more easy if DATA could generate interrupts, so we could be sure that the timeout will work.

3. It would be slow, because the AVR does not have enough memory to hold an image, so it would have to communicate with the PC over and over, OR we would need an additional memory.

All of this could be worked on, but it would be a different device. Furthermore, there already exist solutions to this (like the Pi1541).

So, unless someone has interest in this and starts it on his own, I doubt someone will work on something like this.

closed 2020-05-04 by spiro-trikaliotis

Zibri, 2020-05-04:
no.. probably you misunderstood...
pi1541 is a standalone device like the ultimate cartridge for example.

The idea is to use just a signal converter.. judging from the datasheet of ftd232r it should be capable of doing this 

<tr>
	<td height="23" colspan="3">
	  <table width="100%" border="0" cellspacing="0" cellpadding="0">
	    <tr> 
		  <td id="title0"  bgcolor="#6E8BB1"> <b>
          <font face="Arial" color="#FFFFFF"><a name="KeyHardwareFeatures"></a>&nbsp; 
          Key Hardware Features</font></b></td>
		</tr>
	  </table>	
	</td>
  </tr>
   
  <tr> 
	<td height="471" align="left" valign="top" style="font-family: Arial; font-size: 10pt; text-align: left; text-indent: 0; word-spacing: 0; line-height: 150%; margin: 0" colspan="3">
		<dir style="word-spacing: 0; line-height: 150%; margin-top: 0; margin-bottom: 0">
          <font SIZE="2">
          <li>Single chip USB to asynchronous serial data transfer interface.</li>
          <li>Entire USB protocol handled on the chip - No USB-specific firmware 
          programming required.</li>
          <li>UART interface support for 7 or 8 data bits, 1 or 2 stop bits and 
          odd / even / mark / space / no parity.</li>
          <li>Fully assisted hardware or X-On / X-Off software handshaking.</li>
          <li>Data transfer rates from 300 baud to 3 Megabaud (RS422 / RS485 and 
          at TTL levels) and 300 baud to 1 Megabaud (RS232).</li>
          <li>In-built support for event characters and line break condition.</li>
          <li>New USB FTDIChip-ID™ feature.</li>
          <li>New configurable CBUS I/O pins.</li>
          <li>Auto transmit buffer control for RS485 applications.</li>
          <li>Transmit and receive LED drive signals.</li>
          <li>New 48MHz, 24MHz,12MHz, and 6MHz clock output signal Options for 
          driving external MCU or FPGA.</li>
          <li>FIFO receive and transmit buffers for high data throughput.</li>
          <li>256 Byte receive buffer and 128 Byte transmit buffer utilising 
          buffer smoothing technology to allow for high data throughput.</li>
          <li>Adjustable receive buffer timeout.</li>
          <li>Synchronous and asynchronous bit bang mode interface options with 
          RD# and WR# strobes.</li>
          <li>New CBUS bit bang mode option.</li>
          <li>Integrated 1024 bit internal EEPROM for I/O configuration and 
          storing USB VID, PID, serial number and product description strings.</li>
          <li>Device supplied preprogrammed with unique USB serial number.</li>
          <li>Support for USB suspend / resume.</li>
          <li>Support for bus powered, self powered, and high-power bus powered 
          USB configurations.</li>
          <li>Integrated 3.3V level converter for USB I/O .</li>
          <li>Integrated level converter on UART and CBUS for interfacing to 5V 
          - 1.8V Logic.</li>
          <li>True 5V / 3.3V / 2.8V / 1.8V CMOS drive output and TTL input.</li>
          <li>High I/O pin output drive option.</li>
          <li>Integrated USB resistors.</li>
          <li>Integrated power-on-reset circuit.</li>
          <li>Fully integrated clock - no external crystal, oscillator, or 
          resonator required.</li>
          <li>Fully integrated AVCC supply filtering - No separate AVCC pin and 
          no external R-C filter required.</li>
          <li>UART signal inversion option.</li>
          <li>USB bulk transfer mode.</li>
          <li>3.3V to 5.25V Single Supply Operation.</li>
          <li>Low operating and USB suspend current.</li>
          <li>Low USB bandwidth consumption.</li>
          <li>UHCI / OHCI / EHCI host controller compatible.</li>
          <li>USB 2.0 Full Speed compatible.</li>
          <li>-40°C to 85°C extended operating temperature range.</li>
          <li>Available in compact Pb-free 28 Pin SSOP and QFN-32 packages (both 
          RoHS compliant).
        </font>
        <p align="left" style="margin-top: 0; margin-bottom: 0">&nbsp;<p align="left" style="margin-top: 0; margin-bottom: 0">&nbsp;</td>
  </tr>

Zibri, 2020-05-04:
the features useful for iec to usb are:

- UART signal inversion option.
- 5.25V Supply Operation.
- True 5V / 3.3V / 2.8V / 1.8V CMOS drive output and TTL input.
- 256 Byte receive buffer and 128 Byte transmit buffer utilising buffer smoothing technology to allow for high data throughput.
- Single chip USB to asynchronous serial data transfer interface.
- Entire USB protocol handled on the chip - No USB-specific firmware programming required.
- UART interface support for 7 or 8 data bits, 1 or 2 stop bits and odd / even / mark / space / no parity.
- Fully assisted hardware or X-On / X-Off software handshaking.
- Data transfer rates from 300 baud to 3 Megabaud (RS422 / RS485 and at TTL levels) and 300 baud to 1 Megabaud (RS232).
- In-built support for event characters and line break condition.

silverdr, 2020-05-05:
@Zibri - so, how does this invalidate what spiro wrote?

Zibri, 2020-05-06:
> @Zibri - so, how does this invalidate what spiro wrote?

"Furthermore, there already exist solutions to this"  pi1541 has nothing to do with this,
the most similar I have found uses an AVR cpu. 
If you read my post you will understand what I was proposing is a different type of interface where the "cable" is more a signal converter.
About the timings, both ftd232r and CH341A are fully capable of keeping up and any pc or even an old cell phone are fast enough for the interface to work.

penfold42, 2020-05-06:
I played with this a few years ago https://github.com/Larswad/uno2iec

Please see spiro’s second point.
200uSec timing requires a micro controller at the IEC end - USB to a PC would struggle to meet that hard timing constraint.

silverdr, 2020-05-06:
I read your post. And I read Spiro's. And I still don't know how your response relates to it. Right, with FTDI's bitbanging you could probably handle IEC lines, and "all of this could be worked on". One way or another. You need to take care of quite a few things though. The initial timing is one of them. And it's not only about the pc being fast enough. It has to be consistent in its responses, regardless of what and under what load it is doing at the moment when the request comes. BTW - there is a working "to USB" solution, which I use all the time - it is "ideserv" from IDE64. It is not IEC (some IEC restrictions fall out) to USB but it does what you say, and blazingly fast too. And please note that it is not to discourage you from working on such project. Quite the opposite - please do. I for one would welcome another connection possibility.