OpenCBM/OpenCBM #27: Crashes on Ubuntu 20.04

ops, 2020-05-17:
I just updated my system to Ubuntu 20.04. Now I get crashes with the latest OpenCBM from github

    ops@s761:~/OpenCBM$ git log -1
    commit 52d435cfbdc33569d6d4bba1271ce9ed757edf05 (HEAD -> master, origin/master, origin/HEAD)
    Author: Spiro Trikaliotis <spiro.trikaliotis@gmx.de>
    Date:   Sat May 16 23:32:14 2020 +0200

Works OK

    ops@s761:~/OpenCBM$ cbmctrl status 9
    00, ok,00,00

Works OK

    ops@s761:~/OpenCBM$ cbmctrl dir 9
    0 ."test/demo 12/84 " 84 2a
    11   "how to use"       prg   
    8    "how part two"     prg  
    8    "header change"    prg  
    409 blocks free.             
    00, ok,00,00

Crash

    ops@s761:~/OpenCBM$ cbmctrl detect
    *** stack smashing detected ***: terminated
    Aborted (core dumped)

Crash

    ops@s761:~/OpenCBM$ d64copy 9 image.d64
    *** stack smashing detected ***: terminated
    Aborted (core dumped)

spiro-trikaliotis, 2020-05-17:
Thank you for your report.
Can you execute the following (I hope you have the debugger gdb installed):
- gdb d64copy
Than, on the prompt, give:
- set args detect
and then start it with
- run

When it crashes, use the command
- bt
and give me the output

This gives me the backtrace, that is, the state where cbmctrl was when crashing, so I know where to search for a problem.

ops, 2020-05-17:
ops@s761:~/OpenCBM$ gdb opencbm/cbmctrl/cbmctrl
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from opencbm/cbmctrl/cbmctrl...
(gdb) set args detect
(gdb) run
Starting program: /home/ops/OpenCBM/opencbm/cbmctrl/cbmctrl detect
*** stack smashing detected ***: terminated

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7ddd859 in __GI_abort () at abort.c:79
#2  0x00007ffff7e483ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7f7207c "*** %s ***: terminated\n")
    at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff7eea9ba in __GI___fortify_fail (msg=msg@entry=0x7ffff7f72064 "stack smashing detected") at fortify_fail.c:26
#4  0x00007ffff7eea986 in __stack_chk_fail () at stack_chk_fail.c:24
#5  0x00007ffff7fc76ad in opencbm_plugin_pp_read (f=3) at LINUX/iec.c:157
#6  0x00007ffff7fae4a5 in cbm_pp_read (HandleDevice=3) at cbm.c:1251
#7  0x00007ffff7faf7f2 in cbm_identify_xp1541 (HandleDevice=3, DeviceAddress=9 '\t', CbmDeviceType=0x7fffffffdd90, 
    CableType=0x7fffffffdd94) at detectxp1541.c:296
#8  0x000055555555a129 in do_detect (fd=3, options=0x7fffffffde20) at cbmctrl.c:1328
#9  0x000055555555aba0 in main (argc=2, argv=0x7fffffffdf68) at cbmctrl.c:1980

ops, 2020-05-17:
ops@s761:~/OpenCBM$ gdb opencbm/d64copy/d64copy 
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from opencbm/d64copy/d64copy...
(gdb) set args 9 image.d64
(gdb) run
Starting program: /home/ops/OpenCBM/opencbm/d64copy/d64copy 9 image.d64
*** stack smashing detected ***: terminated

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7ddd859 in __GI_abort () at abort.c:79
#2  0x00007ffff7e483ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7f7207c "*** %s ***: terminated\n")
    at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff7eea9ba in __GI___fortify_fail (msg=msg@entry=0x7ffff7f72064 "stack smashing detected") at fortify_fail.c:26
#4  0x00007ffff7eea986 in __stack_chk_fail () at stack_chk_fail.c:24
#5  0x00007ffff7fc76ad in opencbm_plugin_pp_read (f=3) at LINUX/iec.c:157
#6  0x00007ffff7fae4a5 in cbm_pp_read (HandleDevice=3) at cbm.c:1251
#7  0x00007ffff7faf7f2 in cbm_identify_xp1541 (HandleDevice=3, DeviceAddress=9 '\t', CbmDeviceType=0x7fffffffdaf8, 
    CableType=0x7fffffffdb3c) at detectxp1541.c:296
#8  0x0000555555559368 in d64copy_check_auto_transfer_mode (cbm_fd=3, auto_transfermode=0, drive=9) at ../libd64copy/d64copy.c:730
#9  0x00005555555576d6 in main (argc=3, argv=0x7fffffffdf68) at main.c:387

ops, 2020-05-17:
There are some compile time warnings too:

detectxp1541.c: In function ‘cbm_identify_xp1541’:
detectxp1541.c:245:13: warning: variable ‘value’ set but not used [-Wunused-but-set-variable]
  245 |         int value;
      |             ^~~~~
ar: creating libopencbm.a
detectxp1541.c: In function ‘cbm_identify_xp1541’:
detectxp1541.c:245:13: warning: variable ‘value’ set but not used [-Wunused-but-set-variable]
  245 |         int value;
      |             ^~~~~
ar: creating libtrans.a
../libcbmcopy/cbmcopy.c: In function ‘cbmcopy_write_file’:
../libcbmcopy/cbmcopy.c:531:9: warning: variable ‘error’ set but not used [-Wunused-but-set-variable]
  531 |     int error;
      |         ^~~~~
../libimgcopy/imgcopy.c: In function ‘imgcopy_set_image_type’:
../libimgcopy/imgcopy.c:361:5: warning: enumeration value ‘cbm_it_unknown’ not handled in switch [-Wswitch]
  361 |     switch(settings->image_type)
      |     ^~~~~~
../libimgcopy/imgcopy.c:361:5: warning: enumeration value ‘D64’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c:361:5: warning: enumeration value ‘D71’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c: In function ‘imgcopy_sector_count’:
../libimgcopy/imgcopy.c:398:5: warning: enumeration value ‘cbm_it_unknown’ not handled in switch [-Wswitch]
  398 |     switch(setting->image_type)
      |     ^~~~~~
../libimgcopy/imgcopy.c:398:5: warning: enumeration value ‘D64’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c:398:5: warning: enumeration value ‘D71’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c: In function ‘imgcopy_check_transfer_mode’:
../libimgcopy/imgcopy.c:499:9: warning: unused variable ‘mode_p’ [-Wunused-variable]
  499 |     int mode_p = imgcopy_get_transfer_mode_index("parallel");
      |         ^~~~~~
../libimgcopy/imgcopy.c: In function ‘ReadBAM’:
../libimgcopy/imgcopy.c:661:5: warning: enumeration value ‘cbm_it_unknown’ not handled in switch [-Wswitch]
  661 |     switch(settings->image_type)
      |     ^~~~~~
../libimgcopy/imgcopy.c:661:5: warning: enumeration value ‘D64’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c:661:5: warning: enumeration value ‘D71’ not handled in switch [-Wswitch]
../libimgcopy/imgcopy.c: In function ‘copy_disk’:
../libimgcopy/imgcopy.c:863:13: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  863 |             if(settings->warp>0)
      |             ^~
../libimgcopy/imgcopy.c:865:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  865 |                 settings->warp = 0;
      |                 ^~~~~~~~
../libimgcopy/imgcopy.c:876:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  876 |                 if(settings->warp>0)
      |                 ^~
../libimgcopy/imgcopy.c:878:21: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  878 |                     settings->warp = 0;
      |                     ^~~~~~~~
../libimgcopy/fs.c: In function ‘open_disk’:
../libimgcopy/fs.c:124:31: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘off_t’ {aka ‘long int’} [-Wformat=]
  124 |             printf("filesize=%d, blockcount=%d, calc1=%d, calc2=%d\n", filesize, block_count, block_count * (BLOCKSIZE), block_count * (BLOCKSIZE + 1));
      |                              ~^                                        ~~~~~~~~
      |                               |                                        |
      |                               int                                      off_t {aka long int}
      |                              %ld
rpm1541.c: In function ‘main’:
rpm1541.c:47:32: warning: variable ‘tsize’ set but not used [-Wunused-but-set-variable]
   47 |     unsigned short int currpm, tsize;
      |                                ^~~~~
main.c: In function ‘read_line_status’:
main.c:310:34: warning: variable ‘pp’ set but not used [-Wunused-but-set-variable]
  310 |     int data, clock, atn, reset, pp;
      |                                  ^~
main.c:310:27: warning: variable ‘reset’ set but not used [-Wunused-but-set-variable]
  310 |     int data, clock, atn, reset, pp;
      |                           ^~~~~
main.c:310:22: warning: variable ‘atn’ set but not used [-Wunused-but-set-variable]
  310 |     int data, clock, atn, reset, pp;
      |                      ^~~
main.c:310:15: warning: variable ‘clock’ set but not used [-Wunused-but-set-variable]
  310 |     int data, clock, atn, reset, pp;
      |               ^~~~~
main.c:310:9: warning: variable ‘data’ set but not used [-Wunused-but-set-variable]
  310 |     int data, clock, atn, reset, pp;
      |         ^~~~
At top level:
main.c:332:1: warning: ‘main_testlines’ defined but not used [-Wunused-function]
  332 | main_testlines(int argc, char **argv)
      | ^~~~~~~~~~~~~~

spiro-trikaliotis, 2020-05-17:
First of all, let me understand: Do you really use the xa1541 plugin (that is, the parallel port)?

ops, 2020-05-17:
> First of all, let me understand: Do you really use the xa1541 plugin (that is, the parallel port)?

Yes

spiro-trikaliotis, 2020-05-17:
I am unsure at the moment. Looking at the place where this problem occurs, I do not find any obvious problem.
I compiled the xa1541 and ran it on my only machine that still has a parallel port, and I did not notice any problem. I also used the options "-fstack-protector -D_FORTIFY_SOURCE=2" that seem to be important for your issue.

However, my setup is different than yours, because that machine is a 32 bit Debian 9.12 system. As this machine only has a 32 bit CPU, I cannot upgrade it. I do not even know if the stack protector mentioned above is as good as yours, and if it is even compiled with it.

I will have to create a special version of the module that runs without real hardware, and test it in a VM. However, I will need some time for this.

ops, 2020-05-17:
Found one issue. See my PR. detect works now.

spiro-trikaliotis, 2020-05-17:
Thank you. This bug is embarrassing. It seems I have to re-check every function to make sure there is no other bug of this type.

ops, 2020-05-17:
You can close this. There seems to be another issue with d64copy hanging. I will open a new issue for it.

closed 2020-05-17 by spiro-trikaliotis