OpenCBM/OpenCBM #29: Compile error

ops, 2020-05-19:
Latest OpenCBM

    ops@laptop ~/projects/OpenCBM $ git log -1
    commit ef2f1f1a250efab16d87103c3dfb29187b662328 (HEAD -> master, upstream/master,    origin/master, origin/HEAD)
    Author: Spiro Trikaliotis <spiro.trikaliotis@gmx.de>
    Date:   Sun May 17 21:39:19 2020 +0200

      Add -Werror, calm all warnings

gcc version 9.3.0 gives me this error:

gcc -g -I..//libcbmcopy -O2 -Wall -Werror -I../include -I../include/LINUX -DPREFIX=\"/usr/local\" -DOPENCBM_CONFIG_FILE=\"/etc/opencbm.conf\" -fstack-protector -D_FORTIFY_SOURCE=2  -o main.o -c main.c
In file included from /usr/include/string.h:495,
                 from main.c:18:
In function ‘strncpy’,
    inlined from ‘main’ at main.c:498:33:
/usr/include/bits/string_fortified.h:106:10: error: ‘__builtin_strncpy’ output may be truncated copying 16 bytes from a string of length 16 [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../LINUX/prgrules.make:53: main.o] Error 1
make[1]: Leaving directory '/home/ops/projects/OpenCBM/opencbm/cbmcopy'
make: *** [LINUX/Makefile:69: BUILDSYSTEM.opencbm/cbmcopy.all] Error 2

ops, 2020-05-19:
I'm not sure if  `-Werror` is a good option for software that will be compiled on multiple platforms.

spiro-trikaliotis, 2020-05-19:
You might be right, the -Werror might not be very helpful.

I looked at the warned code: It is safe. The buf has length 48, and the line
                strncpy(buf, fname, 16);

is followed by

                buf[16] = '\0';

So, the warning is not right, as the case is treated right.

ops, 2020-05-19:
OK, compiles now.

closed 2020-05-19 by ops

silverdr, 2020-05-19:
> So, the warning is not right, as the case is treated right.

AFAIU when you terminate the string at `buf[16]` and then you copy "at most" 16 bytes that means you won't copy the terminated `NUL`, which makes the warning valid, right?