OpenCBM/OpenCBM #31: d64copy hanging

ops, 2020-05-20:
I can't get d64copy to work with the latest OpenCBM. I'm using XM1541 cable on 64 bit Linux Machine.

`d64copy 9 image.d64` just hangs and the following messages can be found from system log:

    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: I/O error
    May 20 17:59:30 laptop kernel: cbm_write: I/O error
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present
    May 20 17:59:30 laptop kernel: cbm_write: device not present

Other commands like `cbmctrl detect`, `cbmctrl status 9` and `cbmctrl dir 9` work fine.

spiro-trikaliotis, 2020-05-27:
That's weird. Do you get any output from d64copy in this case?
Can you try the options
-ts1 and -ts2 (one after the other, they change the transfer protocol to serial1 = s1, serial2 = s2) for d64copy? Does it change anything? You can also add -no-warp option, does this change anything?

ops, 2020-05-27:
I have now tested different versions. Something bad happened between these two commits:

    OK	Feb 4, 2019	3d012ae78f5172658b3fa97322d14153b13044fd

    FAIL	Feb 6, 2019	29d42d3627a4bc8d2c347c3b12c0fa00c1632764

spiro-trikaliotis, 2020-05-28:
Thank you for your investigation, this makes more sense now. There was some changes in the kernel infrastructure to allow for FreeBSD in between. It seems some behaviour changed for Linux.

I'll have to dig out my Linux parallel port environment and do some tests myself.

closed 2020-05-28 by spiro-trikaliotis

spiro-trikaliotis, 2020-05-28:
For me, the fix in 677b7e1 solves the issue. Please test and report any success or failure.

ops, 2020-05-29:
Thanks, it works now!

Just a note about the change: why don't you just write `return 0;` instead of `return rv;`?

spiro-trikaliotis, 2020-05-29:
I re-used the value rv. Now, if I set rv and then return another value, one could think this is an error. Thus, I made it explicit that the rv is 0.
I hope a decent compiler will optimize it out, creating the same code in both cases.

commit 677b7e1a (2020-05-28): Fix d64copy and cbmcopy hang on Linux xa1541/xm1541

CBMCTRL_IEC_POLL did not return its value, but returned the value as
return status. Thus, the value never changed, resulting in hangs in
programs that use cbm_iec_poll() or cbm_iec_get().

This fixes #31.