OpenCBM/OpenCBM #32: Merge nibtools with OpenCBM repository (suggestion)

r1me, 2020-05-23:
A short suggestion to merge nibtools source code and include its binaries with builds of OpenCBM.
One depends on each other and with all the issues at the moment on Windows, USB2/USB3 missing device with libusb1 etc. it gets harder to understand where problems lies on.

spiro-trikaliotis, 2020-05-23:
Technically, nibtools can be compiled alongside with OpenCBM. Just put the nibtools repository in opencbm/nibtools, and both Linux and Windows will happily compile them in when compiling OpenCBM.

But: Pete asked not to include them in the Windows binaries, as he wants to proceed faster than OpenCBM does. That's why it is not included automatically.

spiro-trikaliotis, 2020-06-27:
For Linux, the packages at https://software.opensuse.org//download.html?project=home%3Astrik&package=opencbm now also include a package "nibtools" (for Debian based systems), or the nibtools are included in opencbm (for RPM based systems).

For Windows, I will look for a solution.