OpenCBM/OpenCBM #33: d64copy -t original read errors

ops, 2020-05-29:
`d64copy -t original` gives me many read errors. `d64copy -t  serial1` and `d64copy -t serial2` work ok with the same disk.

spiro-trikaliotis, 2020-05-29:
Can you send me the three files you get?
I assume that -t original is correct, and the other do not recognize the read error.

ops, 2020-05-29:
I did the following test:

- write image file to disk: `d64copy cbm1541_demo_disk.d64 9`
- read it back: `d64copy 9 image.d64`
- compare: `diff -s cbm1541_demo_disk.d64 image.d64 `
    Files cbm1541_demo_disk.d64 and image.d64 are identical
- read it back again `d64copy -t original 9 image-orig.d64`
- `diff -s cbm1541_demo_disk.d64 image-orig.d64 `
    Binary files cbm1541_demo_disk.d64 and image-orig.d64 differ

spiro-trikaliotis, 2020-05-31:
Again, can you give me the disk images you wrote and you got via PM?

ops, 2020-06-01:
Email sent.

spiro-trikaliotis, 2020-06-01:
Thank you. I got it, and I can confirm what you are seeing. This is weird, because I always assumed -t original to be the most reliable method.

Interestingly, there are some blocks that differ but that are not marked as having problems. This is even more weird.

I will think about how to proceed from here. Currently, I am a little bit confused.

Does this ALWAYS happen, with all disks?

Can you do me a favor? Please, freshly format a disk and send me the output of the transfer with all three variants?

ops, 2020-06-01:
Do you want disk to be formatted with cbmformat or std drive command (n0:diskname,id)?

spiro-trikaliotis, 2020-06-01:
Both are o.k. I only want to have a "known good" image on the disk.

ops, 2020-06-01:
OK. Another email sent.

spiro-trikaliotis, 2020-06-01:
Your files do not make much sense to me.

I assume you are still using Linux 64 bit with an XM1541 cable, right?

Is the parallel port on your mainboard? Is it on a PCI card?

I assume a general problem with IEC communication. I have had this on a laptop where the parallel port was available only on the docking station. There, the serial communication ("original") could not made be reliable, although it worked in 99% of the cases. Thus, serial1, serial2 and parallel worked, but original did not reliably.

Can we test your communication with the following oneliner:

$ for ((A=1; A<10; A++)); do time cbmctrl download 8 0xc000 0x4000 rom.$A; done

Each call to cbmctrl should take approx. 80 seconds. This line will download the ROM contents of the floppy over and over again. Are all the resulting files identical? (md5sum rom.* should give a good indication)

ops, 2020-06-02:
> Your files do not make much sense to me.
> 
> I assume you are still using Linux 64 bit with an XM1541 cable, right?
> 

Yes.

> Is the parallel port on your mainboard? Is it on a PCI card?
>
> I assume a general problem with IEC communication. I have had this on a laptop where the parallel port was available only on the docking [...]

I guess I'm in that 1% with my HW. Laptop with a docking station.

> Can we test your communication with the following oneliner:
>
> $ for ((A=1; A<10; A++)); do time cbmctrl download 8 0xc000 0x4000 rom.$A; done
>
> Each call to cbmctrl should take approx. 80 seconds. [...]

Can't create files, download just hangs.

I took a quick look at code. `opencbm/lib/upload.c` line 301 doesn't check if `readbytes` is < 0 in case of error.

spiro-trikaliotis, 2020-06-02:
This shows that we found the problem, yes.
I looked into your dump in more detail: I can clearly see that the transfer of some blocks was stopped in between and some value that was there in a previous block remained.

Example:
Track 15/12 if your first D64:
-013240: 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 4f 20 44        *************O D        *************o d
-013250: 49 53 4b 20 42 55 46 46 45 52 20 20 20 20 20 20        ISK BUFFER              isk buffer      
-013260: 20 20 20 20 20 2a 00 82 07 36 01 8f 2a 2a 2a 2a             *...6..****             *...6..****
+013240: 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a 2a        ****************        ****************
+013250: 2a 2a 2a 2a 2a 2a 2a 2a 2a 00 60 0b d2 01 4b b2        *********.`...K.        *********.`.R.k.
+013260: 31 3a 4e 42 28 31 29 b2 c6 28 41 24 28 30 29 29        1:NB(1)..(A$(0))        1:nb(1).F(a$(0))

We see that at 1324D (Track 15/Sector 12), the original.d64 (-) and serial1.d64 (+) differ.
What stumbled me was that I did not know what the data came from? It look somehow valid, even in the original.d64.

But in Track 15/Sector 16, this data is found, too:
013640: 20 2a 00 5e 07 2c 01 8f 2a 20 49 4e 54 4f 20 44          *.^.,..* INTO D         *.~.,..* into d
013650: 49 53 4b 20 42 55 46 46 45 52 20 20 20 20 20 20         ISK BUFFER              isk buffer
013660: 20 20 20 20 20 2a 00 82 07 36 01 8f 2a 2a 2a 2a              *...6..****             *...6..****

Block 15/16 was read before 15/12 due to interleaving. The communication was aborted, d64copy did not recognize it and happily wrote a mixture of old and new data into the d64.

I will have a look and make d64copy (and cbm_download(), btw., which you correctly identified as another problem) more robust.

Thank you for the report!

spiro-trikaliotis, 2020-06-04:
Hello Olli,
I worked on the cbmctrl command. upload and download should be rather robust now when transfer problems occur.
Can you test if the command
$ for ((A=1; A<10; A++)); do time cbmctrl download 8 0xc000 0x4000 rom.$A; done
now creates the rom.X files and if all of them are identical?

If yes, I succeeded at least for these commands. Then, I can try to apply the same to the transfer routines.

spiro-trikaliotis, 2020-06-04:
... and if you are adventurous, you could try the attached patch for d64copy. I am currently testing it.

[d64copy-std-robust.c.diff.gz](https://github.com/OpenCBM/OpenCBM/files/4732801/d64copy-std-robust.c.diff.gz)

ops, 2020-06-05:
> Hello Olli,
> I worked on the cbmctrl command. upload and download should be rather robust now when transfer problems occur.
> Can you test if the command
> $ for ((A=1; A<10; A++)); do time cbmctrl [...]

Yes, seems that you got it right, all files report the same checksum!

ops, 2020-06-05:
> ... and if you are adventurous, you could try the attached patch for d64copy. I am currently testing it.
> 
> [d64copy-std-robust.c.diff.gz](https://github.com/OpenCBM/OpenCBM/files/4732801/d64copy-std-robust.c.diff.gz)

It is friday and I feel brave...

`d64copy -t original` result is now seg. fault :-)

I run it under gdb:

```
Program received signal SIGSEGV, Segmentation fault.
cbm_device_status (HandleDevice=3, DeviceAddress=9 '\t', Buffer=0x7fffffffd680, BufferLength=47) at cbm.c:1562
1562	            bufferToWrite[bytesRead] = '\0';
(gdb) bt
#0  cbm_device_status (HandleDevice=3, DeviceAddress=9 '\t', Buffer=0x7fffffffd680, BufferLength=47) at cbm.c:1562
#1  0x0000000000407c31 in read_block (tr=13 '\r', se=16 '\020', block=0x7fffffffdfd0 "") at ../libd64copy/std.c:33
#2  0x0000000000404657 in copy_disk (fd_cbm=3, settings=0x40f2a0, src=0x40e460 <d64copy_std_transfer>, src_arg=0x9, dst=0x40e340 <d64copy_fs_transfer>, 
    dst_arg=0x7fffffffe9eb, cbm_drive=9 '\t') at ../libd64copy/d64copy.c:514
#3  0x0000000000404f06 in d64copy_read_image (cbm_fd=3, settings=0x40f2a0, src_drive=9, dst_image=0x7fffffffe9eb "xx.d64", msg_cb=0x4025b1 <my_message_cb>, 
    stat_cb=0x4026aa <my_status_cb>) at ../libd64copy/d64copy.c:805
#4  0x000000000040324b in main (argc=5, argv=0x7fffffffe6a8) at main.c:397
```

ops, 2020-06-05:
Another unrelated issue. During debug compilation I get lot of messages like below:

```
gcc -ggdb -O0 -Wall -I../include -I../include/LINUX -DPREFIX=\"/usr/local\" -DOPENCBM_CONFIG_FILE=\"/etc/opencbm.conf\" -fstack-protector -D_FORTIFY_SOURCE=2  -I ./ -I ../libmisc/ -D_REENTRANT -fPIC -c -o LINUX/configuration_name.lo LINUX/configuration_name.c
In file included from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from LINUX/configuration_name.c:23:
/usr/include/features.h:397:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
  397 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
      |    ^~~~~~~
```

spiro-trikaliotis, 2020-06-05:
The warnings should be gone with commit 63062968.

The crash should be gone with commit 7b63e958.

This is a place that is not guarded against transmission errors, and I am not sure if I will add this. At least, it does not crash anymore.

The d64copy patch is not enough. With my artificial test, my d64 is still different from the d64 os serial1 or serial2. At least, the difference is much less than before.

ops, 2020-06-05:
OK. I did a few tests and could not reproduce crash anymore.

ops, 2020-06-05:
Another unrelated issue. Sometimes `cbmctrl dir` hangs. I think that happens when `cbm_raw_read` in line 996 returns -1 (in file cbmctrl.c). Could you check?

spiro-trikaliotis, 2020-06-07:
I deciced not to work anymore on problems with the IEC protocol.
This is really a can of worms.
upload/download are essential for the other operations, so it makes sense to make them more robust. But if I want to make everything else robust, thus is just too much work. And even then, I will not always be able to do something reasonable, because the IEC protocol is not designed to be fault-tolerant. There are errors that do not make much sense to fix.