OpenCBM/OpenCBM #35: Assertion failed error on OSX 10.15

dcutugno, 2020-06-16:
Every time a issue a command with cbmctrl after it perform the operation correctly on xum1541, this error is displayed every time:

Assertion failed: (config >= 0 && config <= UINT8_MAX), function darwin_set_configuration, file os/darwin_usb.c, line 1260.

spiro-trikaliotis, 2020-06-17:
The "official" libusb 1.0.23 sources do not have this assertion, although it is a file of it. It seems whoever added it to MacOS added some instrumentation.

Can you give me a link to the relevant file os/darwin_usb.c of the libusb 1.0 package (however it is called on MacOS)? I have a suspicion what might cause it, but I would like to confirm.

dcutugno, 2020-06-17:
Do you want to send my file on the system OS X? I have to search where is it

spiro-trikaliotis, 2020-06-17:
It's ok, we have it. The problematic assert was removed with this patch: https://github.com/libusb/libusb/commit/e9eec3a680cad3b2c9c5213fb7d60148cf6900da

So, you just have to wait, either for the new libusb 1.0 on MacOS, or for our new patch. We wil remove the dependancy on this shortly, as we found another, better solution to the problem at hand.

spiro-trikaliotis, 2020-06-17:
Oh, and it's a problem of libusb on MacOS, as it does not comply with its own documentation!

spiro-trikaliotis, 2020-06-22:
This problem should be gone with v0.4.99.102. Please check out the latest git HEAD and try again.

spiro-trikaliotis, 2020-06-24:
I close this issue. If the problem still persists, feel free to re-open or to file another ticket.

closed 2020-06-24 by spiro-trikaliotis