OpenCBM/OpenCBM #38: xa1541 issues

janagn, 2020-06-25:
Hello,

I hope I am not taking advantage of this list should it be only for bugs while this might sounds more like a support call. So have followed the instructions to compile and install on an old Linux Slackaware 14.1 distro with a parallel port. To cut the long story short once I do:

modprobe -v cbm 
1. I get the 1541 drive light flash
2. And the log is: 
`Jun 25 16:16:35 puppypc10743 user.warn kernel: cbm_init: using active (XA1541) cable (auto), irq 7

Jun 25 16:16:35 puppypc10743 user.warn kernel: cbm: resetting devices

Jun 25 16:16:35 puppypc10743 user.warn kernel: cbm: waiting for free bus...

Jun 25 16:16:36 puppypc10743 user.warn kernel: cbm: bus is free!`

So long so good. If I issue cbtctl rest then I am getting "cbm: quitting because of time out" and this is fat a I have been. I am wondering if I need to compile with -DDIRECT_PORT_ACCESS

**My lsmod is giving me:**
parport_pc             20273  1
parport                24249  2 cbm,parport_pc

**/etc/modules.conf**
alias parport_lowlevel parport_pc

Any idea as what am I missing will be greatly appreciated.

Kind Regards
Yiannis

spiro-trikaliotis, 2020-06-25:
Hello Yannis,
why do you believe you have a problem? So far, it sounds good. The "cbm: quitting because of time out" sometimes happens with the xa1541 kernel module.

Does cbmctrl reset result in any reaction of the floppy drive? If not, do you get any output other than the kernel log?

Did you try other commands like "cbmctrl detect" or "cbmctrl status 8" (if you have a drive with #8 connected, otherwise replace the 8 with the correct device number)? Even if the floppy does not react on the reset, everything else might work.

How is your parallel port connected? Is it on the mainboard? Is it in a ISA, PCI, EISA, PCI Express or similar slot, or via USB?

The -DDIRECT_PORT_ACCESS is only necessary if you do not have any parport driver. i do not recommend that, it is a relict from Linux 2.2 times.

janagn, 2020-06-25:
Hi Spyro,

Thanks for the prompt response. Sothe only meaningful response I get in the log  is the followring when I issues **cbmctrl reset**:
**Jun 25 21:43:10 puppypc10743 user.warn kernel: cbm: resetting devices
Jun 25 21:43:10 puppypc10743 user.warn kernel: cbm: waiting for free bus...
Jun 25 21:43:11 puppypc10743 user.warn kernel: cbm: bus is free!**

The response to **cbmctrl detect** is always:
**Jun 25 21:46:01 puppypc10743 user.warn kernel: cbm_write: I/O error**
and the terminal is hanging (I have to press ctrl-c) while for **cbmctrl status 8** I get on the terminal cbmctrl: **status: No such device** and on the log:
**Jun 25 21:48:29 puppypc10743 user.warn kernel: cbm_write: I/O error**

The laptop is an old early 2000 celeron with a parallel port. I assume the port must be PCI. Does any of these make any sense?

Kind regards
Yiannis

spiro-trikaliotis, 2020-06-25:
If it is PCI, chances are very high that it cannot deliver the IRQ to the PC.
In this case, you are poor of luck, as OpenCBM needs the interrupt. 
If you have onboard parallel port or an ISA card, try that one instead.

janagn, 2020-06-25:
You mean it won't even though it reports irq 7 when I do modprobe cbm?

**Jun 25 22:04:21 puppypc10743 user.warn kernel: cbm_init: using active (XA1541) cable (auto), irq 7
Jun 25 22:04:21 puppypc10743 user.warn kernel: cbm: resetting devices
Jun 25 22:04:21 puppypc10743 user.warn kernel: cbm: waiting for free bus...
Jun 25 22:04:22 puppypc10743 user.warn kernel: cbm: bus is free!**

janagn, 2020-06-25:
I think it runs now. I changed the BIOS settings to EPP and looks like it is responding as expected. If that was the problem I apologize for the time wasted :)

closed 2020-06-25 by janagn

spiro-trikaliotis, 2020-06-26:
If there are BIOS settings, then it is an onboard parallel port. These normally work.
Your solution was correct, trying ECP, EPP and bidirectional is a good idea, one of them works most often.