OpenCBM/OpenCBM #39: operation timed out libusb osx

dcutugno, 2020-06-25:
Hi, after last update that was working but with assertion failed error, now version 0.4.99.102 give me those errors:
iMacports % cbmctrl detect

USB error: LIBUSB_ERROR_OTHER
libusb/xum1541:20/11: Operation timed out

*Note i'm using Pi1541 but was working fine previously

thierer, 2020-06-26:
Could you please capture the output of `cbmctrl detect` (or even better `cbmctrl status`) with debugging enabled and post it here? Thanks!

```
LIBUSB_DEBUG=9 cbmctrl detect
```

dcutugno, 2020-06-27:
LIBUSB_DEBUG=9 cbmctrl detect

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000003] [00000307] libusb: debug [libusb_init] created default context
[ 0.000022] [00000307] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000412] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6b9e29ba
[ 0.000439] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x6b9e29ba
[ 0.002936] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002947] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002950] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.002954] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002957] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002958] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.002960] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002962] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.002964] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0024
[ 0.002966] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0000
[ 0.002968] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.002970] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.002972] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002974] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002995] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003002] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7fd5a0407180 for with session 0x6b9e29ba
[ 0.003006] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7fd5a0705d0a
[ 0.003130] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6b9e2415
[ 0.003146] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x6b9e2415/0x1a100000 against cached device with sessionID/locationID 0x6b9e29ba/0x1d100000
[ 0.003150] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x6b9e2415
[ 0.003432] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003440] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003442] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.003444] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003446] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003448] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.003450] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.003452] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.003454] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0024
[ 0.003455] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0000
[ 0.003457] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.003459] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.003461] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003462] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003480] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003484] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7fd5a0407180 for with session 0x6b9e2415
[ 0.003487] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7fd5a070439a
[ 0.003600] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x7aaca839
[ 0.003614] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x7aaca839/0x1d180000 against cached device with sessionID/locationID 0x6b9e2415/0x1a100000
[ 0.003617] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x7aaca839/0x1d180000 against cached device with sessionID/locationID 0x6b9e29ba/0x1d100000
[ 0.003620] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x7aaca839
[ 0.003887] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003895] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003897] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.003899] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003901] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[... cut ...]
[ 0.156088] [00000307] libusb: debug [libusb_unref_device] destroy device 26.2
[ 0.156091] [00000307] libusb: debug [libusb_unref_device] destroy device 29.2
[ 0.156094] [00000307] libusb: debug [libusb_unref_device] destroy device 26.1
[ 0.156096] [00000307] libusb: debug [libusb_unref_device] destroy device 29.1
[ 0.156098] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 0.156123] [00001c0b] libusb: debug [darwin_event_thread_main] darwin event thread exiting
libusb/xum1541:20/12: Operation timed out

dcutugno, 2020-06-27:
LIBUSB_DEBUG=9 cbmctrl status

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000005] [00000307] libusb: debug [libusb_init] created default context
[ 0.000023] [00000307] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000215] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6b9e29ba
[ 0.000232] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x6b9e29ba
[ 0.002598] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002608] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002619] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.002621] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002623] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002625] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.002627] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002628] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.002630] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0024
[ 0.002632] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0000
[ 0.002634] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.002650] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.002654] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002656] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002677] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002684] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7f80fbd05fe0 for with session 0x6b9e29ba
[ 0.002696] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7f80fbc07bfa
[ 0.002786] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x6b9e2415
[ 0.002800] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x6b9e2415/0x1a100000 against cached device with sessionID/locationID 0x6b9e29ba/0x1d100000
[ 0.002803] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x6b9e2415
[ 0.003063] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003071] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003073] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.003075] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003077] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003079] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.003081] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.003082] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.003084] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0024
[ 0.003086] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0000
[ 0.003088] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.003090] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.003091] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003093] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003111] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003115] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7f80fbd05fe0 for with session 0x6b9e2415
[ 0.003118] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7f80fbc07e8a
[ 0.003193] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x7aaca839
[ 0.003206] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x7aaca839/0x1d180000 against cached device with sessionID/locationID 0x6b9e2415/0x1a100000
[ 0.003210] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x7aaca839/0x1d180000 against cached device with sessionID/locationID 0x6b9e29ba/0x1d100000
[ 0.003212] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x7aaca839
[ 0.003453] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003461] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003464] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.003466] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003468] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[... cut ...]
[ 0.372747] [00000307] libusb: debug [libusb_unref_device] destroy device 26.2
[ 0.372750] [00000307] libusb: debug [libusb_unref_device] destroy device 29.2
[ 0.372753] [00000307] libusb: debug [libusb_unref_device] destroy device 26.1
[ 0.372755] [00000307] libusb: debug [libusb_unref_device] destroy device 29.1
[ 0.372757] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 0.372783] [0000200b] libusb: debug [darwin_event_thread_main] darwin event thread exiting
libusb/xum1541:20/14: Operation timed out

thierer, 2020-06-27:
Thanks, that is helpful (Though I don't understand *why* it's happening).

From what I understand, you're able to compile opencbm on your machine, right?

If so, could you please try the `limit_set_interface` branch from my repository and see if it works?

https://github.com/thierer/OpenCBM/tree/limit_set_interface

Thanks!

dcutugno, 2020-06-29:
> Thanks, that is helpful (Though I don't understand _why_ it's happening).
>
> From what I understand, you're able to compile opencbm on your machine, right?
>
> If so, could you please try the [...]

Hi, how to compile and install it?

thierer, 2020-06-29:
Sorry, I thought you already compiled and installed the version you have. Where did you get the binaries for version .102 from?

dcutugno, 2020-06-30:
cbmctrl is v .102 from official repo, however i tried to manually compile your source but it complains about some dylib that i don't know

thierer, 2020-06-30:
I'm not sure I understand. Did you download MacOS binaries for .102 (I can't find any neither here on github nor on the sourceforge page) or did were you able to compile the .102 sources and just have trouble compiling the sources from my tree? Could you post the actual error message you get when you try to compile?

spiro-trikaliotis, 2020-06-30:
@dcutugno, are you using MacPorts from http://lallafa.de/blog/c64-projects/opencbm-on-mac/? That one has not been updated and is a little bit dated. The bad thing: It is based on libusb0 (called 'libusb-legacy' on MacOS), but the latest git version needs libusb1!

I asked on the page of lallafa if it can be updated.

I do not know much about the MacPorts system. However, you might want to try the file I attached. I tried to change it as sensible as possible. I looked up that libusb is called this way on MacPorts (by checking https://ports.macports.org/?search=libusb&search_by=name)

However, I cannot guarantee anything, as I do not know MacPorts at all! So, use at own risk!

[opencbm-portfile.zip](https://github.com/OpenCBM/OpenCBM/files/4854199/opencbm-portfile.zip)

thierer, 2020-07-01:
> I'm not sure I understand. Did you download MacOS binaries for .102 (I can't find any neither here on github nor on the sourceforge page) or did were you able to compile the .102 sources and just [...]

Never mind. I was able to reproduce your problem on my macbook and can confirm that the changes from my tree should fix it.

It should work again in Spiro's next release. Sorry for the inconvenience.

dcutugno, 2020-07-04:
confirmed now working, thank you.

closed 2020-07-04 by dcutugno