OpenCBM/OpenCBM #58: GUI Progress Bar

spiro-trikaliotis, 2020-07-06:
Taken over from https://sourceforge.net/p/opencbm/feature-requests/20/:

Just wondering if some sort of progress bar could be added into Gui4Cbm4win so when a D64 image is being written it shows some sort of progress bar with Track/Sector counts?

____________

There are two issues:

First: Noone is actually working on Gui4cbm4win anymore - at least, not on the version on this page. I do not even have a development environment for it, so, I cannot do anything on it.

Second: gui4cbm4win uses the OpenCBM tools in a very dump way. I cannt envision a way how this could be accomplished in the first place. OTOH, I have to admit that I never looked into the gui4cbm4win code, thus, it might be easier than I believe.

I will keep this open in case someone wants to work on it, but I cannot promise you anything, I am sorry.

    Spiro

__________
I have written a new GUI for the OpenCBM tools. You can find it at http://c64.mvgrafx.net
___________

Zirias, 2020-07-22:
Hmm, I'd have doubts maintaining a VB.NET application nowadays makes much sense. Qt seems like a good way to go (supporting all target platforms of OpenCBM), but the project linked above looks pretty unmaintained?

spiro-trikaliotis, 2020-07-22:
The Gui4cbm4win is not the right tool, that's right.
However, it was used as a base for cbmxfer, which IS actually used on Windows.

A progress bar would work there, too. It is only the mechanism that is missing. That's why I did not close this issue yet.

Of course, a much cleaner solution would be if cbmxfer would not use d64copy, cbmcopy and cbmctrl directly, but use the API instead. In fact, there is the libcbmcopy and libd64copy in order to allow exactly for this!

However, it seems this is more work than cbmxfer & Gui4cbm4win are willing to do.

Zirias, 2020-07-22:
Ah, maybe I was confused by seeing this VB.NET GUI being part of the repository here :) Just had a look at cbmxfer, learned it's VB6, ok then ... sorry for the noise ;)