OpenCBM/OpenCBM #69: Suggestion: Change documentation regarding FreeBSD

Zirias, 2020-07-06:
First of all, thank you guys for pulling my FreeBSD-specific patches so quickly :) I just updated the FreeBSD ports for OpenCBM here:
https://www.freshports.org/comms/opencbm/
https://www.freshports.org/comms/opencbm-kmod/
https://www.freshports.org/comms/opencbm-plugin-xa1541/

The ports (and official packages built from them) are confirmed to work fine by a few people. So, I would suggest to change the docs to recommend using them for FreeBSD. The main reason is, for the average user, building and installing from official ports is a lot easier :) If you have any comments on them or think I did something wrong, please let me know. I will do my best to keep these ports always up to date!

Furthermore, the docs currently claim that on FreeBSD, only USB cables are supported. This is no longer true -- at least for me, it works fine with an XMP-1541 cable connected to my trusty old 1541 :) Although I still don't know whether anyone other than me ever tried so far, I'd suggest to change this in the docs as well. As soon as my package builder finishes another large build, I'll check myself whether it still works fine on 0.4.99.102, but I'm pretty sure it will :)

I'm not creating a pull request for this as I really don't want to mess with other people's docs! So, just take it as a suggestion.

Zirias, 2020-07-07:
Tested now, 0.4.99.102 works fine with the xa1541 plugin and an xmp-1541 cable!

spiro-trikaliotis, 2020-07-07:
Hello Felix,
thank you for your commentary. 
I would be happy to merge a PR from you. :)

So, I take it that you want an update the documentation AND also remove the ports file from opencbm/ports/FreeBSD-Ports, right? To be honest, I do not know the ports system good enough to do anything on it, so I would be glad if someone else does this for me. Thus, I do not have any objections.

spiro-trikaliotis, 2020-07-07:
Please also note that I want to bring out a v0.5.0 release. Thus, it is very likely the ports file would not some update soon.

Zirias, 2020-07-07:
Hi Spiro,
also thanks for your response, it's helpful! I totally forgot about the port files integrated here, but now that you mention it, I remember having tried that a long time ago and it didn't work. I assume this was created a long time ago and worked back then, but of course, FreeBSD ports are a moving target :) So yes, I would suggest removing that. When I prepared my ports for inclusion in the official ports tree, I decided to split in three ports, because a kernel module that's only needed for *some* usecases should never be forced onto the user -- and as the xa1541 plugin needs this module, this plugin must be separated as well. Fortunately, your build system supports building just "parts". For building the xa1541 plugin on FreeBSD, I had to "convince" the build system to do that by passing some other variable, as it isn't built on FreeBSD normally -- I think that's another minor thing to change now that the kernel module is available.

If you really think that's fine, I will prepare a PR with my suggestions for documentation changes in the next days. Btw, for the "opencbm-kmod" port, I wrote a manpage to go with the kernel module -- maybe you want to adopt it as well? It can be found here: https://svnweb.freebsd.org/ports/head/comms/opencbm-kmod/files/opencbm.4?view=markup&pathrev=538094
Looks like this when rendered:
[image]

spiro-trikaliotis, 2020-07-07:
Hi Felix,
of course, I would be glad to be able to integrate that! I always like people who know what they do (I do not for FreeBSD, for sure) to document this.

Having said this, I just put up a VM with FreeBSD. That's the first time I installed a BSD since 15 years or so. (I do not count SVR4 on which I worked many, many moons ago for a very short time)

Thus, once the documentation is up, I can be a first tester of it, at least of the USB parts :)

spiro-trikaliotis, 2020-07-13:
Am I correct that we can close it as I merged #70 ?

spiro-trikaliotis, 2020-07-13:
Do you want to have a look into the file README-OSX.txt in the source root, and adjust that one accordingly?

Zirias, 2020-07-14:
> Am I correct that we can close it as I merged #70 ?
As far as FreeBSD is concerned, I would say yes.
> Do you want to have a look into the file README-OSX.txt in the source root, and adjust that one accordingly?
I can have a look, but I'm not sure I can contribute something meaningful as I don't use Mac myself and don't know too much about it ...

spiro-trikaliotis, 2020-07-14:
Erm... My mistake, I am sorry. I read "ports" and immediately, I had FreeBSD in my mind. Of course, this is MacOS. I was a little bit confused.

closed 2020-07-14 by spiro-trikaliotis