OpenCBM/OpenCBM #71: Question: canonical project website?

Zirias, 2020-07-22:
I'm preparing the updated FreeBSD port for 0.4.99.103 (good news, so far it seems no single patch will be necessary!) and I'm wondering whether this in the description is still correct:

WWW: http://opencbm.sourceforge.net/

Right now, it redirects to another page (should I reference that directly), but it still shows 0.4.99.99 as the latest version, therefore I'm not sure about it? Thanks for your input :)

spiro-trikaliotis, 2020-07-22:
I would say the "canonical" webpage is: https://github.com/OpenCBM/OpenCBM/

I will update the SF page, too, but the main development is on github now.

Zirias, 2020-07-22:
Ok, I will change the pkg-descr of the FreeBSD ports accordingly, thanks!

closed 2020-07-22 by Zirias