OpenCBM/OpenCBM #72: Building individual plugin only

Zirias, 2020-07-22:
For the FreeBSD ports, I have the need to build `plugin-xa1541` in isolation. So far, this worked with something like this:

    gmake CC=cc SUBDIRS_PLUGIN_XA1541=opencbm/lib/plugin/xa1541 plugin-xa1541
    gmake CC=cc SUBDIRS_PLUGIN_XA1541=opencbm/lib/plugin/xa1541 install-plugin-xa1541

Now, I assume after https://github.com/OpenCBM/OpenCBM/commit/b8af366e4fa4890ea581f0f7b486f5435756b5f2 -- this also builds and installs the opencbm lib. I could work around by just deleting these unwanted files after staging, but that's not the best solution. Is the dependency causing this really needed to correctly support parallel make? I have to admit I only partially understand your build system :)

The reason for this need is that application packages should never include a kernel module, and as this plugin depends on the kernel module, I end up with 3 packages -- opencbm itself, the kernel module and the xa1541 plugin. I assume a simple way to build it in isolation would come handy for Linux packagers as well.

While analyzing the situation, I found there are some things in SUBDIRS_OPTIONAL, I assume they are never built by default? Should I attempt to package them as well, and is there some documentation on their purpose?

Thanks and BR,
Felix

spiro-trikaliotis, 2020-07-22:
Good point.
It might make sense to add additional build targets (module and install-module come to mind) in order to have only the module compiled.

On Debian, things are quite different, as there, we provide the module source only. The DKMS system makes sure that it is compiled with the right kernel version then.

Zirias, 2020-07-22:
> It might make sense to add additional build targets (module and install-module come to mind) in order to have only the module compiled.

Well, just using the `[install-]plugin-xa1541` target (with custom `SUBDIRS_PLUGIN_XA1541` to avoid also building the kernel module) still works, but with the added dependencies, it builds/installs other stuff as well. I think building it and then removing it to avoid inclusion in the package is a workaround at best. I wonder whether the dependencies leading to that are really necessary? After all, building *only* the xa1541 plugin worked before, so it doesn't seem to depend on other stuff in the source tree? Maybe the other plugins do?

Zirias, 2020-07-22:
IOW, assuming the other plugins indeed depend on some common code, don't you think something like this would work?
https://github.com/Zirias/OpenCBM/commit/33f8706ed02526082d2c731dcbb644776f717504

While at it, are these dependencies for `install` targets needed at all?

spiro-trikaliotis, 2020-09-14:
I must admit that I do not understand the build system myself anymore. :(
It would need some general work to make it work as expected again.

In the meantime, your proposed fix works for me without any negative effect, so I took it over and commited it in 7121ff7496e106.

What do you mean with "While at it, are these dependencies for install targets needed at all?"? What would you change here?

Zirias, 2020-09-26:
Hi Spiro, thanks for your reply!
First of all, I feel the pain (really!) to maintain a build system only based on (GNU) `make`. But I still think it's worthwile for small to medium-size projects cause it avoids large dependencies :) This was btw my motivation to create a "meta" build system: https://github.com/Zirias/zimk -- and I use it e.g. for my D64-tool here: https://github.com/excess-c64/v1541commander -- but it's far from "production quality" and lacks almost any documentation ;)

As for your question: I don't see why for the `install` targets, there are any dependencies *at all*. I see the need for the build targets, as some of them depend on the "base" of course. But installing should work without any dependencies? Please correct me if I'm wrong :)

closed 2022-07-15 by Zirias