OpenCBM/OpenCBM #73: d64copy hangs on missing sector header

ymgve, 2020-09-13:
It seems like some times if d64copy encounters a sector with a missing header, it will hang indefinitely.

This is with an XUM1541 ProMicro board and transfer mode serial1 and serial2. Some times using original transfer works, since it seems to bump halftracks to search for slightly misaligned sectors.

ymgve, 2020-09-13:
To reproduce, try reading tracks 35 to 36 on an un-expanded disk - for a lot of disks, it will hang on track 36 and never finish.

spiro-trikaliotis, 2020-09-13:
Hello,
thank you for your report.

I know that there are some issues in d64copy (and cbmcopy, btw.) when disks do not conform to what they expect.
Your specific problem was not yet known to me, but it fits into the other problems.

IMHO, both tools need a massive rewrite. Fixing them in d64copy and imgcopy might solve the one specific problem, but will not have great impact overall.
That's why I started setting the basis for a rewrite before the USB based cables (XU1541, XUM1541) appeared years ago. The USB based solutions took all of my available time, so this did not happen. Unfortunately, at the moment, I do not have the time to do it.