OpenCBM/OpenCBM #74: building on debian, there is no /LINUX/plugin_helper_tools

abzman, 2020-10-19:
I've been following this guide: https://keathmilligan.net/vintage-c64-computing-opencbm-and-raspberry-pi even though I'm installing on a debian VM and not a raspberry pi.  I get to the point where I run "make -f LINUX/Makefile install install-plugin-xum1541" and I get this output (keep in mind I've run this command several times trying some different things so it has nothing to do until the very end)

abzman@compile-environment:~/OpenCBM$ sudo make -f LINUX/Makefile install install-plugin-xum1541
make -C opencbm/include -f LINUX/Makefile install-files
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/include'
install -d /usr/local/include
install -m 644 opencbm.h d64copy.h cbmcopy.h /usr/local/include
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/include'
make: Circular BUILDSYSTEM.opencbm/arch/linux.all <- BUILDSYSTEM.opencbm/lib.all dependency dropped.
make -C opencbm/arch/linux -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/arch/linux'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/arch/linux'
make: Circular BUILDSYSTEM.opencbm/libmisc.all <- BUILDSYSTEM.opencbm/lib.all dependency dropped.
make -C opencbm/libmisc -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/libmisc'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/libmisc'
make: Circular BUILDSYSTEM.opencbm/lib.all <- BUILDSYSTEM.opencbm/lib.all dependency dropped.
make -C opencbm/lib -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/lib'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/lib'
make -C opencbm/include -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/include'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/include'
make -C opencbm/libtrans -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/libtrans'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/libtrans'
make -C opencbm/cbmctrl -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmctrl'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmctrl'
make -C opencbm/cbmformat -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmformat'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmformat'
make -C opencbm/cbmforng -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmforng'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmforng'
make -C opencbm/d64copy -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/d64copy'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/d64copy'
make -C opencbm/cbmcopy -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmcopy'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmcopy'
make -C opencbm/d82copy -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/d82copy'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/d82copy'
make -C opencbm/imgcopy -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/imgcopy'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/imgcopy'
make -C opencbm/demo/flash -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/demo/flash'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/demo/flash'
make -C opencbm/demo/morse -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/demo/morse'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/demo/morse'
make -C opencbm/demo/rpm1541 -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/demo/rpm1541'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/demo/rpm1541'
make -C opencbm/sample/libtrans -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/sample/libtrans'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/sample/libtrans'
make -C opencbm/sample/testlines -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/sample/testlines'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/sample/testlines'
make -C opencbm/compat -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/compat'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/compat'
test ! -e opencbm/addon/LINUX/Makefile || make -C opencbm/addon -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/addon'
make[2]: Entering directory '/home/abzman/OpenCBM/opencbm/addon/frm_analyzer'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/abzman/OpenCBM/opencbm/addon/frm_analyzer'
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/addon'
test ! -e opencbm/nibtools/LINUX/Makefile || make -C opencbm/nibtools -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/nibtools'
make -f GNU/Makefile CBM_LNX_PATH="../" linux
make[2]: Entering directory '/home/abzman/OpenCBM/opencbm/nibtools'
make CFLAGS="-I include/LINUX/ -I ..//include -W -Wall -Wstrict-prototypes -Wno-unused-parameter -Wpointer-arith   -std=c99" \
	LDFLAGS="-L..//lib -lopencbm" \
	-f GNU/Makefile \
	nibread nibwrite nibconv nibscan nibrepair nibsrqtest
make[3]: Entering directory '/home/abzman/OpenCBM/opencbm/nibtools'
make[3]: 'nibread' is up to date.
make[3]: 'nibwrite' is up to date.
make[3]: 'nibconv' is up to date.
make[3]: 'nibscan' is up to date.
make[3]: 'nibrepair' is up to date.
make[3]: 'nibsrqtest' is up to date.
make[3]: Leaving directory '/home/abzman/OpenCBM/opencbm/nibtools'
make[2]: Leaving directory '/home/abzman/OpenCBM/opencbm/nibtools'
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/nibtools'
test ! -e opencbm/mnib36/LINUX/Makefile || make -C opencbm/mnib36 -f LINUX/Makefile all
test ! -e opencbm/cbmrpm41/LINUX/Makefile || make -C opencbm/cbmrpm41 -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmrpm41'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmrpm41'
test ! -e opencbm/cbmlinetester/LINUX/Makefile || make -C opencbm/cbmlinetester -f LINUX/Makefile all
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/cbmlinetester'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/cbmlinetester'
mkdir -p -m 755 /usr/local/bin /usr/local/lib /usr/local/man/man1 /usr/local/include /usr/local/info
mkdir -p -m 755 /lib/modules/4.19.0-11-amd64/kernel/drivers/char
make -C opencbm/arch/linux -f LINUX/Makefile install-files
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/arch/linux'
make[1]: Nothing to be done for 'install-files'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/arch/linux'
make -C opencbm/libmisc -f LINUX/Makefile install-files
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/libmisc'
make[1]: Nothing to be done for 'install-files'.
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/libmisc'
make -C opencbm/lib -f LINUX/Makefile install-files
make[1]: Entering directory '/home/abzman/OpenCBM/opencbm/lib'
install -d /usr/local/lib
install -m 755 libopencbm.so.0.4.99.103 /usr/local/lib
install -m 644 libopencbm.a /usr/local/lib
cd /usr/local/lib && ln -sf libopencbm.so.0.4.99.103 libopencbm.so.0; ln -sf libopencbm.so.0 libopencbm.so
# Install the plugin helper tool
install -d /usr/local/bin
install -m 755 ..//LINUX/plugin_helper_tools /usr/local/bin/opencbm_plugin_helper_tools
# mark that there is no default plugin yet.
..//LINUX/plugin_helper_tools setdefaultplugin "/etc" 00opencbm.conf ""
/bin/sh: 1: ..//LINUX/plugin_helper_tools: not found
make[1]: *** [../LINUX/librules.make:49: install-lib] Error 127
make[1]: Leaving directory '/home/abzman/OpenCBM/opencbm/lib'
make: *** [LINUX/Makefile:79: BUILDSYSTEM.opencbm/lib.install-files] Error 2

spiro-trikaliotis, 2020-10-19:
Can you check the file ~/OpenCBM/opencbm/LINUX/plugin_helper_tools? Does it exist? Which permissions does it have?

Do you have some unusual mount options, especially for your home (~) and/or ~/OpenCBM/?

If you cd into ~/OpenCBM/opencbm/lib, can you execute
../LINUX/plugin_helper_tools
and
..//LINUX/plugin_helper_tools
?

Can you do the same with a sudo before the commands?

spiro-trikaliotis, 2020-10-19:
Oh, and while you follow the guide: The "sudo" before cbmctrl should not be necessary!

abzman, 2020-10-19:
That file is not present in /LINUX, was it supposed to be put there by another script because if you look in this repo there's only a makefile in there.  

I don't think my mount options are too odd, it's a single partition VM and I just have OpenCBM in my home folder

spiro-trikaliotis, 2020-10-19:
If it is not there, your checkout is broken. I can clearly see it here:
https://github.com/OpenCBM/OpenCBM/tree/master/opencbm/LINUX

Note: There are many LINUX/ directories. You must have a look into the right one.

Please check "git status", and try the command "find . -name plugin_helper_tools"

abzman, 2020-10-19:
It seems like there's an error in that make file then, because if I execute those commands from withing the lowercase opencbm directory that contains the LINUX directory that has plugin_helper_tools then they won't run because that's not where Makefile is located.  If I execute those commands from the OpenCBM directory then the first one completes but the second one is looking for program_helper_tools in the wrong LINUX directory.  

I can move plugin_helper_tools to the other folder where the makefile is looking for it, but that's not where it is now

spiro-trikaliotis, 2020-10-20:
You cannot execute the commands from arbitrary directories. As can be seen in your make output, the plugin_helper_tools are executed with a current directory of ~/OpenCBM/opencbm/lib as ..//LINUX/plugin_helper_tools, so plugin_helper_tools must be in ~/OpenCBM/opencbm/LINUX/, where it is.

Can you please literally (!) execute the commands and give me the outputs:
cd ~/OpenCBM/opencbm/lib; ../LINUX/plugin_helper_tools
and
cd ~/OpenCBM/opencbm/lib; ..//LINUX/plugin_helper_tools

abzman, 2020-10-20:
here's what I get from those commands:

abzman@compile-environment:~$ cd ~/OpenCBM/opencbm/lib; ../LINUX/plugin_helper_tools
bash: ../LINUX/plugin_helper_tools: /bin/sh^M: bad interpreter: No such file or directory
abzman@compile-environment:~/OpenCBM/opencbm/lib$ cd ~/OpenCBM/opencbm/lib; ..//LINUX/plugin_helper_tools
bash: ..//LINUX/plugin_helper_tools: /bin/sh^M: bad interpreter: No such file or directory

abzman, 2020-10-20:
I checked, /bin/sh exists as a link to dash and after the dos2unix treatment I get this:

```
abzman@compile-environment:~/OpenCBM/opencbm/lib$ cd ~/OpenCBM/opencbm/lib; ..//LINUX/plugin_helper_tools
..//LINUX/plugin_helper_tools: 114: shift: can't shift that many
abzman@compile-environment:~/OpenCBM/opencbm/lib$ cd ~/OpenCBM/opencbm/lib; ../LINUX/plugin_helper_tools
../LINUX/plugin_helper_tools: 114: shift: can't shift that many
```

spiro-trikaliotis, 2020-10-21:
Ok, that's good. No, plugin_helper_tools complaints because it does not get any parameters.

Now, if you try to install again, it should work.

abzman, 2020-10-23:
It did, thanks for all your help.  Now I can get to work an my modded version of the XUM1541 based on a teensy 2.0, a 7406, and the IEEE488 bus driver ICs (I don't think any current design uses the last one on that list).

closed 2020-10-23 by abzman