OpenCBM/OpenCBM #77: Imgcopy error using OpenCBM 0.4.99.103 release with xu1541

DondInvents, 2021-02-08:
Hi, I'm a first time bug submitter. I have some C64 floppies I would like to back up to a more reliable medium. I just bought a xu1541 board (USB) and was really pleased that countless folks have put together openCBM to help in the endeavor. Thanks. 

Playing around with the OpenCBM 0.4.99.103 release and noticed some odd errors when trying to perform an imgcopy. My OS is Windows 10 Ver 10.0.18362 and the floppy drive is a first generation VIC 1541. The xu1541 appears to be talking to the VIC 1541 and have been able to use the d64copy app multiple times without errors or warnings on some of my floppies, but some of the floppies are copy protected and thought using imgcopy would be best, but that is where it gets odd. For example:

 C:\Program Files\opencbm>imgcopy -vvv 8 test.d64
[Debug] transfer mode is 0
[Debug] decided to use transfer mode 1
[Debug] imagetype D80 from drive type
[Warning] invalid imagetype for this drive type
[Info] Trying to identify drive type
[Fatal] drive is not supported

No certain why the imagetype of the drive type is D80? I thought it should be D64 for the 1541 drive
Trying again using -d 1541

C:\Program Files\opencbm>imgcopy -vvv -d 1541 8 test.d64
[Debug] transfer mode is 0
[Debug] decided to use transfer mode 1
[Debug] block count: -538976288
[Fatal] invalid drive or image type

Curious about block count -538976288 ? Tried going with d80

C:\Program Files\opencbm>imgcopy -vvv  8 test.d80
[Debug] transfer mode is 0
[Debug] decided to use transfer mode 1
[Debug] imagetype D80 from file extension
[Debug] imagetype D80 from drive type
[Debug] block count: 2083
[Info] trying to identify drive type
[Fatal] drive is not supported

Block count no long negative, but think should be 680 something. Lastly tried adding -d 1541

C:\Program Files\opencbm>imgcopy -vvv  -d 1541 8 test.d80
[Debug] transfer mode is 0
[Debug] decided to use transfer mode 1
[Debug] imagetype D80 from file extension
[Warning] invalid imagetype for this drive type
[Warning] invalid imagetype for this drive type
[Fatal] invalid imagetype for this drive type

Hope this testing helps.
Thanks again.
Don

(P.S. 1) Was able to install on Virtualbox the identical Windows installation as mentioned above and got the same debug and fatal errors, but the block count was no longer -538976288, but -137864329. I also noticed the access Led on the xu1541 never turns on when using any of the imgcopy variations above, but works fine with d64copy. Tried using d64copy on some of the copy protected floppies and get LIBUSB_ERROR_TIMEOUT or LIBUSB_ERROR_PIPE. )

spiro-trikaliotis, 2021-02-08:
Well... It is not simple.

The simple answer: Use d64copy instead of imgcopy.

The more complicated answer: imgcopy was written, but is completely unmaintained at the moment. In fact, I do not even know how accurate it is at the moment. I would not recommend anyone to use it at the current state.

DondInvents, 2021-02-08:
Thanks  spiro-trikaliotis for the reply and suggestion. As mentioned in my previous message d64copy fails on the copy protected floppies I get either LIBUSB_ERROR_TIMEOUT or LIBUSB_ERROR_PIPE  messages. I have tried running on an older slower system as well as my I7 laptop using virtual machines and get the same errors. Any thoughts would be appreciated. Thanks - Don

detlefgerhardt, 2021-03-05:
Hi, I just tried to make an image of an 1581 disk connected to a xum1541 and this didn't work either (imgcopy aborts without any message an leaves the connection open).
I can not use d64copy with 1581. Does that mean that I can not use OpenCBM to make image from my 1581 at all?

spiro-trikaliotis, 2021-10-06:
@detlefgerhardt, see #89. imgcopy is broken. The description in #89 has a work-around for the current problems.

closed 2021-10-06 by spiro-trikaliotis