OpenCBM/OpenCBM #78: make breaks with syntax error in usbcommon.h with Debian Buster

jwich71, 2021-03-11:
Building opencbm version 0.4.99.103 under Debian Buster breaks at usbcommon.h  at line 36 with a syntax error because of missing quotation marks. After adding quotation marks to this line the build works like expected:

#error "Could not find the libusb 1.0 development packages. Please install them and retry!"

I think this could be a bug.

spiro-trikaliotis, 2021-03-11:
Hello @jwich71, 
my main development machine is a Debian Buster (10.8). I cannot see the problem you are mentioning.

What is the output of the following commands?

$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0

and

$ cpp --version
cpp (Debian 8.3.0-6) 8.3.0

I have had a look at the documentation at https://gcc.gnu.org/onlinedocs/cpp/Diagnostics.html. While the examples all show a quotation mark, I see the following text:

The directive ‘#error’ causes the preprocessor to report a fatal error. **The tokens forming the rest of the line following ‘#error’ are used as the error message.**

I understand this as the quotation marks are not required - and, in fact, that line works on my machine as expected, regardless if it is used with or without quotation marks.

Can you give me the output of the above line, and the exact output of your failing compilation?

jwich71, 2021-03-12:
Unfortunately I can not reproduce the issue anymore. After removing the quotation marks again from usbcommon.h the make -f LINUX/Makefile opencbm plugin-xa1541 still works. When the make broke, it gave me the above message about missing libusb-dev although it was already installed. So I assumed, the error message for it self was wrong. And after adding quotation marks to the error message, the make also worked like expected. But now I removed the quotation marks again and the make still works. Strange thing! I'm no developer, but that's what I observed.

That's my environment:

$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cpp --version
cpp (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 8
model name      : Pentium III (Coppermine)
stepping        : 10
microcode       : 0x4
cpu MHz         : 846.263
cache size      : 256 KB
physical id     : 0
siblings        : 1
core id         : 0
cpu cores       : 1
apicid          : 0
initial apicid  : 0
fdiv_bug        : no
f00f_bug        : no
coma_bug        : no
fpu             : yes
fpu_exception   : yes
cpuid level     : 2
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 sep mtrr pge mca cmov pse36 mmx fxsr sse cpuid pti
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 1692.52
clflush size    : 32
cache_alignment : 32
address sizes   : 36 bits physical, 32 bits virtual
power management:

$ uname -a
Linux q4os-desktop 4.19.0-14-686-pae #1 SMP Debian 4.19.171-2 (2021-01-30) i686 GNU/Linux

I love my old HP OmniBook XE3 i386 with 128MB RAM because it sill works with xa1541 cable and it's perfect to copy C64 images to real Floppies ;)

spiro-trikaliotis, 2021-03-12:
Well, I missed the point that you are building the xa1541 plugin. In this case, libusb should not be needed at all. I am not sure if it works as expected, though.

I will check.

Anyway, I will close this issue here.

closed 2021-03-12 by spiro-trikaliotis