OpenCBM/OpenCBM #79: cbmcopy custom path

TjOsT1975, 2021-03-28:
it would be nice if I could ouse a custom path so save a prp from floppy to my computer.

like this : cbmcopy -r 8 file /users/me/c64/file.prg

spiro-trikaliotis, 2021-03-28:
Hello,

did you try the following command:

`cbmcopy -r -o /users/me/c64/file.prg 8 file`

TjOsT1975, 2021-03-28:
sadly it does not work. it save the file in the directory where I am in terminal and renamed it to __Users_marcoraap_c64_file.prg

spiro-trikaliotis, 2021-03-28:
You are right. I just looked into the sources: For reading, it intentionally replaces slashes with underlines.
Why? I do not know.

I will have a look into it, it should not be too complicated to add this option.

TjOsT1975, 2021-03-29:
hi, thx for quick response.
just one wish. while copying a .prg file to real disk the name of the file include the .prg. is the something you can fix?

thierer, 2021-04-05:
> You are right. I just looked into the sources: For reading, it intentionally replaces slashes with underlines.
Why? I do not know.

It **does** make sense when the local filename is derived from the CBM name, just not when it is provided by the "-o" option.

> just one wish. while copying a .prg file to real disk the name of the file include the .prg. is the something you can fix?

I agree, this bothers me too.

I just opened PR #80, which fixes both issues (and some minor others that I found in the process).