OpenCBM/OpenCBM #82: Use OpenCBM to emulate a peripheral (eg drive) rather than emulating a host?

tlokovic, 2021-06-29:
The OpenCBM commands are all based around emulating an IEC host, with the assumption that the ZoomFloppy (or xu1541 or whatever) is connecting a modern machine to a legacy peripheral.

Can it be used in reverse, to emulate a peripheral, and use the ZoomFloppy to talk to a legacy host (eg a C64)?  I was hoping to connect my C64 to my Windows machine (via xu1541/USB) and implement an emulated 1541 on the Windows machine.

Essentially I'm imagining something like Pi1541 but running on Windows, and using the xu1541 to deal with the realtime serial bus interfacing.  (I know Pi1541 relies on baremetal realtime on the Pi, and Windows wouldn't support that, but I figured the xu1541 would manage the realtime part.)  More generally, I'm imagining using my Windows machine to emulate multiple peripherals--imagine multiple floppies and a printer--and have the C64 connected to that via IEC/xu1541/USB.

I thought this would be possible, using OpenCBM to talk to the serial bus as a peripheral would, rather than talking to it the way a host would.  But digging into the code, now I'm not so sure.

tlokovic, 2021-06-30:
In particular, based on https://web.archive.org/web/20170116173410/https://www.zimmers.net/anonftp/pub/cbm/programming/serial-bus.pdf
I guess I'm asking if I can write OpenCBM code to listen to the ATN line rather than driving the ATN line, so that I'm acting like a peripheral rather than the host.

spiro-trikaliotis, 2021-06-30:
In principle, this is possible on the XUM1541. However, you would have to modify the firmware of the USB device, as it is not supported at the moment.

BTW: Are you speaking about the Zoomfloppy (XU**M**1541) or the older XU1541 device?

closed 2021-06-30 by spiro-trikaliotis

tlokovic, 2021-06-30:
Thank you for your response.  I happen to own a XU1541 but I was asking if OpenCBM lets you express peripheral-side semantics with any of these devices (XUM1541, XU1541, etc).  You say the XUM1541 might be able to support this with modified firmware (which is good to know), but I'm assuming then you'd need a modified driver and a modified version of OpenCBM (or something else) that supports the subtly different semantics.  In particular, a peripheral needs to be listening on the bus, reacting when the host mention of its device number, *not* reacting when the host mentions other devices, etc.  I don't see a way in OpenCBM's API to express that side the conversation.  I'm sure it's possible, just much more involved than I had hoped.

Thank you for clarifying!