OpenCBM/OpenCBM #87: OpenCBM doesn't appear in device manager and Gui won't install properly

daniel-strang, 2021-09-28:
Hello. I have a ZoomFloppy which I am trying to use to copy diskettes from a 1581 drive to my laptop which is running on Windows 10 using a USB cable.  I was able to install it using the old software but it wouldn't make .d81 files. Presently I am trying to install the OpenCBM 9.4.99.103. After installation the 1581 does respond to commands on the command line. However, it doesn't appear in the device manager. Furthermore, when I try to install CBMTransfer it can't find my floppy drive. I am at a loss as to what to try next.

daniel-strang, 2021-09-28:
I beg your pardon. I have found zum1541 in the device manager under Universal Serial Bus Devices. In the case of the previous installation it was under Other Devices.

wbochar, 2021-09-28:
I had the same issue, I tired the zadig method shown here and things worked fine..
https://retrotinker.net/opencbm-installation-unter-windows-10/

spiro-trikaliotis, 2021-09-28:
@wbochar 
Using zadig is a way to do it, but the way it is described in the link, it does not work with 0.4.99.103 anymore. These instructions are outdated.
Additionally, if the driver would not work,

@daniel-strang Regarding CBMTransfer, I cannot help you, as I do not know that program. You must ask the author of that tool.

However, your report sounds as if the basic installation does not work at all.
First, if the ZF appears under "universal serial bus devices", then this is a good sign, that the driver was installed successfully.
Are you able to use the command line? If so, please try the following commands:

`cbmctrl reset`
`cbmctrl detect`

Do you get any output?

If not, try again with `set XUM1541_DEBUG=9`, and then issue both commands again, one after the other. Please attach the output you get here.

wbochar, 2021-09-28:
I installed 0.4.99.103, none of the command line tools like cbmctrl would detect anything. It would show up in device manager. Before this, I was on .99 everything was working. I uninstalled opencbm from the control panel, then installed .130 --> then an tool would throw errors that it couldn’t find the zoomfloppy. I did the zadig thing .103 and then the tools started to work.

spiro-trikaliotis, 2021-09-28:
> I installed 0.4.99.103, none of the command line tools like cbmctrl would detect anything. It would show up in device manager.

Yes, but not as "universal serial bus device" (which suggest WinUSB, that is, libusb v1.0 API) but as "libusb-win32 device" (which suggest libusb-win32, that is, libusb v0.1 API), right?

> Before this, I was on .99 everything was working.

From .99 to .103, the requirement changed: Before, it had to be libusb-win32 (which implements the libusb v0.1 API). With .103, it must be WinUSB (which implements the libusb v1.0 API).

The version must match! That is, .99 will not work with WinUSB, and .103 will not work with libusb-win32.

>  I did the zadig thing .103 and then the tools started to work.

But your link above specifically asks to use the libusb-win32 driver with zadig:

Citing from the site:
> If the “xum1541 floppy adapter” is selected, the next step is to find the “libUSB” driver from the list. It is important that it must be the “libusb-win32” driver, even if it is a 64 bit Windows. 

And that's the part that will not work. In fact, your suggestion most probably will break the installation of @daniel-strang more than it is broken now!

So, to be more precise: Your use of zadig only worked because you did not follow the path on the link you provided! If you would have read everything, your setup would not work.

daniel-strang, 2021-09-28:
Yes. In a later revision of my message I said that the commands on the 
command line do work.

spiro-trikaliotis, 2021-09-28:
As I do not know what CBMTransfer does, I do not know how I could handle this problem now.
Would you be willing to talk with the author of CBMTransfer? (Who is it, anyway?)

daniel-strang, 2021-09-28:
I found this GUI in a blog called Retrotinker. There were two 
suggestions and CBMTransfer sounded like the best option.

https://retrotinker.net/opencbm-installation-unter-windows-10/

This is the site of CBMTransfer. I found it to be quite helpful. It is 
run by Steve Gray ***@***.***). I've seen him on various forums. 
I will indeed contact him. However, I suspect that the problem isn't 
with his software.

http://www.6502.org/users/sjgray/software/cbmxfer/cbmxfer.html

daniel-strang, 2021-09-28:
I have looked in my OpenCBM folder in Program Files and found that 
WinUSB wan't there. Therefore, I have just downloaded it and added it to 
the folder. Should I run it or do I have to uninstall and reinstall 
OpenCBM? Should I delete libusb-win32?

daniel-strang, 2021-09-28:
I moved the libusb v1.0 dll to another place for safe keeping. Then I 
added WinUSB into the OpenCBM directory in Program Files. Result is that 
CBMTransfer doesn't see the drive anymore although it does cause a 
reset. I have sent a message to CBMTransfer asking them what I can do 
about this. I am not sure what my next move should be.

spiro-trikaliotis, 2021-09-29:
@daniel-strang , please stop putting random files into the directory! You are most likely breaking your setup more and more!
WinUSB is *not* meant for manual installation; it has to be done with an installer (either ZADIG, that was already suggested above by @wbochar, or my tool which is based on the same code-base as ZADIG).

So, do you remember what you have done, so can you undo your latest changes? (Copying WinUSB, moving libusb, and so on?) If not, then please uninstall OpenCBM (from your control center), then delete c:\program files\opencbm, and start over again with installation (running install.cmd)

If after this, your setup does not work, use ZADIG (https://zadig.akeo.ie/) as suggested above, but make sure to choose the WinUSB driver (I think this is the default now; it was not in v2.2 or so).

spiro-trikaliotis, 2021-09-29:
I was not aware that CBMTransfer is the same is CBMXfer. I know it, and I know Steve, too.
You said in a previous message that the commands on the command-line work. If this is true, then OpenCBM is running and working correctly.
Just to be sure: Have you tested commands like `cbmctrl detect`, `cbmctrl status 8`, `cbmctrl dir 8`? Do they all work?
Do you have a 1541, 1570 or 1571 drive and tested reading from a disk with `d64copy 8 read.d64` to read a disk image from the floppy into the file read.d64?
If all these work, then the problem most probably is with CBMTransfer. If they do not work, we will have to investigate here.

daniel-strang, 2021-09-29:
I have tried using gui4cbm4win but it says that it doesn't support 1581 
drives. Therefore, my only hope seems to be with CBMXfer. Let's hope 
that I hear from Steve in short order.

daniel-strang, 2021-09-29:
I just realized that I didn't send this answer to all. So I am sending 
it again.

I have just installed the WinUSB driver but it unfortunately hasn't made 
any difference. I am still waiting for an answer from Steve. Meanwhile, 
I will try another GUI.

daniel-strang, 2021-10-03:
Hello,

I have been in contact with Steve who has been trying to help me. 
However, it seems that the problem does after all have to do with 
OpenCBM. When I first installed it I did some tests on Command Line but 
not trying to copy files. Upon doing this yesterday I got the message 
"[Fatal] 1581 drives are not supported" while in the OpenCBM folder in 
Program Files. What can I do about this?

Greetings,

Daniel Strang

closed 2021-10-06 by spiro-trikaliotis