OpenCBM/OpenCBM #88: ZoomFloppy hangs on Linux, but works on Windows

dawn-minion, 2021-10-02:
Hi there, I'm trying to get cbmctrl working with my new ZoomFloppy and a Commodore 1520 Plotter. I can't seem to get this project to work correctly when running on my AMD64 Linux laptop. When plugged into a Surface Pro running Windows and the latest cbm4win build it works just fine, but under Linux every command just hangs and ends up blinking the light on the ZoomFloppy. If I run the command again it'll reset the plotter, but then hangs again on the "checking for status" info, never returning.

Here are the debug logs I get when trying `cbmctrl detect`:
```
[XUM1541] scanning usb ...
[XUM1541] device 1d6b:0002
[XUM1541] device 0483:2016
[XUM1541] device 1d6b:0001
[XUM1541] device 0a5c:21e8
[XUM1541] device 1d6b:0001
[XUM1541] device 1d6b:0001
[XUM1541] device 16d0:0504
[XUM1541] found xu/xum1541 version 0208 on bus 2, device 4
[XUM1541] xum1541 name: xum1541 floppy adapter (ZOOMFLOPPY)
[XUM1541] xum1541 serial number:   0
[XUM1541] firmware version 8, library version 8
[XUM1541] device capabilities 1b status 00
[XUM1541] [xum1541_init] Tape supported, disk mode entered.
[XUM1541] firmware git revision is 00673ba6
[XUM1541] compiled with avr-gcc version 10.3.1 2
[XUM1541] and using avr-libc version 2.0.0
[XUM1541] write 16 2 bytes from address 0x7fffdb8a00ce flags 2
[XUM1541] wrote 2 bytes (28 6f)
[XUM1541] xum1541_wait_status checking for status
```

I cannot get any command to ever work under Linux. I tried compiling the firmware in this project and flashing it in case it was something fixed, but the issue remains. It will get to this "checking for status" line and simply never return no matter how long I wait. The equivalent commands on the Windows version return near instantly. 

Any idea on what I can try here? The laptop isn't very special, it's just an older Thinkpad T60 running openSUSE. The Plotter itself works just fine too with the same cables connected to my actual C64.

dawn-minion, 2021-10-02:
Weird, I've just tried it on Debian on PPC64 and it works flawlessly there - the problem seems to be isolated to openSUSE on this laptop. Really not sure why. Unless there's something obvious I can try here, I guess it's not too big of an issue :)

spiro-trikaliotis, 2021-10-02:
@thierer, might be USB related. Can you have a look?

thierer, 2021-10-02:
@dawn-minion I'm not entirely sure what you're trying to achieve with the plotter? I'm not even sure how `cbmctrl detect` would behave, because it's primary purpose is to detect disc drives. What exactly does work on Windows and the PPC machine, assuming you're also using the plotter for these tests? (That said, of course `cbmctrl` shouldn't hang in any case).

Do you also have a CBM disc drive that you could try with `cbmctrl status` or `cbmctrl dir`?

You write "... under Linux every command just hangs ...". Is it really **every** command? I had a similar problem, but the first command always worked and only the next commands could hang. Is really the **first** command after plugging the Zoomfloppy into the usb port already hanging?

Could you please check:

1. which Linux kernel version you are running (`uname -r`)
2. which USB driver is used for the Zoomfloppy. After you plug it in, `dmesg` should report something like "`new full-speed USB device number <x> using <y>`". I'm interested in "y". If it's "`xhci_hcd`", then it's probably an USB3 port. If the laptop has an USB2 port: Does the problem persist if you use that?
3. which libusb version you are using? (I know nothing about the package manager of opensuse, so I can't help you with that.)
4. Could you compile and flash the firmware from revision eedaa3f and check if it makes a difference?

Thanks!

dawn-minion, 2021-10-02:
Is was at least every command I tried, which was open, listen, status detect and put  I was trying detect, but in actual usage just `cbmctrl put` and open for controlling it. On the PPC64 machine I can draw just fine with the plotter using just these two without issue, and all of these hang on the SUSE box. 

I did just try and if I have the printer disconnected, the first command does succeed and all subsequent fail. With it plugged in it fails every time, and never succeeds. 

Kernel was 5.10, USB 2 only on this machine. Will check the other details in a bit. I'm going to try with another distro on this machine, at least to figure out if it's the laptop or the distro's software. Thanks for the help here :)

thierer, 2021-10-03:
> It was at least every command I tried, which was open, listen, status detect and put I was trying detect, 

Don't use `status` and `detect` with the plotter. Both are meant to be used with floppy drives and might confuse the plotter (and, as a result, us).

> in actual usage just `cbmctrl put` and open for controlling it

Ok, that makes sense. Could you then please post a `XUM1541_DEBUG` log similar to the one in your first post for a `cbmctrl open` for the plotter after you plugged the Zoomfloppy in (I understand this will already hang?).

To have a clean state maybe also unplug the IEC cable from the Zoomfloppy before you plug it into the computer, then turn on all devices on the IEC bus (that would only be the plotter, right?) and only then connect the IEC cable to the ZF.

> I did just try and if I have the printer disconnected, the first command does succeed and all subsequent fail. With it plugged in it fails every time, and never succeeds.

Just to make sure I understand that correctly: With "fail" you strictly mean `cbmctrl` hangs, not just that the outcome of the command isn't what you'd expect (for example `cbmctrl` prints an error message)?

dawn-minion, 2021-10-04:
> Don't use status and detect with the plotter. Both are meant to be used with floppy drives and might confuse the plotter (and, as a result, us).

Noted, didn't know. It does return instantly on the other working machines, so I thought it was ok to test. 

> Just to make sure I understand that correctly: With "fail" you strictly mean cbmctrl hangs, not just that the outcome of the command isn't what you'd expect (for example cbmctrl prints an error message)?

Apologies for being vague there - I meant the command never returns, and the last debug line I get is `[XUM1541] xum1541_wait_status checking for status`

I've just tried another distro on this laptop (Manjaro), and have the same issue again, so it doesn't seem to be just openSUSE but just this laptop.

Here is the debug output for executing the command `XUM1541_DEBUG=99 cbmctrl put "bug"` (which spits out the literal word "bug" on the paper on the PPC64 system):
```
[XUM1541] scanning usb ...
[XUM1541] device 1d6b:0002
[XUM1541] device 0483:2016
[XUM1541] device 1d6b:0001
[XUM1541] device 0a5c:21e8
[XUM1541] device 1d6b:0001
[XUM1541] device 1d6b:0001
[XUM1541] device 16d0:0504
[XUM1541] found xu/xum1541 version 0208 on bus 2, device 6
[XUM1541] xum1541 name: xum1541 floppy adapter (ZOOMFLOPPY)
[XUM1541] xum1541 serial number:   0
[XUM1541] firmware version 8, library version 8
[XUM1541] device capabilities 1b status 02
[XUM1541] [xum1541_init] Tape supported, disk mode entered.
[XUM1541] firmware git revision is 00673ba6
[XUM1541] compiled with avr-gcc version 10.3.1 2
[XUM1541] and using avr-libc version 2.0.0
[XUM1541] write 16 3 bytes from address 0x56489ff24f60 flags 0
[XUM1541] wrote 3 bytes (42 75 67)
[XUM1541] xum1541_wait_status checking for status
```

This is after plugging in the ZoomFloppy, powering on the printer, letting it complete it's startup sequence, plugging in the DIN cable, and then executing the above command. The command at this point never continues beyond the "checking for status", and always prints this same output. Following the exact same sequence on the Windows and PPC64 box results in the command completing near instantly. 

If it helps, here's the system info:
Kernel - `5.13.19-2-MANJARO`
libusb - `1.0.24-2`
gcc - `11.1.0`
USB Controller - `Intel Corporation NM10/ICH7 Family USB UHCI Controller`
USB kernel logs:
```
[  511.475316] usb 2-2: new full-speed USB device number 6 using uhci_hcd
[  511.683443] usb 2-2: New USB device found, idVendor=16d0, idProduct=0504, bcdDevice= 2.08
[  511.683459] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  511.683466] usb 2-2: Product: xum1541 floppy adapter (ZOOMFLOPPY)
[  511.683471] usb 2-2: Manufacturer: Nate Lawson and OpenCBM team
[  511.683476] usb 2-2: SerialNumber: 000
```

Really quite baffled here, as every other USB device I use on this laptop is fine. I routinely use an ST-Link V3 which also is implemented using libusb and it's quite happy, along with USB to serial adapters. I have a cardbus card which has a number of USB ports on it, going to try if that makes any difference at all.

thierer, 2021-10-04:
Thanks for the detailed report!

> Here is the debug output for executing the command XUM1541_DEBUG=99 cbmctrl put "bug" (which spits out the literal word "bug" on the paper on the PPC64 system):
[...]
> This is after plugging in the ZoomFloppy, powering on the printer, letting it complete it's startup sequence, plugging in the DIN cable, and then executing the above command. The command at this [...]

I'm confused, because I wouldn't expect this to work? I think the `cbmctrl put` would have to be preceded by a `cbmctrl listen 6 0` (or a `cbmctrl open 6 0 ""`), otherwise the plotter wouldn't know that it's addressed? (This still does neither explain nor justify that it would hang, I'm just suspicious if this is really the result of the **first** command you issued. That's important, that's why I'm pedantic).

> If it helps, here's the system info:
[...]

I have a system with similar specs (chipset is ICH8, but also using uhci_hcd). And I just checked, it seems to work just fine, at least with a 1541-II. (As you haven't answered my question earlier: You don't happen to have access to a CBM floppy drive to test? Right now, I don't think it matters, but it would still help if we could rule out that it's a specific problem of communicating with the plotter).

One more question: Is there a difference in the state of the Zoomfloppy's activity LED after you plugged it into a machine where it works vs. the laptop where it doesn't (just after plugging it in and maybe giving it 10s to settle, not issuing any commands)?

Btw, if you haven't already (otherwise: my apologies), don't bother with my suggestion to test the firmware from revision eedaa3f. I realized that was wrong thinking on my part; you'd want to test with the **xum1541 plugin** from that revision, not the **firmware**. But no need to do that now, neither, let's first try to get a better understanding of the problem.

dawn-minion, 2021-10-06:
> I have a system with similar specs (chipset is ICH8, but also using uhci_hcd). And I just checked, it seems to work just fine, at least with a 1541-II. (As you haven't answered my question earlier: [...]

I do have a 1541 to try - running `cbmctrl detect` on the PPC64 host gives me this:

```
raven@TheBig970:~$ cbmctrl detect
 8: 1540 or 1541 
```
I can also format disks successfully using `cbmformat`:
```
raven@TheBig970:~$ cbmformat -s 8 "HELLO",0
00, OK,00,00
```

And on the Thinkpad laptop the exact same issue as before where it sits after printing "checking for status". The format command also hangs forever. 

So it seems it's not related to the device, as the behaviour is the same on both. 

> I'm confused, because I wouldn't expect this to work? I think the cbmctrl put would have to be preceded by a cbmctrl listen 6 0 (or a cbmctrl open 6 0 ""), otherwise the plotter wouldn't know that [...]

Ah you're right, I was indeed doing an "open" on the PPC64 host before. I retried just with `cbmctrl open 6 0 ""` on both, and it is the same result - instant return on PPC64, and hangs on "checking for status" on the Thinkpad. 

> One more question: Is there a difference in the state of the Zoomfloppy's activity LED after you plugged it into a machine where it works vs. the laptop where it doesn't (just after plugging it in and maybe giving it 10s to settle, not issuing any commands)?

No difference with the 1541 I can see. On both it lights up for a moment, the floppy drive resets, and then the light goes out after a second. Seems to be the same with the plotter too. 

Thanks again for your assistance here, it's really appreciated. I do have an AVR debugger here too, if it helps looking at the firmware side.

spiro-trikaliotis, 2021-10-07:
Wild guess: Have you tried another USB port? Do you have a USB hub? Can you measure the voltage the uC gets?
Have you tried adding another device to the IEC bus (so there are more pull-ups)?

BTW: cbmlinetester changes the IEC bus if you want to, one line after the other. So, you could measure the voltage of the lines one after the other.