OpenCBM/OpenCBM #89: imgcopy is broken

spiro-trikaliotis, 2021-10-06:
imgcopy has some serious issues:

1. the protocol s3 is not implemented in the xum1541. Thus, you should never use s3 protocol!
2. For 1581 drives, it uses original as transfer protocol, but sets warp mode. Thus, it tries to send the track map while using original transfer, which will result either in a segmentation fault, or in a hang. This is the problem in #87.
Work-around: Use --no-warp
3. When using a 1581 drive, if `-d 1581` is not specified, it will "think" it is writing to a D80 file, but the file extension is D81.  This way, imgcopy is totally confused and behaves erroneously
4. At least on my machine, serial1 does not work with a 1581 drive. imgcopy waits for the start-up of the drive code, and never gets beyond this point.

Thus, in order to use imgcopy to read from a 1581 drive as drive number 8, use:

`imgcopy --no-warp -ts2 -d 1581 8 test.d81`

To write the image back, use

`imgcopy --no-warp -ts2 -d 1581 test.d81 8`

Other commands will most likely fail!

spiro-trikaliotis, 2021-10-06:
TODO: Test for other drives, if they have the same issues (likely)

vddiabolic, 2021-11-24:
imgcopy 1.17 is returning the following message using "imgcopy -d 1581 9 test.d81" with ZoomFloppy:
[Fatal] drive 09 (CBM-1581): 73,(C) 1989 JIFFYDOS 6.0 1581,00,00

Code 73 is handled as an error in libimgcopy/imgcopy.c as it is not exempted. I do not think "JiffyDOS" is causing the issue in this case.

Could modifying imgcopy.c from "if(cnt)" to "if(cnt && cnt != 73)" fix the issue?