OpenCBM/OpenCBM #91: plugin-xa1541 build fails

ops, 2021-11-28:
Seems that kernel 5.14 broke Linux driver.

```
make -C /lib/modules/5.14.16-205.current/build here=`pwd` OPENCBM_KERNEL_FLAGS="" M=`pwd` modules
make[2]: Entering directory '/usr/src/linux-headers-5.14.16-205.current'
  CC [M]  /home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.o
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘timeout_us’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:414:14: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  414 |     current->state = TASK_INTERRUPTIBLE;
      |              ^~~~~
      |              __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘do_reset’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:485:14: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  485 |     current->state = TASK_INTERRUPTIBLE;
      |              ^~~~~
      |              __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘wait_for_listener’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:536:14: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  536 |     current->state = TASK_INTERRUPTIBLE;
      |              ^~~~~
      |              __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘cbm_read’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:565:26: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  565 |                 current->state = TASK_INTERRUPTIBLE;
      |                          ^~~~~
      |                          __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘cbm_raw_write’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:652:14: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  652 |     current->state = TASK_INTERRUPTIBLE;
      |              ^~~~~
      |              __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘cbm_unlocked_ioctl’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:791:26: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
  791 |                 current->state = TASK_INTERRUPTIBLE;
      |                          ^~~~~
      |                          __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘cbm_init’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:1154:14: error: ‘struct task_struct’ has no member named ‘state’; did you mean ‘__state’?
 1154 |     current->state = TASK_INTERRUPTIBLE;
      |              ^~~~~
      |              __state
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c: In function ‘cbm_unlocked_ioctl’:
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:789:32: warning: this statement may fall through [-Wimplicit-fallthrough=]
  789 |         while ((POLL() & mask) == state) {
      |                ~~~~~~~~~~~~~~~~^~~~~~~~
/home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.c:802:5: note: here
  802 |     case CBMCTRL_IEC_POLL:
      |     ^~~~
make[3]: *** [scripts/Makefile.build:271: /home/ops/projects/OpenCBM/opencbm/sys/linux/cbm_module.o] Error 1
make[2]: *** [Makefile:1858: /home/ops/projects/OpenCBM/opencbm/sys/linux] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.14.16-205.current'
make[1]: *** [LINUX/Makefile:49: cbm.o] Error 2
make[1]: Leaving directory '/home/ops/projects/OpenCBM/opencbm/sys/linux'
make: *** [LINUX/Makefile:79: BUILDSYSTEM.opencbm/sys/linux/.all] Error 2
```

spiro-trikaliotis, 2021-11-28:
Hello,
thank you for your report.
Please try the attached patch:
[xa1541-linux5.14.diff.gz](https://github.com/OpenCBM/OpenCBM/files/7614354/xa1541-linux5.14.diff.gz)

If it works, I will try to find out with which Kernel version we can / have to use it.

Please note that the warning regarding the "fall through" (lines 789, 802) is not relevant, because this is intentional (and is even marked with a "magic comment", but it seems gcc has changed its behaviour regarding this.

ops, 2021-11-29:
Thanks for the quick patch.

Quickly tested with d64copy and everything seems to be working!

commit 3665230b (2021-11-30): xa1541 linux: Use set_current_state() if available

Using current->state directly is deprecated since some time. The macro
set_current_state() was introduced in Kernel 2.2.18. Use that if it is
available.

Fixes #91
closed 2021-11-30 by spiro-trikaliotis