OpenCBM/OpenCBM #93: Cannot read a file from disk that is listed in dir

spiro-trikaliotis, 2022-02-23:
This is a bug report from SF (https://sourceforge.net/p/opencbm/bugs/74/):
If i list disk files using the command "cbmctrl dir 8" i obtain a file with the following name:
"a1.@@@@@@@@@@@@." and type prg.
If i try to read this file using following command:
cbmcopy -r -t original 8 'a1.@@@@@@@@@@@@.' -o TEST
i obtain this output:
[Info] reading a1.@@@@@@@@@@@@. -> TEST
[Info] identified a 1571 drive
[Fatal] could not open file for reading: 62, FILE NOT FOUND,00,00
[Warning] error reading A1.@@@@@@@@@@@@.

I'm using "opencbm-git" version "r1480.05c04338-1" on Arch Linux

spiro-trikaliotis, 2022-02-23:
Answer written on SF:

This happens because the cbmctrl command and cbmcopy do different mappings for tfie characters. This is a known limitation at the moment.

Is the file really named with at-signs (@)?

To make sure, can you create a copy of the disk as an image (d64 file)?

andreafrati1969, 2022-02-23:
andrea@archserver64 ~ $ cbmctrl dir 8
0 ."once upon a time" -rem-
88   "aliens+9&p  /rem" prg   
69   "a1.@@@@@@@@@@@@." prg   
31   "a2]   aliens   ]" prg   
31   "a3]   ......   ]" prg   
30   "a4]the computer]" prg   
24   "a5]    game    ]" prg   
24   "a6]            ]" prg   
22   "a7]  (c) 1986  ]" prg   
22   "a8] activision ]" prg   
20   "a9]incorporated]" prg   
20   "a0.@@@@@@@@@@@@." prg   
283 blocks free.             
00, ok,00,00

andrea@archserver64 ~ :( $ nibread -@xum1541:0 OneDrivePersonale/Commodore/disks/DISK43B.nib

nibread - Commodore 1541/1571 disk image nibbler
(C) Peter Rittwage
http://c64preservation.com
Built Nov 13 2021 08:06:23

* Using OpenCBM adapter xum1541:0

Drive Version: 73,CBM DOS V3.0 1571,00,00
Drive type: 1571
Bumping...
Initializing
Sending 1571 SRQ support code...
Uploading floppy-side code ($0454 bytes, $300-$754)...done.
Starting custom drive code...Started!
Testing communication...done.
Passed basic communication test.

18.0: (2) 
Cosmetic Disk ID: '-R'
Format Disk ID: '

 1.0: (3) 7694 [CBM OK] (weakgcr:17) 
 2.0: (3) 7690 [CBM OK] (weakgcr:22) 
 3.0: (3) 7711 [E2S4]
      (3) 7712 [E2S4]
      (3) 7712 [E2S4]
      (3) 7712 [E2S4]
      (3) 7713 [E2S4]
      (3) 7713 [E2S4]
      (3) 7713 [E2S4]
      (3) 7713 [E2S4]
      (3) 7714 [E2S4]
      (3) 7713 [E2S4]
      (3) 7713 [E2S4] (weakgcr:21) 
 4.0: (3) 7709 [E2S20]
      (3) 7710 [E2S20]
      (3) 7711 [E2S20]
      (3) 7710 [E2S20]
      (3) 7711 [E2S20]
      (3) 7711 [E2S20]
      (3) 7712 [E2S20]
      (3) 7709 [E2S20]
      (3) 7712 [E2S20]
      (3) 7717 [E2S20]
      (3) 7712 [E2S20] (weakgcr:18) 
 5.0: (3) 7689 [E2S13]
      (3) 7689 [E2S13]
      (3) 7689 [E2S13]
      (3) 7689 [E2S13]
      (3) 7689 [E2S13]
      (3) 7689 [E2S13]
      (3) 7689 [E2S13]
[... 213 lines cut ...]
Aligning tracks...
Writing G64 file...
G64 Track Length = 7928
Successfully saved G64 file

andrea@archserver64 ~ $ c1541 -a OneDrivePersonale/Commodore/disks/DISK43B.g64 -dir
0 "once upon a time" -rem-
88    "aliens+9&p  /rem"  prg 
69    "a1.............."  prg 
31    "a2.   aliens   ."  prg 
31    "a3.   ......   ."  prg 
30    "a4.the computer."  prg 
24    "a5.    game    ."  prg 
24    "a6.            ."  prg 
22    "a7.  (c) 1986  ."  prg 
22    "a8. activision ."  prg 
20    "a9.incorporated."  prg 
20    "a0.............."  prg 
283 blocks free.

283 blocks free.
andrea@archserver64 ~ $ c1541 -a OneDrivePersonale/Commodore/disks/DISK43B.g64 -read "a1.........." OneDrivePersonale/Commodore/disks/DISK43B/02
ERR = 62, FILE NOT FOUND, 00, 00
cannot read `a1..........' on unit 8
invalid filename

andreafrati1969, 2022-02-25:
[image]

andreafrati1969, 2022-02-25:
[image]

spiro-trikaliotis, 2022-02-25:
What I had asked was the disk image (.D64) of your disk, if this is possible.
Additionally, in order to create a D64, nibread is a little bit of overkill. d64copy should be enough (d64copy 8 disk43b.d64).

andreafrati1969, 2022-02-26:
[image]

andreafrati1969, 2022-02-28:
Using wildcard character i'm able to copy files