OpenCBM/OpenCBM #94: modprobe cbm hangs

wilhelmy, 2022-04-27:
I've recompiled OpenCBM after about 8 years, now on an entirely different Linux box, and I still have the issue that loading the module hangs indefinitely (except now I can press Ctrl+C to get out of modprobe). The module is loaded correctly and I can copy floppies.

Any ideas?

spiro-trikaliotis, 2022-04-27:
What kind of cable do you use? xa1541, xm1541, xu1541, xum1541, ...?

wilhelmy, 2022-04-28:
It's something I presume is either XA1541 or XM1541 (could be X1541 but I'd be surprised if it still works on this modern parallel port). I got it off of ebay. Does it matter whether or not the cable is attached while loading the module?

wilhelmy, 2022-04-29:
I think I waited 5s, but can't remember for sure (haven't rebooted since). Will update as soon as I do.
I noticed however that `cbmctrl reset` also hangs until I press Ctrl+C. Possibly related?

spiro-trikaliotis, 2022-04-29:
Yes, on startup, the same function is caked as with cbmctrl reset. 
Without a connected device, the PC can hang for 5 seconds.

spiro-trikaliotis, 2022-10-31:
Do you have any updates?

wilhelmy, 2022-11-01:
I think it ended up working in the end but I honestly can't remember anymore. I'll close this and reopen with more info if it turns out I was wrong. Thank you!

closed 2022-11-01 by wilhelmy