OpenCBM/OpenCBM #95: Segmentation fault on M1 Macbook

alaslums, 2022-06-28:
I built OpenCBM following the Mac OS X guide via Ports, with the portfile specifying version 0.4.99 revision 88 (although the git fetch method appears to pull master?). This built and installed successfully.

Unfortunately all the resulting binaries segmentation fault when attempting to access the device, with or without a device plugged in:

`
# cbmctrl detect
zsh: segmentation fault  cbmctrl detect
`

The xum1541cfg command does at least show some info:

`
xum1541cfg devinfo
finding device...
xum1541 device, model 5 (PROMICRO), firmware version 8
`

Here's output from lldb:

`
lldb cbmctrl detect
(lldb) target create "cbmctrl"
Current executable set to 'cbmctrl' (arm64).
(lldb) settings set -- target.run-args  "detect"
(lldb) run
Process 63110 launched: '/opt/local/bin/cbmctrl' (arm64)
Process 63110 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x000000019ab6e6c4 libsystem_c.dylib`strtoul_l + 88
libsystem_c.dylib`strtoul_l:
->  0x19ab6e6c4 <+88>:  ldrsb  w9, [x22], #0x1
    0x19ab6e6c8 <+92>:  and    x12, x9, #0xff
    0x19ab6e6cc <+96>:  tbnz   w9, #0x1f, 0x19ab6e6dc    ; <+112>
    0x19ab6e6d0 <+100>: add    x12, x10, x12, lsl #2
Target 0: (cbmctrl) stopped.
`

If you need any additional information to help troubleshoot this let me know.

alaslums, 2022-06-28:
Forgot to mention I'm on macOS Monterey 12.4 and MacPorts 2.7.2 with Apple clang version 13.1.6

spiro-trikaliotis, 2022-06-29:
Hello,
I am sorry, I am not a Mac guy, so I cannot help much.

However: Can you give me a link to the instructions and/or the portfile? 0.4.99.88 is quite old (really old!), and it used libusb0.

The current git head uses libusb1 instead. This might be related, but it does not have to.

Can you create a backtrace from the crash? On Linux, you can use gdb for this and use the "bt" command? This way, I could see more precisely where this happend.

Your output (which I cannot understand completely) suggests to me that there is something going wrong with calling a dynamic library, so this might be related to the libusb0 vs. libusb1 issue I mentioned above, or it might be related to the way the opencbm libraries (opencbm itself, or the plugin) are loaded.

As said, I am not sure if I read the output ccrrectly.

@all Is there some Mac guy who can help?

alaslums, 2022-07-09:
@spiro-trikaliotis Thanks for coming back to me and trying to help.

First off, ignore the version number - the Portfile is using git-head for downloading and compiling the source, and simply has an out of date version for the Portfile itself. You can see this in the source tree here: ./opencbm/ports/MacPorts/Portfile

So we are indeed using libusb1.

Unfortunately the Mac build ecosystem is all llvm/clang these days, which I'm not mega familiar with. I did install gdb11 from MacPorts and built OpenCBM with this successfully, but it segfaults as well. Unfortunately I couldn't get gdb on Mac to work, I would need to code-sign it for it to be able to function, which is a big rabbit hole I'd like to avoid.

However here is a backtrace from lldb, apologies for not including this previously:

```
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x26624c)
  * frame #0: 0x000000019ab6e6c4 libsystem_c.dylib`strtoul_l + 88
    frame #1: 0x00000001001e5180 libopencbm.0.dylib`initialize_plugin_configuration(plugin_name="xum1541", Plugin_information=0x00000001001f0450, handle_configuration=0x0000600000208000) at cbm.c:374:21
    frame #2: 0x00000001001e5690 libopencbm.0.dylib`initialize_plugin_pointer(Plugin_information=0x00000001001f0450, Adapter=0x0000000000000000) at cbm.c:552:21
    frame #3: 0x00000001001e5784 libopencbm.0.dylib`initialize_plugin(Adapter=0x0000000000000000) at cbm.c:595:17
    frame #4: 0x00000001001e5a0c libopencbm.0.dylib`cbm_driver_open_ex(HandleDevice=0x000000016fdff6b8, Adapter=0x0000000000000000) at cbm.c:810:13
    frame #5: 0x00000001000073d4 cbmctrl`main + 168
    frame #6: 0x000000010001d08c dyld`start + 520
```

cbm.c 374 is:

```
            tmpUL = strtoul(tmpData, &endPtr, 0);
```

The wider context of this is:

```
 361         counter = -1;
 362         while (configData[++counter].type != OPENCBM_PLUGIN_CONFIGURATION_DATA_TYPE_LASTENTRY) {
 363             configData[counter].isValid = 0;
 364
 365             char          *endPtr;
 366             unsigned long tmpUL;
 367             int           isUL;
 368             long          tmpL;
 369             int           isL;
 370
 371             error = opencbm_configuration_get_data(handle_configuration, plugin_name,
 372                     configData[counter].name, &tmpData);
 373
 374             tmpUL = strtoul(tmpData, &endPtr, 0);
 375             isUL = (*endPtr == 0);
 376
 377             tmpL = strtol(tmpData, &endPtr, 0);
 378             isL = (*endPtr == 0);
 379
 380             if (error) {
 381                 /* could not get the data, continue */
 382                 error = 0; /* this is not an error */
 383                 continue;
 384             }
```

I'm not the best C developer in the world, but given the if(error) handler below, is it possible that on Mac OS we get an illegal access and a segfault, but on Linux it just continues? So this code isn't Mac friendly? Just a guess.

Any ideas welcome.

alaslums, 2022-07-09:
Okay - I've got OpenCBM working! I noticed this code appeared in February, so I downloaded release 104 from Jan, and compiled it.

This doesn't segfault, and can talk to my 1541-II successfully:

```
alasdair@Alasdairs-MacBook-Air OpenCBM-0_4_99_104 % ~/opencbm-104/bin/cbmctrl reset
alasdair@Alasdairs-MacBook-Air OpenCBM-0_4_99_104 % ~/opencbm-104/bin/cbmctrl detect
 8: 1541-II
```

So I guess there's something Mac unfriendly about this commit: https://github.com/OpenCBM/OpenCBM/commit/cfc628d4daeec7c829bae6273737700dd1f97681

spiro-trikaliotis, 2022-07-10:
> So I guess there's something Mac unfriendly about this commit: [cfc628d](https://github.com/OpenCBM/OpenCBM/commit/cfc628d4daeec7c829bae6273737700dd1f97681)

Why do you think it is that patch? Commit 3476a1884c1 is the other good candidate.
Can you somehow check which one is the culprit? This should make it easier for me to debug.

spiro-trikaliotis, 2022-07-10:
Ouch... Forget it. The error handling should be *before* the data is used!
Can you test with this patch applied?

diff --git a/opencbm/lib/cbm.c b/opencbm/lib/cbm.c
index 7da3e6e3..c7bdfd5c 100644
--- a/opencbm/lib/cbm.c
+++ b/opencbm/lib/cbm.c
@@ -371,18 +371,18 @@ static int initialize_plugin_configuration(char *plugin_name, plugin_information
             error = opencbm_configuration_get_data(handle_configuration, plugin_name,
                     configData[counter].name, &tmpData);
 
-            tmpUL = strtoul(tmpData, &endPtr, 0);
-            isUL = (*endPtr == 0);
-
-            tmpL = strtol(tmpData, &endPtr, 0);
-            isL = (*endPtr == 0);
-
             if (error) {
                 /* could not get the data, continue */
                 error = 0; /* this is not an error */
                 continue;
             }
 
+            tmpUL = strtoul(tmpData, &endPtr, 0);
+            isUL = (*endPtr == 0);
+
+            tmpL = strtol(tmpData, &endPtr, 0);
+            isL = (*endPtr == 0);
+
             switch (configData[counter].type) {
                 case OPENCBM_PLUGIN_CONFIGURATION_DATA_TYPE_BOOLEAN:
                     if (  (arch_strcasecmp(tmpData, "false") == 0)

alaslums, 2022-07-10:
Yup that fixed it:

```
alasdair@Alasdairs-Air OpenCBM-git % ~/opencbm-patched/bin/cbmctrl reset
alasdair@Alasdairs-Air OpenCBM-git % ~/opencbm-patched/bin/cbmctrl detect
 8: 1541-II
```

:-)

commit af707913 (2022-07-10): Fix crash if no configuration data is available

initialize_plugin_configuration() called opencbm_configuration_get_data() and
used the result *before* testing for errors. This resulted in a crash when the call failed.

Fixed this by moving the error handling before using of the data.

This fixes #95.
closed 2022-07-10 by spiro-trikaliotis