OpenCBM/OpenCBM #99: XM1541 Linux + PCI Parallel port card - "DRIVE ERROR"

vintagepc, 2022-10-30:
Preamble - I've read the disclaimers this may or may not work, but hoping to get some input on my issue.

Issue:
`d64copy 8 test.d64` returns with an error:

- Built from source on OpenSuSE 15.3; cannot use OBS package as GLIBC is too new. 
- Kernel module seems to load OK, but has a long delay before modprobe returns. 
- XM1541 Cable behaves correctly according to cbmlinetester, though voltages are ~3.6v when a line is high, not 5v. 
- 1541C drive is known-good and works on a real C64. 
-

spiro-trikaliotis, 2022-10-30:
Unfortunately, the 99,DRIVER ERROR is very generic.
You say that the drive is identified with "cbmctrl detect"? This is good, that means that the interrupt is working. Normally, this is the biggest problem with PCI cards.

The 3.6V might be a problem, though. I am not sure if this is related to the XM1541 cable (which only has diodes). What type of diodes are you using (cf. http://sta.c64.org/xm1541c.html)?

Do you have the option to try with an XA1541 cable (with the recommended transistors BSV52 (!), cf. http://sta.c64.org/xa1541.html) instead?

vintagepc, 2022-10-30:
Thanks for the reply!

I'm using the recommended 1N5819 diodes. Unfortunately I'm not really equipped to do an SMT board for the -A design. 

Let me see if I can get pin value measurements without the diodes, that will answer whether the card is only driving ~3v vs if it is a problem with the cable construction :+1:

vintagepc, 2022-10-30:
I probed the pins, it is the card driving the lines at ~3.4v, so it's not due to the diodes or cable length.

spiro-trikaliotis, 2022-10-30:
The parallel port outputs act as a sink only. That is, if the parallel port is at high signal, then the voltage should be from the Commodore floppy side.

However: If the parallel port output is at 3V only when high, then it will act as sink, anyway. 3V + 0.6V (the voltage drop of the 1N5819 according to https://components101.com/diodes/1n5819-pinout-datasheet) gives exactly the 3.6V you are measuring on the Commodore side.

Thus: Yes, I suspect that your parallel port does not provide the necessary voltages here.

And 3.6V is a little bit low, especially when we take the timing behaviour into account.

vintagepc, 2022-10-30:
hmm, I should double check with the drive connected then, My tests were all done with the IEC end disconnected on the assumption the port was driving the signals.

spiro-trikaliotis, 2022-10-30:
I see you answered while I wrote my answer.

Well... One option might be to add 7406 open collector inverters instead; but then, you are far away from the normal XM1541 cable (in fact, it would act as an XA1541). Additionally, as far as I know, no one has ever tested this. Additionally, you would have to add some power supply for the 7406.

spiro-trikaliotis, 2022-10-30:
Oh: Yes, please check with connected drive! The pull-up resistors are inside of the drive.

vintagepc, 2022-10-30:
At the drive it is 5v with no cable connected, but drops to the observed 3.6 when connected.

vintagepc, 2022-11-05:
Shy of buying some actual BVS52s. I decided to try with some BC548s I already have in my stash; they have similarly low on voltages and seem "similar enough". 

Now the line voltages at the drive are ~4.6 volts, which seems much more reasonable. However, the same symptoms remain. 

DMESG output during the modprobe is as follows, which seems interesting:
```
cbm_init: using active (XA1541) cable, irq 16
cbm: resetting devices
cbm: waiting for free bus...
cbm: quitting because of timeout
```

Looking at the "Check if bus free" code... manually doing this with cbmlinetester seems to work correctly - setting ATN causes DATA to get set, and releasing it causes DATA to also get released by the drive. Which smells like there might be a timing issue here. I can get the bus free check to pass with the "hold_clk=0" module argument. (but this makes no difference in attempts to use d64copy)

vintagepc, 2022-11-23:
Thanks - I'll keep poking at things and let you know; thanks for the link on the transistors - I'll see if I have any of those in my parts stash.

vintagepc, 2022-11-27:
I made a little breadboard to make it easy for me to play with various transistors I have on hand and it was interesting to see how some would report complete failures on `cbmctrl status` and others would work there but fail on the `dir` step. 

I found a cheap source of some 2N7000s so have ordered a few of those just to try it out - though at this point it's mostly just curiosity to see if it will work at all and add a data point for this particular PCI parallel port card.

vintagepc, 2022-12-01:
Progress! with the 2N7000s, `cbmctrl dir 9` will actually give directory listing sometimes!

```
cbmctrl dir 9
0 ."test/demo  1/85 " 84 2a
14   "how to use"       prg   
8    "how part two"     prg  
7    "how part three"   prg  
4    "vic-20 wedge"     prg  
1    "c-64 wedge"       prg  
4    "dos 5.1"          prg  
9    "printer test"     prg  
6    "disk addr change" prg  
12   "view bam"         prg   
15   "display t&s"      prg   
4    "check disk"       prg  
11   "performance test" prg   
5    "seq.file.demo"    prg  
18   "rel.file.demo"    prg   
7    "sd.backup.c16"    prg  
7    "sd.backup.plus4"  prg  
10   "sd.backup.c64"    prg   
7    "print.64.util"    prg  
7    "print.c16.util"   prg  
7    "print.+4.util"    prg  
13   "uni-copy"         prg   
30   "c64 basic demo"   prg   
35   "+4 basic demo"    prg   
8    "load address"     prg  
7    "unscratch"        prg  
5    "header change"    prg  
403 blocks free.             
00, ok,00,00
```

It still fails sporadically with "driver error". However, I suspect that this is because I have just dropped them in as a direct replacement for the transistors - the base transistor probably needs to be removed (this can add signal delay) and possibly replaced with a pulldown.

Overall this looks rather promising that it *should* be possible to get this working with this PCI card.

vintagepc, 2022-12-01:
Success - I found the mailing list message you referenced and it noted no gate resistors were needed, so I removed them. Sometimes it still hiccups but once it starts it seems to complete just fine (dir, d64copy). 

```
d64copy 9 1541.d64 -v -v -v
[Debug] transfer mode is 0
[Debug] decided to use transfer mode 3
[Info] Trying to identify drive type
[Info] drive 09 (1541): 00, OK,00,00
[Warning] growing image file to 683 blocks
[Info] copying tracks 1-35 (683 sectors)
 1: *********************                  
 2: *********************                  
 3: *********************                  
 4: *********************                  
 5: *********************                  
 6: *********************                  
 7: *********************                  
 8: *********************                  
 9: *********************                  
10: *********************                  
11: *********************                  
12: *********************                  
13: *********************                  
14: *********************                  
15: *********************                  
16: *********************                  
17: *********************                  
18: *******************                    
19: *******************                    
20: *******************                    
21: *******************                    
22: *******************                    
23: *******************                    
24: *******************                    
25: ******************                     
26: ******************                     
27: ******************                     
28: ******************                     
29: ******************                     
30: ******************                     
31: *****************                      
32: *****************                      
33: *****************                      
34: *****************                      
35: *****************       100%   683/683
683 blocks copied.

```

spiro-trikaliotis, 2022-12-03:
> When it fails, the failure is immediate, almost like it's not waiting quite long enough for the drive to wake up and respond; if I retry right away it starts without a problem. It's entirely possible this is just due to my prototype wiring and this issue will go away once I actually solder it up permanently into a shielded plug.

On start of d64copy, it uses the CBM IEC protocol to upload drive code into the floppy. The drive code is rather picky and time sensitive.

When the transfer starts, it uses its own protocol which is implemented by the code that was just uploaded. This protocol is much less picky, that's why it will fail almost always on startup only.

A good test if the IEC protocol works reliable it the command

`cbmctrl download 8 0xc000 0x4000 rom1541.bin`

which reads out the ROM image of the 1541 in drive 8 into a file named rom1541.bin. After this command (which takes some minutes!), you should get a file of exactly 16384 byte. If you retry this command multiple times and always get a file of this length, and the file is always identical, then the IEC protocol transfer from floppy to PC works.

vintagepc, 2022-12-06:
Thanks for the follow-up. I ran some tests with the command you suggest and most of the time it does work, successfully writing the 16k file with the same md5 checksum. Though it does indeed sporadically fail partway through  on occasion, 8 times out of a total of 31 attempts.