OpenCBM/OpenCBM #102: OpenCBM won't detect 1581 on IEC bus

Schwefelholz, 2023-03-27:
I have a problem with a physical 1581 which won't be detected and can't be addressed on the IEC bus. The drive's ROM is 312558-01 (which seems to be identical to 318045-01.
No matter where I place it in the daisy chain (before or after a working 1571), the drive won't be detected.

Connected to a Commodore computer, the same setup is working as expected.

Interestingly, a standalone used Ultimate II+ cartridge pretending to be a 1581 is detected by OpenCBM, no matter which ROM I configure it with: 318045-01, 318045-02 or JiffyDOS.

Is there anything else I could try to get the drive working with OpenCBM?

spiro-trikaliotis, 2023-03-27:
Are you using an xum1541 device or an xu1541 device? If it is an xum1541 device, does it have an active driver for the IC lines (that is, a 7406N)?

Schwefelholz, 2023-04-03:
Yesterday, I had the chance to try this again with another 1581 with the same outcome. `cbmctrl detect` was not able to see that device on the IEC bus.
I'm really wondering why there is a difference between detecting physical drives and the FPGA-based Ultimate-II+ pretending to be a 1581.
Is there any way to see what's happening on the bus?

Tiny69, 2024-03-16:
before this post turns 1 I'd like to add my UX,
just yesterday I put together a XUM1541 ProMicro on a breadboard and connected it to a 1581 (original, ROM -01),
also complied the master-branch of this repo, 
then this is the result:

```
~ $ cbmctrl -V
cbmctrl version 0.4.99.104, built on Mar 16 2024 at 00:16:23

~ $ cbmctrl detect
 8: 1581

~ $ cbmctrl status 8
73,copyright cbm dos v10 1581,00,00

```

spiro-trikaliotis, 2024-04-21:
This might be a duplicate of #111. Can you check:

1. try the command "cbmlinetester.exe --SRQ" before trying to access the 1581. Does it work?
2. if 1. worked, does the experimental firmware given in #111 fix the problem?

spiro-trikaliotis, 2024-05-07:
Duplicate of #111

closed 2025-06-19 by spiro-trikaliotis