OpenCBM/OpenCBM #103: cbmformat --verify runs "endless"

schorsch3000, 2023-08-01:
`cbmformat` (and all other tested tools from opencbm) seems to work fine, but running `cbmformat` with `--verify` runs endlessly.

My setup:
opencbm 0.4.99.104 from debian 12
xum1541
1541-II with Jiffydos
running for example 
`cbmformat -sopcv 10 test,33`
prints 
`###################################` 
in a normal timeline
prints a new line and takes at least 3 hours while spinning the disk.

schorsch3000, 2023-08-04:
this semms to happen only with bad disks.
i'm using cmbforng now, that seems to work better

spiro-trikaliotis, 2023-08-08:
Unfortunately, this is a known restriction (some wold call it a bug) of cbmformat: If the verify fails, it tries the format over and over again, and the verify will always fail.

Note that this can be an advantage: With disks that are "a little bit" problematic, trying to format again and again might allow for the format to succeed at some point of time.

I will think about a proper solution. I see the following options:
1. As default, stop formatting with an error code, and having a special option which allows for continuously formatting, or
2. at least, give the user a warning that there is a problem, that cbmformat tries to format the disk anyway, and if it takes too long, the user should abort the process with Ctrl+C / Ctrl+Break

It might be possible to combine 1. and 2.