OpenCBM/OpenCBM #105: zoomfloppy resets when using cbm_srq_burst_read

michielboland, 2023-10-11:
Hi. I was trying to play around with bulk transfer with a 1571 and a zoomfloppy, but I cannot get it to work.
For some context, the ultimate goal is to image MFM disks, since nibtools does not appear to be able to do that, and my PC floppy drive died.

sample code:

[test.c.txt](https://github.com/OpenCBM/OpenCBM/files/12874595/test.c.txt)

if I run this I get

```$ ./a.out 
USB error in xum1541_wait_status: LIBUSB_ERROR_NO_DEVICE
```
and a message in syslog that indicates that the USB device has reset, apparently.

Maybe I am going about it all wrong, and the API is not supposed to be used in this way. (Nibtools does some magic with loading custom code on the 1571.)

Cheers
Michiel

spiro-trikaliotis, 2023-10-12:
Unfortunately, yes, you get it wrong.

I have to admit that the naming is not very good. The SRQ burst routines are meant for the nibtools only. They use a special protocol, so, you cannot use them with the U0 commands of the floppy drive.

The "normal" SRQ burst routines are not implemented with OpenCBM, nor with the xum1541 yet. I am not sure if it is worth to try to implement it, though, as the only benefit would be the ability to read MFM disks - something which can also be achieved by programming the floppy drive differently and using the "normal" transfer routines.

Regarding your target of reading MFM disks: You are aware that the 1571 drive will not be able to read HD disks (i.e., 1.2MB)? It could only read DD disks (360 KB, 320 KB, 180 KB, 160 KB) with proper programing.

michielboland, 2023-10-12:
Thanks, I figured it was too good to be true. :)
I'll try to see if I can read the disks some other way.
The diskettes I have here all say DS-DD on the label (ie not HD), so they should be readable.

Cheers
Michiel