OpenCBM/OpenCBM #106: cmbcopy can’t specify drive for dual disk models (8050)

root42, 2023-11-19:
The cbmcopy tool uses a random drive in CBM 8050 and similar models. The files get copied fine, but sometimes end up on drive 0, sometimes on drive 1.

spiro-trikaliotis, 2023-11-20:
Just some wild guess: Can it be that it uses the last drive used?
Does it help if you prepend the drive number before the filename ("0:" and "1:")?

root42, 2023-11-20:
That could be the case, I will try that tonight.

For reading the "0:" or "1:" might help, will try that, too. However I am especially interested in writing files, where this won't work...

root42, 2023-11-20:
I just tried by running `cbmctrl dir 8 '1:'` before the cbmcopy command, but it still wrote to unit 0. Not sure how that is decided. Let me see if I can get it to write to drive unit 1 again.

root42, 2023-11-20:
Nope, it seems I can only write to Unit 0. Seems I was confused. So maybe we should change the title to "cbmcopy can't write to drive unit 1" or similar.

root42, 2023-11-20:
Ok, I have to revert my comment AGAIN. Now it DID write to unit 1. This is VERY confusing. There is something wrong with cbmcopy and dual drive units...

root42, 2023-11-20:
A bit more detail. I now did the following commands:
```
$ cbmctrl dir 8 1                      
1 ."drive0          " yy 2c
2052 blocks free.             
00, ok,00,00,1
```
This was successful and shows floppy disk named "drive0" (because it was formatted in drive 0) in unit 1, and it's empty.

Then copy one file using this command:

```
$ cbmcopy -w 8 volcano.prg -o '1:volcano,p'
[Info] writing volcano.prg -> 1 volcano p,p,w
[Info] identified a 8250 dos2.7 drive
................/
[Info] 00, OK,00,00,1
```

I thought maybe I can specify the unit in the output file name, as well as the file type. Well, that mangled the filename a bit.

Please note that THE LED ON UNIT 0 lit during the write! And NOT the LED in Unit 1!!!

Let's have a look at unit 1:

```
$ cbmctrl dir 8 1                          
1 ."drive0          " yy 2c
62   "1 volcano p"      prg   
1990 blocks free.             
00, ok,00,00,1
```

Even though drive unit 0 lit up the file was written to drive unit 1! I would say there is some weird communication happening between OpenCBM's cbmcopy and the 8050 drive...

root42, 2023-11-20:
PS: When I insert only one diskette, the usage gets better. The drive will bump the head of the empty unit once, but after that copying works reasonably well. Until a patch can be made for unit selection I will use this as a workaround.

root42, 2023-11-20:
PPS: The "1:" trick when reading works, but with limitations! The maximum file name length still applies. For example I had to run this command to copy a file named "thermochemistry,p" from unit 1:
```
$ cbmcopy -r 8 "1:thermochemis*" -o "thermochemistry.prg"
[Info] reading 1:thermochemis* -> thermochemistry.prg
[Info] identified a 8250 dos2.7 drive
........\[Info] 00, OK,00,00,1
```

Without the asterisk the y would be cut off:

```
$ cbmcopy -r 8 "1:thermochemistry"
[Info] reading 1:thermochemistry -> 1:thermochemistry.prg
[Info] identified a 8250 dos2.7 drive
[Fatal] could not open file for reading: 62, FILE NOT FOUND,00,00,1
[Warning] error reading 1:THERMOCHEMISTR
```

thierer, 2024-05-10:
[That's my fault](https://github.com/OpenCBM/OpenCBM/blame/db80d1be041c0683973a4a704a1f2d8e744c4e24/opencbm/cbmcopy/main.c#L504) (at least the write case). I didn't think of that use case, my apologies! 

I'll prepare a fix and open a PR.

closed 2024-05-25 by spiro-trikaliotis

root42, 2024-06-07:
Perfect! Thank you for your work!

PR #112 (2024-05-25): cbmcopy: Fixes for remote filename handling
These two commits should fix #106 and #107.

6cf5a832 limits filtering for the remote filename (introduced in 2937f54b) to cases when the filename is auto-generated from the local filename.

7a13a217 removes the 16 byte filename limit for both read and write except for the case when the remote filename is auto-generated for writing. I don't see why limiting the filename length would be needed, because as far as I can tell CBM DOS ignores extra filename characters anyway. And it hurts in the cases as discussed in #106 and #107 and even more so when wo [... cut]