OpenCBM/OpenCBM #107: cbmcopy fails to read SEQ files from 8050 drive

root42, 2023-11-27:
I found another issue with my 8050 drive. When trying to read a SEQ file, cbmcopy will fail. PRG files can be copied off of the floppy fine, though:
```
$ cbmcopy -r 8 'game-pet'     
[Info] reading game-pet -> game-pet.prg
[Info] identified a 8250 dos2.7 drive
.......-[Info] 00, OK,00,00,0
$ cbmcopy -r 8 'z3 story'
[Info] reading z3 story -> z3 story.prg
[Info] identified a 8250 dos2.7 drive
[Fatal] could not open file for reading: 64, FILE TYPE MISMATCH,00,00,0
[Warning] error reading Z3 STORY
$ cbmcopy -r 8 -f s 'z3 story'
[Warning] --file-type ignored
[Info] reading z3 story -> z3 story.prg
[Info] identified a 8250 dos2.7 drive
[Fatal] could not open file for reading: 64, FILE TYPE MISMATCH,00,00,0
[Warning] error reading Z3 STORY
$ cbmctrl dir 8 0
0 ."hibernated1     " hi 2c
359  "z3 story"         seq    
28   "game-pet"         prg   
35   "game-vic"         prg   
1    "config"           prg  
1629 blocks free.             
00, ok,00,00,0
```

spiro-trikaliotis, 2023-11-28:
Untested, but please try:
`cbmcopy -r 8  'z3 story,s'`

root42, 2023-11-28:
That seems to be a workaround. Probably will hit the 16 character limit eventually, though, right?
```
cbmcopy -r 8 'z3 story,s'   
[Info] reading z3 story -> z3 story.seq
[Info] identified a 8250 dos2.7 drive
........................................................................................../[Info] 00, OK,00,00,0
```