OpenCBM/OpenCBM #108: Zoom Floppy - Commands only work once

mase76, 2023-12-12:
Hi,
I use a Zoom Floppy device on my Debian unstable System with Kernel  6.5.0. I have connected a 1541 floppy drive.
However I can only place a command once. When I want to place a second command, the device does not react. No error message visible. The LED on the ZF turns on after the operation finished.
When I place a command before the LED turns on, it works as expected. But if I wait a bit too long until the LED turns on again, there is no reaction.. Even when replugging the USB cable, I have to fire a command before the LED turns on. I built the latest Master branch.
The issue is similar to this one:
https://groups.google.com/g/zoomfloppy-users/c/Z3cSZgO5wV4/m/uC1O-szyAAAJ

spiro-trikaliotis, 2023-12-12:
Can you give the version of the xum1541 device (the output of "xum1541cfg devinfo")?
Can you set XUM1541_DEBUG=9 and run the commands again, and give us the output?
Do you have a possibility to create a video (with the device and the screen visible), so we can see what is happening exactly?

mase76, 2023-12-12:
xum1541 device, model 2 (ZOOMFLOPPY), firmware version 8
cbmctrl reset works all the time. But after this, the LED turns on again and no more command is possible.
This is the debug output:

[XUM1541] scanning usb ...
[XUM1541] device 0bda:0316
[XUM1541] device 1d6b:0003
[XUM1541] device 5986:2113
[XUM1541] device 8087:0a2b
[XUM1541] device 5986:1141
[XUM1541] device 058f:9540
[XUM1541] device 16d0:0504
[XUM1541] found xu/xum1541 version 0208 on bus 1, device 30
[XUM1541] xum1541 name: xum1541 floppy adapter (ZOOMFLOPPY)
[XUM1541] xum1541 serial number:   0
[XUM1541] firmware version 8, library version 8
[XUM1541] device capabilities 1b status 03
previous command was interrupted, resetting
[XUM1541] [xum1541_init] Tape supported, disk mode entered.
[XUM1541] firmware git revision is 3ef4fc0d
[XUM1541] compiled with avr-gcc version 5.4.0
[XUM1541] and using avr-libc version 2.0.0
[XUM1541] write 16 2 bytes from address 0x7ffe4f1c872e flags 2
[XUM1541] wrote 2 bytes (28 6f)
[XUM1541] xum1541_wait_status checking for status

I will try do record a video.

mase76, 2023-12-12:
I have uploaded a video. I hope you can see what you need. After the LED turned on, the device hangs. When I fire a command before the LED turns on, it works, even when I fire a second command immediately.
[URL=https://www.file-upload.net/download-15237507/VID_20231212_211659.mp4.html]VID_20231212_211659.mp4[/URL]

spiro-trikaliotis, 2023-12-16:
Thank you for the input. I see and understand the problem now, but I cannot explain it yet.

The LED going on means that either the ZF is going in an error state, or it gets some information from USB that is not complete.

The debugging output is from the command that did not work? Or is it from the command that did work, before the next one failed?

@thierer: Do you have any idea? Would it make sense to ask for an USB trace, so we know what is going on on the bus?

thierer, 2023-12-16:
I didn't watch the video, but if "LED turns on" means it turns on solid (not blinking), this would suggest that a "device disconnect" event is triggered for some reason (no idea why). A reset caused by brown-out detection might also be the reason, but I *think* the device should recover from that.

@mase76 To test my theory I built a firmware (attached) with some of the "device disconnect" code removed. Could you please flash it and report back if the problem persists?

[xum1541-ZOOMFLOPPY-v08-7da18934.zip](https://github.com/OpenCBM/OpenCBM/files/13694016/xum1541-ZOOMFLOPPY-v08-7da18934.zip)

spiro-trikaliotis, 2023-12-16:
> I didn't watch the video, but if "LED turns on" means it turns on solid (not blinking)

Exactly that.

After the command, there is a time window of 1 or 2 seconds. In this time frame, a new command can be send.
If not, then the LED turns on "solid", and nothing works anymore until the USB cable is disconnected.

mase76, 2023-12-16:
I flashed the modified firmware and the problem disappeared.

thierer, 2023-12-16:
Thanks for the feedback! Could you please also test the two firmwares from the attached zip-File? aea7e768 is a different approach, which might well not work, but if it does might give an indication what's going on. 11ce3435 is just a better version of what I did for the 7da18934 firmware and should work (and if it does, you should prefer it over 7da18934).

[xum1541-ZOOMFLOPPY-v08.zip](https://github.com/OpenCBM/OpenCBM/files/13694161/xum1541-ZOOMFLOPPY-v08.zip)

mase76, 2023-12-16:
aea7e768 has the issue with the solid LED again. 11ce3435 works fine.

thierer, 2023-12-17:
> aea7e768 has the issue with the solid LED again.

Thanks, that was somewhat expected. Could you please also test the attached firmware? If my current theory is correct, I would expect the LED to still turn on after ~2s, but it should turn off again when you issue a command and the command should work. After the command finished, the LED would probably turn on again.

[xum1541-ZOOMFLOPPY-v08-4156def4.zip](https://github.com/OpenCBM/OpenCBM/files/13695935/xum1541-ZOOMFLOPPY-v08-4156def4.zip)

mase76, 2023-12-17:
Exactly as you described. The LED turns on solid but commands work.

thierer, 2023-12-17:
Ok, thanks for the confirmation! I'll have to discuss with Spiro how exactly we want to handle this, but for now I'll say you can keep using the 11ce3435 firmware, the final solution will probably be similar (or 4156def4, if you don't mind the LED turning on).

In case you didn't keep it (or others running into the same problem might need it) I'll attach it again. I removed it above, as I don't want random known non-working firmwares hanging around.

[xum1541-ZOOMFLOPPY-v08-11ce3435.zip](https://github.com/OpenCBM/OpenCBM/files/13697578/xum1541-ZOOMFLOPPY-v08-11ce3435.zip)

closed 2024-06-26 by spiro-trikaliotis