OpenCBM/OpenCBM #111: XUM1581 won't interact with 1581 unless SRQ line is unset.

SteveGuidi, 2024-04-18:
Hi there!  I recently built a XUM1541 with a 7406 and tested it with a 1581, which I also built. After some tinkering, I am able to use the drive through the XUM1541, but only after manually unsettling the SRQ line.  I'm not sure if this is normal, but I thought I would raise the issue at it seems unusual.

When the 1581 starts up, the SRQ pin (1) is held high at just under 5V.  This happens with both the stock and JiffyDOS ROMs.  The drive works fine when used with a CBM machine, but almost all commands are rejected by the XUM1541 until the SRQ line is unset (drive reset works).

I read that the SRQ line is used to request burst mode from a C128.  Could this be the issue?  I suppose the same problem would occur with a 1571 too, but unfortunately I don't have that drive to confirm.

For reference, I'm using the V8 firmware for the 7406 pro micro variant, currently from the master branch of the repo.

spiro-trikaliotis, 2024-04-19:
Hello Steve,
can you tell how you "unset" the SRQ line? Do you use a command? Do you cut the trace?
And: Do I understand it correctly? If you say "unset" the SRQ line, you mean grounding it?

SteveGuidi, 2024-04-19:
Hi @spiro-trikaliotis -- sorry for the confusion!

When I was testing my implementation, I was using the OpenCBM `cbmlinetester` command to poll and set the logic level of the IEC/serial lines.

Here is the output of `cbmlinetester` for the XUM1541 (7406) with the 1581 turned off.

```
PS C:\Program Files\opencbm> .\cbmlinetester.exe --poll
line status: 0017:       ATN   DATA  CLOCK SRQ
```

Here is the same command output after the 1581 is first turned on.

```
PS C:\Program Files\opencbm> .\cbmlinetester.exe --poll
line status: 0010:                         SRQ
```

To "unset" the SRQ line, I used the `cbmlinetester` command as follows.

```
PS C:\Program Files\opencbm> .\cbmlinetester.exe --SRQ
PS C:\Program Files\opencbm> .\cbmlinetester.exe --poll
line status: 0000:
```

spiro-trikaliotis, 2024-04-19:
Hi Steve,

ok, understood. So, after a RESET, SRQ is *low*, because the lines are low-active. That's why I asked.

The SRQ line is not initialized on RESET. This is problematic for some xum1541 devices, but not for all. That's why noone noticed yet.

Can you apply this patched firmware file to your xum1541 and test again?

[xum1541-PROMICRO_7406-v00.hex.gz](https://github.com/OpenCBM/OpenCBM/files/15045268/xum1541-PROMICRO_7406-v00.hex.gz)

Don't  be fooled by the version number "00"; this marks this as a test version, so OpenCBM accepts it, but we can see in logs that this is not an official released versoin.

The diff to the current git head is attached, too:
[xum1541-init-srq.diff.gz](https://github.com/OpenCBM/OpenCBM/files/15045272/xum1541-init-srq.diff.gz)

SteveGuidi, 2024-04-19:
@spiro-trikaliotis Fantastic -- thank you for the quick response and patch.  Everything works well now!

```
PS C:\Program Files\opencbm> .\cbmlinetester.exe --poll
xum1541 firmware version 0 is a TEST VERSION!
There is no support for this version!
PLEASE UPDATE to the officially supported firmware 8.

line status: 0017:       ATN   DATA  CLOCK SRQ
PS C:\Program Files\opencbm> .\cbmlinetester.exe --poll
xum1541 firmware version 0 is a TEST VERSION!
There is no support for this version!
PLEASE UPDATE to the officially supported firmware 8.

line status: 0000:
PS C:\Program Files\opencbm> .\cbmctrl.exe status 8
xum1541 firmware version 0 is a TEST VERSION!
There is no support for this version!
PLEASE UPDATE to the officially supported firmware 8.

73,(c) 1989 jiffydos 6.0 1581,00,00
```

spiro-trikaliotis, 2024-04-20:
Ok, this is good.
I will discuss this patch with others. If all agree, it will be part of the next firmware release.

SteveGuidi, 2024-04-20:
Thank you! 

For what it's worth, I've asked the members of my local C= club to test the voltage of pin 1 (SRQ) when the drive is first powered on.  We've confirmed that a 1571 also keeps the line at 5V, but don't have any data yet for a 1541 and 1541-II.

The hypothesis is that the startup state of SRQ might be related to burst mode and thus only the 1571 and 1581 are impacted.

SteveGuidi, 2024-05-14:
Confirming that on the 1541 and 1541-II, the SRQ pin on the serial port value has an opposite startup value than on the 1571 and 1581.  This is confirmed by the linked schematic drawings: SRQ on the [1541/1541-II](https://www.zimmers.net/anonftp/pub/cbm/schematics/drives/new/1541/1541-II.340503.gif) is pulled via a 1K resistor, where-as on the [1571](https://www.zimmers.net/anonftp/pub/cbm/schematics/drives/new/1571/310420-full.gif) and [1581](https://www.zimmers.net/anonftp/pub/cbm/schematics/drives/new/1581/1581-17.gif), SRQ is pulled down to GND.

closed 2024-06-11 by spiro-trikaliotis

mvbhz, 2024-12-14:
Hello. Will the XUM1541-II have the same problem? It won't work with the Commodore 1581 drive. The cbmctrl detect command doesn't show anything.

spiro-trikaliotis, 2024-12-14:
As there is no (official) XUM1541-II device, it is hard to tell how that one behaves. However, it is very likely it has the same issue if it is based on the official firmware, and it did not get any firmware update yet.

Please contact the one where you got that device and ask for a firmware update. Of course, you could try one of the official ones, but as I do not know the device, I cannot tell if any of them will work, and if yes, which one.

mvbhz, 2024-12-15:
https://github.com/tebl/C64-XUM1541-II

spiro-trikaliotis, 2024-12-15:
Cool...
https://github.com/tebl/C64-XUM1541-II/tree/main/software/firmware

He himself says that he does not have any interest in supporting this. Instead, he links to an outdated version with known issues.

Why should we support his work?

His documentation is also outdated, btw.

Again, ask him and not us on support. I do not feel as spending my time just finding out what others have done in order to find out if my advise helps there a good way to spend my time, I am sorry for that.

commit 7ebe4e47 (2024-04-20): Init SRQ on reset.

This fixes #111, issues with a 1581 with xum1541 implementations where
the serial outputs are inverted. There, the SRQ is active, that is, held
low, which results in a non-working 1581.
commit a263290d (2024-06-11): Release SRQ on init and reset of xum1541

This fixes #111.