OpenCBM/OpenCBM #115: d64copy appends error map to .d64 when -b or -B flags are used

movvny, 2024-08-18:
When I use `--bam-only` or `--bam-save` flag, `d64copy` always appends an error map to the created `.d64` image file, even when there was no errors. It does it even with `--error-map=on_error` flag. The disk image size is 175351 bytes.
When dumping the very same disk without `--bam-only` or `--bam-save` flags, and no errors are found, **no** error map is appended to `.d64` image, which is the expected behaviour based on [the documentaion](https://opencbm.trikaliotis.net/opencbm-21.html). The disk image size is 174848 bytes.
Could you please look into it?

spiro-trikaliotis, 2024-08-18:
I can confirm this behaviour: If the disk is not completely filled, the error states for blocks that are skipped because they are not marked in the BAM is left as "0". However, the error state "1" is without errors.

So, d64copy assumes an error occurred because it gets error states of "0", so it writes the error map.

I will make a fix for this.

movvny, 2024-08-18:
Thank you for a quick response!
It's strange nobody reported this earlier. I've only been using this software for a couple of days do dump my C64 "backup disks" that were sitting in boxes for years.

spiro-trikaliotis, 2024-08-18:
Actually, I don't expect many people use the --bam* options. If the BAM is corrupt, you might leave out essential information that you might need to fix your disk image.

spiro-trikaliotis, 2024-08-18:
This patch would fix the problem.

However, I just wrote bug #116. I am not sure if it is good to actually fix this issue.

Are there any downsides of the error information? Any problems that you encounter?

[d64copy-errormap.patch.gz](https://github.com/user-attachments/files/16648902/d64copy-errormap.patch.gz)

movvny, 2024-08-18:
Well, I just found this behaviour inconsistent with the situation when no `-b` or `-B` flags are used.
Also, why append an **error** map, when there where no actual errors in the used, read sectors?
Besides, I compare all my dumps against TOSEC database and against duplicates in my dump directories. Both methods use CRC32 or MD5, and of course `.d64` image with appended error map containing only `0`s and `1`s results in different CRC32/MD5 than a standard 174848-byte image.
I admit, I have little knowledge of C64/1541/d64 disks/images internals. I do have much more experience in Amiga `.adf`, Atari ST `.st` and Atari 8-bit `.atr` disk imaging, so maybe I am misplacing these other platforms experience to C64. I do have intention to learn though :) while dumping C64 disks I collected from different sources in the recent years.
Anyway, thanks a lot for a quick patch. Let's wait and see other's opinions on the subject in #116.