OpenCBM/OpenCBM #116: d64copy writes error maps with "0" as error code

spiro-trikaliotis, 2024-08-18:
While fixing #115, it was found out that d64copy, imgcopy and d82copy write out image files that contain error codes of "0" if an error map is to be written.

The error code "0" is written if a block was not even transferred because it is not marked in the BAM as used, but an error occurred somewhere on the disk.

The error code of "0" is not even defined for .d64 files, because the drives (2031, 1540, 1541, 1570, 1571) do not even use that error code. Success is an "1".

One might argue that converting the "0" to "1" might be a good idea before writing out the error map. However, this "0" also allows to find out which block were skipped when creating the image. There might be a benefit in having this information.

So, for the time being, I am not sure which way to go.

This is open for discussion!

movvny, 2024-08-18:
How about adding another flag or another value to `--error-map` flag to control the behaviour while doing BAM based image?

UffeJakobsen, 2024-08-19:
Just my few cents:

As Spiro mentioned - only dumping the allocated bam entries can fail in so many ways

A defective bam will prevent you from obtaining a complete dump of the disk - short-cutting sort of defeats the whole purpose of preserving the disk - use the few extra seconds to get a full copy of that disk :-)

Another trivial example - most of the full-side (trackloader) demos that are released nowadays will not be properly dumped using bam-only

The (Disk Preservation) Transfer Team (https://csdb.dk/group/?id=8137) has a rule of thumb for dumping disks: always dump your disks with the error block/map - it is a simple and powerful journal of how well that dump went.

The Disk Transfer Team uses the error block intensively - if you leave out the error block it is impossible to determine the state of the dump later.

Also from the error-block it will be possible to determine if the disk also should be g64 nibble-dumped during later inspection

PS: The Disk Transfer Team has built up their experience and routines over the last 20+ years - they have dumped an estimated total of 500000+ disks

PPS: I'm not an active (dumping) member of the Disk Transfer Team - but I'm closely affiliated with them.

movvny, 2024-08-19:
I see your point @UffeJakobsen, but in all cases you mentioned `--error-map=always` comes in handy.
The thing is, if a user explicitly wants for some special reason use `--bam-only` or `--bam-save` combined with `--error-map=on_error` the software behaves like `--error-map=always` anyway, so not like the user expects.
Since there is `--error-map=never` flag and it works as expected (although is strongly not recommeded), then why `--error-map=on_errors` should not work accordingly to its name?

BTW, what dumping tools / devices does The Transfer Team use?

PS. I'll be away for a couple of days.

UffeJakobsen, 2024-08-21:
> 
> BTW, what dumping tools / devices does The Transfer Team use?
> 

For many years the preferred C64 dumping toolchain setup was:

C64/1541 running WarpCopy server, RR-Net ethernet port/connection via cartridge (clockport extension) 
PC (Windows) with WarpCopy client GUI, ethernet port (directly) connected to the C64 RR-Net ethernet port

Links:
https://www.c64-wiki.com/wiki/RR-Net
https://icomp.de/shop-icomp/en/shop/product/rr-net-mk3.html
https://www.oxyron.de/html/wc64.html
https://csdb.dk/release/?id=17891

Nowadays the preferred C64 dumping toolchain setup is:

PC (Windows/Linux) running IECHost client GUI
IECHost device connected to PC via USB
1541 connected to IECHost device via serial cable

https://www.luigidifraia.com/hardware/#IECHost
https://www.luigidifraia.com/sales/#IECHost
https://www.luigidifraia.com/software/#IECHost_Clients

For G64 nibbling:

PC with OpenCBM/nibtools client software
ZoomFloppy/XUM1541 device
1541 with SpeedDOS compatible parallel cable *or* 1571 (without parallel cable)

PS: IECHost has recently got nibbling client software support - and could replace the ZoomFloppy/XUM1541 setup

thierer, 2024-10-16:
I think that the errormap should not contain invalid values (like 0).

- If the mode is `on_error` and there are no errors in the blocks that are actually transferred, no errormap should be appended. Else the errors for the affected sectors should be set as encountered and all other sectors (including omitted sectors) should have the status code 1.
- Likewise, if the mode is `always`, all sectors without explicit errors should get the status code 1, regardless if they have been actually copied or not.

I personally don't see the point of storing the information which sectors have been skipped in the errormap. (You wouldn't have that information for a physical, BAM-driven copy, where the destination disk has been formatted without errors before, either).

But if you really would like to preserve that information, instead of status code 0 I'd suggest to use status code 2 (header not found) for sectors which have not been copied and maybe 3 (no sync) for sectors on tracks which had no sector copied at all, but the latter would require a bit more effort.

UffeJakobsen, 2024-10-23:
Seen from a preservation perspective - I guess - that the information whether a sector is skipped is useful...