OpenCBM/OpenCBM #119: Undeclared compile-time dependency (cl65)? Is there a workaround?

KarlLevik, 2024-09-29:
On Fedora Linux:

Having first installed `libusb1` and `libusb1-devel` packages as advised by the installation instructions (chapter 4.3), and then downloaded the OpenCBM source code from https://github.com/OpenCBM/OpenCBM/archive/refs/tags/v0_4_99_104.tar.gz and attempting to compile, I receive the following error:

\<snip\>

```
make[1]: Entering directory '[REDACTED]/OpenCBM-0_4_99_104/opencbm/libtrans'
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
grep: warning: stray \ before #
cl65 -c --feature labels_without_colons --feature pc_assignment --feature loose_char_term --asm-include-dir .. -o pp1541.tmp pp1541.a65
make[1]: cl65: No such file or directory
make[1]: *** [../LINUX/config.make:236: pp1541.o65] Error 127
make[1]: Leaving directory '[REDACTED]/OpenCBM-0_4_99_104/opencbm/libtrans'
make: *** [LINUX/Makefile:83: BUILDSYSTEM.opencbm/libtrans.all] Error 2
``` 

Is there a workaround for this?

penfold42, 2024-09-30:
Try grabbing the latest source from master - ideally using git but 
https://github.com/OpenCBM/OpenCBM/archive/refs/heads/master.zip will get you a zip.

KarlLevik, 2024-09-30:
Hi @penfold42 

I have just attempted to compile using the most up-to-date code from the master branch, but I'm afraid I still get exactly the same compilation error.

penfold42, 2024-09-30:
on closer read it looks like you need cc65 installed:
https://github.com/cc65/

KarlLevik, 2024-09-30:
Thank you, @penfold42  - cc65 is available in the Fedora repo, so I did ``dnf install cc65``, then re-ran the ``make`` command, which was successful. Finally, I ran the "make install" command as given in the documentation, and this was also a success.

So, all that needs doing now is clearly stating that cc65 is a dependency in the documentation.

KarlLevik, 2024-09-30:
My XoomFloppy is scheduled to arrive in about two weeks, so fingers crossed it will work ...!

spiro-trikaliotis, 2024-09-30:
As stated in #118 , the sources should come with the pre-built o65 files (which are generated with the help of cc65). So, if the source tarball would be correct, there wouldn't be the need to have cc65 available.

Compiling from git, cc65 is always needed, however.

KarlLevik, 2024-09-30:
Thank you, @spiro-trikaliotis , that makes a lot of sense!