OpenCBM/OpenCBM #121: cbmforng and cbmformat not parsing the cmdline the same way

Pokefinder-org, 2024-11-12:
Been fiddling around with parameters to cbmforng for a gui but cannot work around this.

# cbmforng -nxvo 8 "- LALA,ID -"
cbmforng: invalid option -- ' '
Try `cbmforng' -h for more information.

To resolve this I checked:

# cbmformat -nxvo -- 8 "- LALA,ID -"

works fine to avoid the invalid option output cbmfornq returns:

cbmforng -nxvo -- 8 "- LALA,ID -"
Missing `,' in disk name or ID field not equal to two characters

Should be easy to fix? :)

spiro-trikaliotis, 2024-11-13:
I see that they should behave consistently.

However, can you tell me the purpose of an ID "ID -"? Or do you want the disk name to be "- LADA,ID -"? In this case, what ID should the disk have?

I am trying to understand what you want to achieve.

Pokefinder-org, 2024-11-13:
Oh, sorry - the ",ID -" part was just one of my test cases (including cbmcopy with " - " filenames, etc.) and was working with cbmformat -- so, sorta copy/paste error.

Dove into IDs a little more and found

"cbmforng -nxvo -- 8 "- LALA,ID" works as expected and the issue should really be smthg like:

* why does it work with cbmformat to pass (any) id size ?  - which is ok whenever cut off after two chars as it seems currently :)
* cbmforng does not accept IDs with a single char either while cbmformat does (and then produces the same "broken" directory as the original DOS. Custom formatters (e.g. Action Replay) fill single char IDs with a space to create a proper formatted disk.

Sorry - the issue name still fits but got slightly extended by these observations.