OpenCBM/OpenCBM #130: d82copy can't specify drive 1 as source or target

root42, 2025-05-25:
For the dual disk drives it would be nice to be able to specify the drive which should be used as target, i.e. drive 0 or drive 1. Right now it will default to drive 0 if there is a disk in drive 0.

spiro-trikaliotis, 2025-06-04:
Hi,

this feature request is feasible. However, I am not completely sure how the interface should look like.

I could image adding the drive number to the unit number. That is, doing something like

`d82copy 8:0 image0.d82`

to read from drive 0 on disk unit 8, or

`d82copy 8:1 image0.d82`

to read from drive 1 on disk unit 8.

Another option would be to add a command line parameter like `--drive=0` or `--drive=1` or `--drive0` or `--drive1`.

However, I have a problem with the name "drive". While Commodore used the name "unit number" and "drive number", the other OpenCBM tools use drive number to mean what Commodore called unit. So, adding an option like the above might add to the confusion.

Do you have any opinion or even a better idea?

spiro-trikaliotis, 2025-06-04:
As I do not have a 8250/8050/SFD1001 setup ready, I cannot test it here at the moment.

This patch [d82copy-add-medium.diff.gz](https://github.com/user-attachments/files/20598970/d82copy-add-medium.diff.gz) tries the approach with adding the drive number with a colon and the number. For example, 8:0 would be drive 0 on unit 8, while 8:1 would be drive 1 on unit 8. If you specify just 8, you still get the old behaviour (that is, "8" and "8:0" or synonymous).

Does this work for you? Please note that this patch does not test if the drive exists. That is, you can happily give drive 1 for an SFD1001.

root42, 2025-06-05:
Ok, I built it, but am having issues:
```
➜  ~ d82copy -d 8050 -1 /Users/aschmitz/Downloads/PET/chemistry.d80 8:1         
[Warning] non-standard number or tracks: 154                                    
 1: *****************************                                               
 2: ***--------------------------  1%    32/2083[Warning] write error: 02/03: 1 
 2: ***?-------------------------  1%    33/2083[Warning] write error: 02/03: 1 
 2: ***?-------------------------  1%    34/2083[Warning] write error: 02/03: 1 
...
```
It seems to work insofar as this is sending to the right drive, but it always aborts after the first track or so. Copying using cbmcopy works fine (but there I also can't select the drive 0/1, so same change might be useful there).

What is weird is that the d80 file SUPPOSEDLY has 1541 tracks, which is wrong. It is a standard 77 track image. Also I HAVE to specify 8050 and single sided or it will assume an 8250, because cbmctrl detect incorrectly returns an 8250 drive:
```
➜  ~ cbmctrl detect 
8: 8250 dos2.7     
```

root42, 2025-06-05:
The other way around (dumping an image) however works fine:
```
➜  PET d82copy -d 8050 -1 8:0 punter_bbs.d80
[Warning] growing image file to 2083 blocks 
 1: *****************************           
 2: *****************************           
 3: *****************************           
 4: *****************************           
...
```

root42, 2025-06-06:
PS: cbmctrl dir might also want do support this:
```
cbmtrl dir 8:1
```

spiro-trikaliotis, 2025-06-06:
> Only thing to mention: cbmcopy could use the 8:0 and 8:1 syntax as well. That would make everything quite perfect.

To be honest, at the moment, I don't want  to change cbmcopy. Its code is a mess, and I fear I might break something.
I am currently working on  a big rewrite of d64copy, d81copy, d82copy, imgcopy (because the code is not good and was copy-n-pasted from one into the other), and later also cbmcopy. I would like to leave it for later.

Regarding cbmctrl dir, it might make sense to add that syntax there, too, although you have found a work-around yourself.

spiro-trikaliotis, 2025-06-06:
BTW: This patch was added with commit 97d997e. Commit 40639d7 cleaned up the code a little bit.